evcc-io/evcc #19402: falsche Berechnung% Anteil Sonne am Ladevorgang

StevieC121176, 2025-03-04:
### Describe the bug

Ich habe einen HomematicIP Zwischenstecker als Loadpoint eingerichtet. Lade ich über MIN+PV oder SCHNELL wird der % Anteil Sonne falsch berechnet (Als ob Energie aus dem Netz bezogen wird, was aber nicht der Fall ist (ca. 7kw Einspeisung) Daher stimmen die ganzen Werte nicht wie Preis usw.  Lade ich über PV werden die Werte korrekt angezeigt.

Am Beispiel (Screenshot): 1,4kwh geladen 1,2kwh Sonnenanteil. Obwohl es 100% Sonnenanteil waren

[image]
[image]

### Steps to reproduce

Laden starten über min+PV oder Schnell

### Configuration details

```yaml
über UI Konfiguriert
```

### Log details

```text
log Level Track ist aktiviert, leider gibt er mir für den Loadpoint nur debug aus

[lp-3  ] DEBUG 2025/03/04 13:01:20 charge power: 976W
[lp-3  ] DEBUG 2025/03/04 13:01:30 charge power: 957W
[lp-3  ] DEBUG 2025/03/04 13:01:33 charge total import: 3.522kWh
[lp-3  ] DEBUG 2025/03/04 13:01:33 charger status: C
[lp-3  ] DEBUG 2025/03/04 13:01:33 vehicle soc: 20%
[lp-3  ] DEBUG 2025/03/04 13:01:33 pv charge current: 42.4A = 16A + 26.4A (-6063W @ 1p)
[lp-3  ] DEBUG 2025/03/04 13:01:40 charge power: 957W
[lp-3  ] DEBUG 2025/03/04 13:01:50 charge power: 957W
[lp-3  ] DEBUG 2025/03/04 13:02:00 charge power: 945W
[lp-3  ] DEBUG 2025/03/04 13:02:02 charge total import: 3.530kWh
[lp-3  ] DEBUG 2025/03/04 13:02:02 charger status: C
[lp-3  ] DEBUG 2025/03/04 13:02:02 vehicle soc: 20%
[lp-3  ] DEBUG 2025/03/04 13:02:02 pv charge current: 43A = 16A + 27A (-6203W @ 1p)
[lp-3  ] DEBUG 2025/03/04 13:02:10 charge power: 945W
[lp-3  ] DEBUG 2025/03/04 13:02:20 charge power: 945W
[lp-3  ] DEBUG 2025/03/04 13:02:30 charge power: 945W
[lp-3  ] DEBUG 2025/03/04 13:02:32 charge total import: 3.530kWh
[lp-3  ] DEBUG 2025/03/04 13:02:32 charger status: C
[lp-3  ] DEBUG 2025/03/04 13:02:32 vehicle soc: 20%
[lp-3  ] DEBUG 2025/03/04 13:02:32 pv charge current: 42.9A = 16A + 26.9A (-6192W @ 1p)
[lp-3  ] DEBUG 2025/03/04 13:02:40 charge power: 945W
[lp-3  ] DEBUG 2025/03/04 13:02:50 charge power: 945W
[lp-3  ] DEBUG 2025/03/04 13:03:00 charge power: 945W
[lp-3  ] DEBUG 2025/03/04 13:03:04 charge total import: 3.530kWh
[lp-3  ] DEBUG 2025/03/04 13:03:04 charger status: C
[lp-3  ] DEBUG 2025/03/04 13:03:04 vehicle soc: 20%
[lp-3  ] DEBUG 2025/03/04 13:03:04 pv charge current: 42A = 16A + 26A (-5972W @ 1p)
[lp-3  ] DEBUG 2025/03/04 13:03:10 charge power: 945W
[lp-3  ] DEBUG 2025/03/04 13:03:20 charge power: 945W
[lp-3  ] DEBUG 2025/03/04 13:03:30 charge power: 934W
[lp-3  ] DEBUG 2025/03/04 13:03:33 charge total import: 3.552kWh
[lp-3  ] DEBUG 2025/03/04 13:03:33 charger status: C
[lp-3  ] DEBUG 2025/03/04 13:03:33 vehicle soc: 20%
[lp-3  ] DEBUG 2025/03/04 13:03:33 pv charge current: 42.7A = 16A + 26.7A (-6132W @ 1p)
[lp-3  ] DEBUG 2025/03/04 13:03:40 charge power: 934W
[lp-3  ] DEBUG 2025/03/04 13:03:50 charge power: 934W
[lp-3  ] DEBUG 2025/03/04 13:04:00 charge power: 934W
[lp-3  ] DEBUG 2025/03/04 13:04:02 charge total import: 3.552kWh
[lp-3  ] DEBUG 2025/03/04 13:04:02 charger status: C
[lp-3  ] DEBUG 2025/03/04 13:04:02 vehicle soc: 20%
[lp-3  ] DEBUG 2025/03/04 13:04:02 pv charge current: 42.8A = 16A + 26.8A (-6172W @ 1p)
[lp-3  ] DEBUG 2025/03/04 13:04:10 charge power: 934W
[lp-3  ] DEBUG 2025/03/04 13:04:20 charge power: 934W
[lp-3  ] DEBUG 2025/03/04 13:04:30 charge power: 934W
[lp-3  ] DEBUG 2025/03/04 13:04:32 charge total import: 3.552kWh
[lp-3  ] DEBUG 2025/03/04 13:04:32 charger status: C
[... 402 lines cut ...]
[lp-3  ] DEBUG 2025/03/04 13:07:33 pv charge current: 41.3A = 16A + 25.3A (-5823W @ 1p)
[lp-3  ] DEBUG 2025/03/04 13:07:40 charge power: 923W
[lp-3  ] DEBUG 2025/03/04 13:07:50 charge power: 923W
[lp-3  ] DEBUG 2025/03/04 13:08:00 charge power: 914W
[lp-3  ] DEBUG 2025/03/04 13:08:02 charge total import: 3.623kWh
[lp-3  ] DEBUG 2025/03/04 13:08:02 charger status: C
[lp-3  ] DEBUG 2025/03/04 13:08:02 vehicle soc: 20%
[lp-3  ] DEBUG 2025/03/04 13:08:02 pv charge current: 42A = 16A + 26A (-5972W @ 1p)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5 (575d20cd)

VolkerK62, 2025-03-04:
OT:  Ich vermute, die Heizung hat einen "festen" Verbrauch. Dann solltest du  Min.- und Max.Ladestrom entsprechend der Leistung der Heizung anpassen. Da werden 16A freigegeben.

closed 2025-03-04 by andig

StevieC121176, 2025-03-04:
> Config fehlt

Hier die relevante config (Charger und vehicle)

```
chargers: 
  - type: template
    template: homematic
    host: 192.168.178.68
    port: 2010
    device: '0001D8XXXXXXX'
    standbypower: 15
    user: XXXX
    password: 'XXXX'
    meterchannel: 6
    switchchannel: 3
    integrateddevice: true
    Heating: true
    name: Heizung

vehicles:
  - name: heat
    type: custom
    title: Heizung Noel # display name for UI
    soc: # battery soc (%)
      source: mqtt
      topic: TempSZ
```