evcc-io/evcc #19407: Solar charging max current per phase

mmargi, 2025-03-04:
**Is your feature request related to a problem? Please describe.**

I have an installation with two 22kW chargers (Wallbox Pulsar Pro) and currently only a single-phase car with 1x32A, i.e., 7.2kW. Additionally, I have a Solax Ultra solar inverter, which can handle (under overload conditions) approximately 25A on a single phase. The house's grid connection is also limited to 25A per phase.

Due to the installation and the inverter's capabilities (after the latest update), I can charge up to 32A on a single phase in FAST mode. Therefore, I have set the charging current limit in EVCC to 32A. In FAST mode, everything works without issues.

However, if I want to use the Solar or Min+Solar mode, the limit remains set to 32A, and the charging current starts oscillating. EVCC detects power export to the grid or battery charging because the total solar power is sufficient, so it sets the maximum charging current to 32A. However, since the inverter itself cannot provide more than 25A per phase, this leads to grid import, which then triggers a reduction in power, and this cycle repeats continuously.

**Describe the solution you'd like**
The solution would be to have the ability to set different charging currents for each mode – one for FAST and a different one for SOLAR.

**Describe alternatives you've considered**
While writing this request, I realized that I could temporarily solve it with an automation in Home Assistant, which will set the max charging current to 25A when the Solar mode is activated. However, I still believe that having separate configuration options would be useful. :)

Thank you for your work. M.

mfuchs1984, 2025-03-04:
Load management should help, set the maximum current of the main circuit to 25.
https://docs.evcc.io/en/docs/features/loadmanagement

mmargi, 2025-03-04:
ok, i will try it. But is possible to use circuits somewhere in PV definititon? Because i can't tell to EVCC how value of current, which my PV system can provide to 1 phase.

Maybe will be interesting, when child (load point) has higher value ( 32A breaker)  then master 25A (main home breaker) in my case .:)

mfuchs1984, 2025-03-04:
Why is the maximum phase current of the inverter important? Is your grid energy meter not balancing phases? Ours (Germany) are. So when you pull 6 A on phase 1 and feed 3 A each on phases 2 and 3, the total power is 0.

mfuchs1984, 2025-03-04:
Please note there is a pending issue concerning maxacpower. A fix is in the nightly, but not yet in the release.