evcc-io/evcc #19420: OCPP: wrong "charge power" calculation

TheAgentHH, 2025-03-04:
### Describe the bug

OCPP charger, running on 1phase with 3,4kw result in charge power of 22080w

### Steps to reproduce

1. current build
2. OCPP start charging
3. charger display does not match with results in evcc
...

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
#  site: trace

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.220.16
  port: 1502
  usage: pv
  modbus: tcpip
  endianness: little
  minsoc: 5
  maxsoc: 100
  watchdog: 60s
  name: pv1
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.220.16
  port: 1502
  usage: battery
  modbus: tcpip
  endianness: little
  minsoc: 5
  maxsoc: 100
  watchdog: 60s
  name: battery2
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: 192.168.220.16
  port: 1502
  usage: grid
  modbus: tcpip
[... 332 lines cut ...]
[ocpp  ] TRACE 2025/03/04 22:07:02 send 84702764194358: [2,"3864442745","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/04 22:07:02 recv 84702764194358: [3,"3864442745",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/03/04 22:07:02 recv 84702764194358: [2,"16310653","MeterValues",{"connectorId":1,"transactionId":1741096347,"meterValue":[{"timestamp":"2025-03-04T21:06:53.333Z","sampledValue":[{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"113500"},{"measurand":"Power.Active.Import","unit":"kW","value":"3.4"},{"measurand":"Temperature","unit":"Celsius","value":"32"},{"measurand":"Voltage","phase":"L1-N","unit":"V","value":"229.5"},{"measurand":"Voltage","phase":"L2-N","unit":"V","value":"233.7"},{"measurand":"Voltage","phase":"L3-N","unit":"V","value":"236.5"},{"measurand":"Current.Import","phase":"L1","unit":"A","value":"14.9"},{"measurand":"Current.Import","phase":"L2","unit":"A","value":"0.0"},{"measurand":"Current.Import","phase":"L3","unit":"A","value":"0.0"}]}]}]
[ocpp  ] TRACE 2025/03/04 22:07:02 send 84702764194358: [3,"16310653",{}]
[ocpp  ] TRACE 2025/03/04 22:07:04 send 84702764194358: [2,"1860016795","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/04 22:07:04 recv 84702764194358: [3,"1860016795",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/03/04 22:07:04 recv 84702764194358: [2,"16310655","MeterValues",{"connectorId":1,"transactionId":1741096347,"meterValue":[{"timestamp":"2025-03-04T21:06:55.333Z","sampledValue":[{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"113500"},{"measurand":"Power.Active.Import","unit":"kW","value":"3.4"},{"measurand":"Temperature","unit":"Celsius","value":"32"},{"measurand":"Voltage","phase":"L1-N","unit":"V","value":"229.7"},{"measurand":"Voltage","phase":"L2-N","unit":"V","value":"233.8"},{"measurand":"Voltage","phase":"L3-N","unit":"V","value":"236.6"},{"measurand":"Current.Import","phase":"L1","unit":"A","value":"14.9"},{"measurand":"Current.Import","phase":"L2","unit":"A","value":"0.0"},{"measurand":"Current.Import","phase":"L3","unit":"A","value":"0.0"}]}]}]
[ocpp  ] TRACE 2025/03/04 22:07:04 send 84702764194358: [3,"16310655",{}]
[ocpp  ] TRACE 2025/03/04 22:07:06 send 84702764194358: [2,"3388824067","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/04 22:07:06 recv 84702764194358: [3,"3388824067",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/03/04 22:07:06 recv 84702764194358: [2,"16310656","MeterValues",{"connectorId":1,"transactionId":1741096347,"meterValue":[{"timestamp":"2025-03-04T21:06:56.333Z","sampledValue":[{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"113500"},{"measurand":"Power.Active.Import","unit":"kW","value":"3.4"},{"measurand":"Temperature","unit":"Celsius","value":"32"},{"measurand":"Voltage","phase":"L1-N","unit":"V","value":"229.7"},{"measurand":"Voltage","phase":"L2-N","unit":"V","value":"233.7"},{"measurand":"Voltage","phase":"L3-N","unit":"V","value":"236.4"},{"measurand":"Current.Import","phase":"L1","unit":"A","value":"14.9"},{"measurand":"Current.Import","phase":"L2","unit":"A","value":"0.0"},{"measurand":"Current.Import","phase":"L3","unit":"A","value":"0.0"}]}]}]
[ocpp  ] TRACE 2025/03/04 22:07:06 send 84702764194358: [3,"16310656",{}]
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

mfuchs1984, 2025-03-04:
Your charger shows some ocpp issues. Particularly the fact that it rejects all measurand configuration attempts  with `NotSupported makes` evcc fall back to not using the reported values at all. Evcc requests 32 A, this makes 230*3*32=22080.

But there is a way out of that misery. First, clean up your charger config, it contains numerous obsolete options, then explicitly add the measurands it is able to provide despite rejection.

```
chargers:
- type: template
  template: ocpp
  stationid: 84702764194358
  name: wallbox2
  metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import,Voltage
```

TheAgentHH, 2025-03-04:
cleanup done and will test the adjusted chargers config with next battery load.

Details about charger:
Em2go (aka EVSE) EM2GO Model EM022AS0OCPP FW:AC_SCE_3.12EG

TheAgentHH, 2025-03-04:
Glücklich bin ich mit dem auch nicht, aber er hängt leider schon an der Wand..
Aber richtig gut integrieren lässt er sich leider wirklich nicht..

mfuchs1984, 2025-03-04:
Gibt schlimmere ;).

premultiply, 2025-03-05:
/cc @dniakamal