evcc-io/evcc #19421: open-meteo solar forecast not working

VolkerK62, 2025-03-05:
### Describe the bug

Fehler beim Starten. Es wird keine Vorhersage angezeigt.

### Steps to reproduce

.

### Configuration details

```yaml
"GUI configuration"

currency: EUR
grid:
    type: template
    bzn: DE-LU
    charges: 0.17
    tax: 0.19
    template: energy-charts-api
feedin:
    type: fixed
    price: 0.11
co2:
    type: ""
planner:
    type: ""
solar:
    type: template
    template: open-meteo
    lat: 49.31
    lon: 8.54
    dec: 42
    az: -28
    kwp: 7.5
    ac: 8
    interval: 1h
```

### Log details

```text
"evcc tariff solar"

[main  ] INFO 2025/03/05 06:17:58 evcc 0.200.5 (b6384457)
[main  ] INFO 2025/03/05 06:17:58 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/05 06:17:58 using sqlite database: /home/wsl/.evcc/evcc.db
[db    ] TRACE 2025/03/05 06:17:58 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/03/05 06:17:58 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/03/05 06:17:58 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/03/05 06:17:58 SELECT * FROM `settings` 14 <nil>
[mqtt  ] INFO 2025/03/05 06:17:59 connecting evcc-1049903813 at tcp://192.168.178.25:1883
[mqtt  ] DEBUG 2025/03/05 06:17:59 tcp://192.168.178.25:1883 connected
[db    ] TRACE 2025/03/05 06:17:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2025/03/05 06:17:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2025/03/05 06:17:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/03/05 06:17:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/03/05 06:17:59 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/03/05 06:17:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2025/03/05 06:17:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main  ] ERROR 2025/03/05 06:17:59 creating tariff solar failed: cannot create tariff type 'template': cannot create tariff type 'custom': forecast: invalid jq query 'def alphatemp: -0.004; # temperature coefficient
def rossmodel: '0.0342'; # cooling type
def eff: 100 / 100; # efficiency 1 = 100%
def kwp: 7.5 * 1000 ; # kWp
def ac: 8 * 1000 ; # AC
def dm: 0 / 100; # 1 = 100% damping morning (0 = no damping) 
def de: 0 / 100; # damping evening
def clamp(min; x; max):
  if x < min then min elif x > max then max else x end;
def midday(sunrise; sunset):
  sunrise + ((sunset - sunrise) / 2);
def calculate_damping(time; sunrise; sunset; m; e):
  if time < sunrise then m
  elif time < midday(sunrise; sunset) then
    m * (1 - ((time - sunrise) / (midday(sunrise; sunset) - sunrise)))
  elif time < sunset then
    e * ((time - midday(sunrise; sunset)) / (sunset - midday(sunrise; sunset)))
  else
    e
  end;
.hourly as $h
| .daily as $d
| [ range(0; ($h.time | length))
    | . as $i
    | $h.time[$i] as $time
    | ($i / 24 | floor) as $day
    | {
        start: ($time | todateiso8601),
        end: (($time + 3600) | todateiso8601),
        price: (
          kwp
          * ($h.global_tilted_irradiance[$i] / 1000)
          * (1 + ( alphatemp *
                (
                  ( ($h.temperature_2m[$i]
                      + ($h.temperature_2m[$i-1] // $h.temperature_2m[$i])
                    ) / 2)
                  + ($h.global_tilted_irradiance[$i] / 800.0) * rossmodel - 25.0
                )
              ))
          * (1 - calculate_damping($time;
                $d.sunrise[$day];
                $d.sunset[$day];
                dm; de))
          * eff
          | clamp(0; .; ac)
        )
      }
  ]
| tostring
': unexpected token "'"
[main  ] FATAL 2025/03/05 06:17:59 tariff not available: cannot create tariff type 'template': cannot create tariff type 'custom': forecast: invalid jq query 'def alphatemp: -0.004; # temperature coefficient
def rossmodel: '0.0342'; # cooling type
def eff: 100 / 100; # efficiency 1 = 100%
def kwp: 7.5 * 1000 ; # kWp
def ac: 8 * 1000 ; # AC
def dm: 0 / 100; # 1 = 100% damping morning (0 = no damping) 
def de: 0 / 100; # damping evening
def clamp(min; x; max):
  if x < min then min elif x > max then max else x end;
def midday(sunrise; sunset):
  sunrise + ((sunset - sunrise) / 2);
def calculate_damping(time; sunrise; sunset; m; e):
  if time < sunrise then m
  elif time < midday(sunrise; sunset) then
    m * (1 - ((time - sunrise) / (midday(sunrise; sunset) - sunrise)))
  elif time < sunset then
    e * ((time - midday(sunrise; sunset)) / (sunset - midday(sunrise; sunset)))
  else
    e
  end;
.hourly as $h
| .daily as $d
| [ range(0; ($h.time | length))
    | . as $i
    | $h.time[$i] as $time
    | ($i / 24 | floor) as $day
    | {
        start: ($time | todateiso8601),
        end: (($time + 3600) | todateiso8601),
        price: (
          kwp
          * ($h.global_tilted_irradiance[$i] / 1000)
          * (1 + ( alphatemp *
                (
                  ( ($h.temperature_2m[$i]
                      + ($h.temperature_2m[$i-1] // $h.temperature_2m[$i])
                    ) / 2)
                  + ($h.global_tilted_irradiance[$i] / 800.0) * rossmodel - 25.0
                )
              ))
          * (1 - calculate_damping($time;
                $d.sunrise[$day];
                $d.sunset[$day];
                dm; de))
          * eff
          | clamp(0; .; ac)
        )
      }
  ]
| tostring
': unexpected token "'"
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5 (b6384457)

andig, 2025-03-05:
/cc @thecem

CiNcH83, 2025-03-05:
Da muss wohl das rossmodel "unquotet" werden?

closed 2025-03-05 by andig

VolkerK62, 2025-03-05:
>Evtl. mal probieren, das rossmodel in die Config mit aufnehmen?

Hab ich gemacht. Gleiches Ergebnis