evcc-io/evcc #19423: target soc Always 100% after reconnect

heatpumper, 2025-03-05:
### Describe the bug

Dear all,

Since moving to the gui config I Encounter the following issue:
- i cannot Set a target soc in the gui
- whenever I replug my car, the target soc ist Set top 100%
How can i Limit IT to 80% as a Standard?
Wallbox: Go e charger
Car VW ID 3

I did Check only with the latest released Version.

```
[lp-2 ] DEBUG 2025/03/05 07:44:31 set session soc limit: 80
[lp-2 ] DEBUG 2025/03/05 07:44:31 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:44:31 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:44:31 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:44:31 charger status: B
[lp-2 ] DEBUG 2025/03/05 07:44:31 !! pvScalePhases DOWN activePhases: 1, available power: -86W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:44:31 !! pvScalePhases UP maxPhases: 3, available power: -86W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:44:31 !! pvScalePhases EXIT
[lp-2 ] DEBUG 2025/03/05 07:44:31 pv charge current: 0A = 0A + -0.374A (86W @ 1p)
[lp-2 ] DEBUG 2025/03/05 07:44:41 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:44:41 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:44:41 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:44:41 charger status: B
[lp-2 ] DEBUG 2025/03/05 07:44:41 !! pvScalePhases DOWN activePhases: 1, available power: 82W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:44:41 !! pvScalePhases UP maxPhases: 3, available power: 82W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:44:41 !! pvScalePhases EXIT
[lp-2 ] DEBUG 2025/03/05 07:44:41 pv charge current: 0.358A = 0A + 0.358A (-82W @ 1p)
[lp-2 ] DEBUG 2025/03/05 07:44:56 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:44:56 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:11 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:45:11 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:11 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:45:11 charger status: A
[lp-2 ] INFO 2025/03/05 07:45:11 car disconnected
[lp-2 ] DEBUG 2025/03/05 07:45:11 set charge mode: pv
[lp-2 ] DEBUG 2025/03/05 07:45:11 set battery boost: false
[lp-2 ] DEBUG 2025/03/05 07:45:11 set session soc limit: 0
[lp-2 ] DEBUG 2025/03/05 07:45:11 set session energy limit: 0
[lp-2 ] WARN 2025/03/05 07:45:11 charger out of sync: expected disabled, got enabled
[lp-2 ] DEBUG 2025/03/05 07:45:11 charger enable
[lp-2 ] DEBUG 2025/03/05 07:45:12 charger disable
[lp-2 ] DEBUG 2025/03/05 07:45:12 wake-up timer: stop
[lp-2 ] DEBUG 2025/03/05 07:45:12 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:45:12 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:12 vehicle odometer: 12442km
[lp-2 ] DEBUG 2025/03/05 07:45:12 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:45:12 charger status: A
[lp-2 ] DEBUG 2025/03/05 07:45:26 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:45:26 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:45:41 charger status: B
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge start energy: 2475.388kWh
[lp-2 ] INFO 2025/03/05 07:45:41 car connected
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv timer elapse
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv timer inactive
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:45:41 vehicle status: B
[lp-2 ] DEBUG 2025/03/05 07:45:41 vehicle soc: 78%
[lp-2 ] DEBUG 2025/03/05 07:45:41 vehicle soc limit: 100%
[lp-2 ] DEBUG 2025/03/05 07:45:41 vehicle range: 231km
[lp-2 ] DEBUG 2025/03/05 07:45:41 !! pvScalePhases DOWN activePhases: 1, available power: 43W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:45:41 !! pvScalePhases UP maxPhases: 3, available power: 43W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:45:41 !! pvScalePhases EXIT
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv charge current: 0.189A = 0A + 0.189A (-43W @ 1p)
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv enable timer reset
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv timer inactive 
```

### Steps to reproduce

1.set target soc to 80%
2. Unplug and Plug the car
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
[lp-2 ] DEBUG 2025/03/05 07:44:31 set session soc limit: 80
[lp-2 ] DEBUG 2025/03/05 07:44:31 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:44:31 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:44:31 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:44:31 charger status: B
[lp-2 ] DEBUG 2025/03/05 07:44:31 !! pvScalePhases DOWN activePhases: 1, available power: -86W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:44:31 !! pvScalePhases UP maxPhases: 3, available power: -86W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:44:31 !! pvScalePhases EXIT
[lp-2 ] DEBUG 2025/03/05 07:44:31 pv charge current: 0A = 0A + -0.374A (86W @ 1p)
[lp-2 ] DEBUG 2025/03/05 07:44:41 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:44:41 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:44:41 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:44:41 charger status: B
[lp-2 ] DEBUG 2025/03/05 07:44:41 !! pvScalePhases DOWN activePhases: 1, available power: 82W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:44:41 !! pvScalePhases UP maxPhases: 3, available power: 82W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:44:41 !! pvScalePhases EXIT
[lp-2 ] DEBUG 2025/03/05 07:44:41 pv charge current: 0.358A = 0A + 0.358A (-82W @ 1p)
[lp-2 ] DEBUG 2025/03/05 07:44:56 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:44:56 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:11 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:45:11 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:11 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:45:11 charger status: A
[lp-2 ] INFO 2025/03/05 07:45:11 car disconnected
[lp-2 ] DEBUG 2025/03/05 07:45:11 set charge mode: pv
[lp-2 ] DEBUG 2025/03/05 07:45:11 set battery boost: false
[lp-2 ] DEBUG 2025/03/05 07:45:11 set session soc limit: 0
[lp-2 ] DEBUG 2025/03/05 07:45:11 set session energy limit: 0
[lp-2 ] WARN 2025/03/05 07:45:11 charger out of sync: expected disabled, got enabled
[lp-2 ] DEBUG 2025/03/05 07:45:11 charger enable
[lp-2 ] DEBUG 2025/03/05 07:45:12 charger disable
[lp-2 ] DEBUG 2025/03/05 07:45:12 wake-up timer: stop
[lp-2 ] DEBUG 2025/03/05 07:45:12 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:45:12 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:12 vehicle odometer: 12442km
[lp-2 ] DEBUG 2025/03/05 07:45:12 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:45:12 charger status: A
[lp-2 ] DEBUG 2025/03/05 07:45:26 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:45:26 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge power: 0W
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:45:41 charger status: B
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge start energy: 2475.388kWh
[lp-2 ] INFO 2025/03/05 07:45:41 car connected
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv timer elapse
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv timer inactive
[lp-2 ] DEBUG 2025/03/05 07:45:41 charge total import: 2475.388kWh
[lp-2 ] DEBUG 2025/03/05 07:45:41 vehicle status: B
[lp-2 ] DEBUG 2025/03/05 07:45:41 vehicle soc: 78%
[lp-2 ] DEBUG 2025/03/05 07:45:41 vehicle soc limit: 100%
[lp-2 ] DEBUG 2025/03/05 07:45:41 vehicle range: 231km
[lp-2 ] DEBUG 2025/03/05 07:45:41 !! pvScalePhases DOWN activePhases: 1, available power: 43W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:45:41 !! pvScalePhases UP maxPhases: 3, available power: 43W, scalable: false
[lp-2 ] DEBUG 2025/03/05 07:45:41 !! pvScalePhases EXIT
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv charge current: 0.189A = 0A + 0.189A (-43W @ 1p)
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv enable timer reset
[lp-2 ] DEBUG 2025/03/05 07:45:41 pv timer inactive
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

200.5

bastian1607, 2025-03-05:
Hast du was in der yaml konfiguriert? Kannst du die auch noch posten?

heatpumper, 2025-03-05:
Die yaml ist für diesen Loadpoint komplett leer geräumt

StefanSchoof, 2025-03-05:
Hast du das Standard Ladelimit gesetzt: https://docs.evcc.io/docs/features/limits#einstellung

heatpumper, 2025-03-05:
Perfekt danke, das hab ich gesucht!

closed 2025-03-05 by heatpumper