evcc-io/evcc #19440: Lastmanagement: charger flapping on and off - evauates 0W->7360W twice

pocki80, 2025-03-05:
### Describe the bug

Hi, I use a Shelly 3EM Pro on the mains and an EVBox-Livo charger with a car that's capable of 7.4kW charging power. My main circuit is limited to power=10000W and currents=25A.

When I change the charging mode from off to fast, the car charges for some seconds and then stops. In the log's I see that evcc recognizes and over-power > 10000W - that's wrong because evcc has the 7.4kW load from shelly already befor it evaluates the "remaining power availablily" to switch the loadpoint from 0W to 7360W and thus stops the charge.

### Steps to reproduce

1. switch mode to off, wait some seconds to get everything into sync
2. switch mode to fast
3. evcc will show "waiting for vehicle to load"...
4. load will start, consumption will jump up for 7000W
5. some seconds later changing stops and consumption drops 7000W again to normal
6. some 15-20s later steps repeat from point 3

### Configuration details

```yaml
log: debug
levels:
  cache: error

interval: 10s

meters:
- type: template
  template: shelly-pro-3em
  usage: grid
  host: 192.168.8.207
  name: grid1
- name: solarpower
  type: custom
  power:
    source: http
    uri: http://localhost/pv.py
    method: GET
    timeout: 3s

circuits:
  - name: main
    title: Hauszuleitung
    maxCurrent: 25
    maxPower: 10000
    meter: grid1

chargers:
- type: template
  template: livo
  ip: 192.168.x.x
  name: wallbox3

vehicles:
- type: template
  template: audi
  title: A6 Avant TFSIe
  capacity: 11
  phases: 2
  minCurrent: 5
  maxCurrent: 16
  cache: 15m
  name: ev2
  mode: pv

loadpoints:
- title: KSC-EVBox
  charger: wallbox3
  mode: off
  circuit: main

site:
  title: KSC
  meters:
    grid: grid1
    pv: solarpower

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          -----END CERTIFICATE-----
```

### Log details

```text
[site ] DEBUG 2025/03/05 10:00:54 ----
[lp-1 ] DEBUG 2025/03/05 10:00:54 charge power: 0W
[circuit-main] DEBUG 2025/03/05 10:00:55 power: 277.39W
[circuit-main] DEBUG 2025/03/05 10:00:55 current: 1.96A
[site ] DEBUG 2025/03/05 10:00:55 pv 1 power: 242W
[site ] DEBUG 2025/03/05 10:00:55 grid power: 290W
[site ] DEBUG 2025/03/05 10:00:55 grid powers: [-101 359 32]W
[site ] DEBUG 2025/03/05 10:00:55 grid currents: [-1.12 1.96 0.467]A
[site ] DEBUG 2025/03/05 10:00:55 site power: 290W
[lp-1 ] DEBUG 2025/03/05 10:00:55 charger status: B

[site ] DEBUG 2025/03/05 10:01:04 ----
[lp-1 ] DEBUG 2025/03/05 10:01:04 charge power: 0W
[circuit-main] DEBUG 2025/03/05 10:01:05 power: 7454.4W
[circuit-main] DEBUG 2025/03/05 10:01:05 current: 15.4A
[site ] DEBUG 2025/03/05 10:01:05 pv 1 power: 242W
[site ] DEBUG 2025/03/05 10:01:05 grid power: 7458W
[site ] DEBUG 2025/03/05 10:01:05 grid powers: [3506 373 3579]W
[site ] DEBUG 2025/03/05 10:01:05 grid currents: [15.2 2.08 15.3]A
[site ] DEBUG 2025/03/05 10:01:05 site power: 7458W
[lp-1 ] DEBUG 2025/03/05 10:01:05 charger status: C
[lp-1 ] INFO 2025/03/05 10:01:05 start charging ->
[circuit-main] DEBUG 2025/03/05 10:01:05 validate power: 7454.4W + (0W -> 7360W) > 10000W capped at 2545.6W
[lp-1 ] DEBUG 2025/03/05 10:01:05 charger disable

[site ] DEBUG 2025/03/05 10:01:14 ----
[lp-1 ] DEBUG 2025/03/05 10:01:14 charge power: 0W
[circuit-main] DEBUG 2025/03/05 10:01:15 power: 269.73W
[circuit-main] DEBUG 2025/03/05 10:01:15 current: 2A
[site ] DEBUG 2025/03/05 10:01:15 pv 1 power: 242W
[site ] DEBUG 2025/03/05 10:01:15 grid power: 291W
[site ] DEBUG 2025/03/05 10:01:15 grid powers: [-99 358 32]W
[site ] DEBUG 2025/03/05 10:01:15 grid currents: [-0.853 2 0.267]A
[site ] DEBUG 2025/03/05 10:01:15 site power: 291W
[lp-1 ] DEBUG 2025/03/05 10:01:15 charger status: B
[lp-1 ] INFO 2025/03/05 10:01:15 stop charging <-
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.5

andig, 2025-03-05:
10s interval sind zu kurz. Warnungen beachten!

andig, 2025-03-05:
Sehe ich nicht- hier ist nur ein Einschaltvorgang, kein flapping.

pocki80, 2025-03-05:
Hier ein längerer Log mit an/aus, an/aus,...

30s interval

Vorgefiltert:
```
[site ] DEBUG 2025/03/05 12:34:33 ----
[lp-1 ] DEBUG 2025/03/05 12:34:34 charger status: B
[lp-1 ] DEBUG 2025/03/05 12:34:34 charger enable

[site ] DEBUG 2025/03/05 12:35:09 ----
[lp-1 ] INFO 2025/03/05 12:35:10 start charging ->
[circuit-main] DEBUG 2025/03/05 12:35:12 validate power: 7260.6W + (0W -> 7360W) > 10000W capped at 2739.4W
[lp-1 ] DEBUG 2025/03/05 12:35:12 charger disable

[site ] DEBUG 2025/03/05 12:35:39 ----
[lp-1 ] INFO 2025/03/05 12:35:40 stop charging <-
[lp-1 ] DEBUG 2025/03/05 12:35:41 charger enable

[site ] DEBUG 2025/03/05 12:36:09 ----
[lp-1 ] INFO 2025/03/05 12:36:10 start charging ->
[circuit-main] DEBUG 2025/03/05 12:36:10 validate power: 8367.7W + (0W -> 7360W) > 10000W capped at 1632.3W
[lp-1 ] DEBUG 2025/03/05 12:36:10 charger disable

[site ] DEBUG 2025/03/05 12:36:39 ----
[lp-1 ] INFO 2025/03/05 12:36:40 stop charging <-
[lp-1 ] DEBUG 2025/03/05 12:36:41 charger enable

[site ] DEBUG 2025/03/05 12:37:09 ----[lp-1 ] INFO 2025/03/05 12:37:10 start charging ->
[circuit-main] DEBUG 2025/03/05 12:37:10 validate power: 8203.6W + (0W -> 7360W) > 10000W capped at 1796.4W
[lp-1 ] DEBUG 2025/03/05 12:37:10 charger disable
```

und hier der komplette Log ohne eebus
```
[site ] DEBUG 2025/03/05 12:34:33 ----
[lp-1 ] DEBUG 2025/03/05 12:34:33 charge power: 0W
[circuit-main] DEBUG 2025/03/05 12:34:33 power: 51.83W
[circuit-main] DEBUG 2025/03/05 12:34:33 current: 1.59A
[site ] DEBUG 2025/03/05 12:34:33 pv 1 power: 293W
[site ] DEBUG 2025/03/05 12:34:33 grid power: 52W
[site ] DEBUG 2025/03/05 12:34:33 grid powers: [-188 209 31]W
[site ] DEBUG 2025/03/05 12:34:33 grid currents: [-1.16 1.59 0.26]A
[site ] DEBUG 2025/03/05 12:34:34 site power: 52W
[lp-1 ] DEBUG 2025/03/05 12:34:34 charger status: B
[lp-1 ] DEBUG 2025/03/05 12:34:34 plan: charge 59m43s between 2025-03-05 13:00:00 +0100 CET until 2025-03-05 16:00:00 +0100 CET (power: 7360W, avg cost: 0.144)
[lp-1 ] DEBUG 2025/03/05 12:34:34 max charge current: 16A
[lp-1 ] DEBUG 2025/03/05 12:34:34 charger enable
[lp-1 ] DEBUG 2025/03/05 12:34:34 wake-up timer: start

[site ] DEBUG 2025/03/05 12:34:39 ----
[lp-1 ] DEBUG 2025/03/05 12:34:39 charge power: 0W
[circuit-main] DEBUG 2025/03/05 12:34:40 power: 53.614W
[circuit-main] DEBUG 2025/03/05 12:34:40 current: 1.61A
[site ] DEBUG 2025/03/05 12:34:40 pv 1 power: 292W
[site ] DEBUG 2025/03/05 12:34:40 grid power: 53W
[site ] DEBUG 2025/03/05 12:34:40 grid powers: [-192 215 31]W
[site ] DEBUG 2025/03/05 12:34:40 grid currents: [-1.48 1.61 0.524]A
[site ] DEBUG 2025/03/05 12:34:40 site power: 53W
[lp-1 ] DEBUG 2025/03/05 12:34:40 charger status: B
[lp-1 ] DEBUG 2025/03/05 12:34:40 plan: charge 59m43s between 2025-03-05 13:00:00 +0100 CET until 2025-03-05 16:00:00 +0100 CET (power: 7360W, avg cost: 0.144)

[site ] DEBUG 2025/03/05 12:35:09 ----
[lp-1 ] DEBUG 2025/03/05 12:35:09 charge power: 0W
[circuit-main] DEBUG 2025/03/05 12:35:10 power: 7260.6W
[circuit-main] DEBUG 2025/03/05 12:35:10 current: 15.4A
[site ] DEBUG 2025/03/05 12:35:10 pv 1 power: 293W
[site ] DEBUG 2025/03/05 12:35:10 grid power: 7266W
[site ] DEBUG 2025/03/05 12:35:10 grid powers: [3483 220 3564]W
[site ] DEBUG 2025/03/05 12:35:10 grid currents: [15 1.62 15.4]A
[... 65 lines cut ...]
[lp-1 ] DEBUG 2025/03/05 12:36:41 vehicle range: 14km
[lp-1 ] DEBUG 2025/03/05 12:36:41 plan: charge 14s between 2025-03-05 13:00:00 +0100 CET until 2025-03-05 16:00:00 +0100 CET (power: 7360W, avg cost: 0.144)
[lp-1 ] DEBUG 2025/03/05 12:36:41 max charge current: 16A
[lp-1 ] DEBUG 2025/03/05 12:36:41 charger enable
[lp-1 ] DEBUG 2025/03/05 12:36:41 wake-up timer: start

[site ] DEBUG 2025/03/05 12:37:09 ----
[lp-1 ] DEBUG 2025/03/05 12:37:09 charge power: 0W
[circuit-main] DEBUG 2025/03/05 12:37:10 power: 8203.6W
[circuit-main] DEBUG 2025/03/05 12:37:10 current: 15.4A
[site ] DEBUG 2025/03/05 12:37:10 pv 1 power: 292W
[site ] DEBUG 2025/03/05 12:37:10 grid power: 8484W
[site ] DEBUG 2025/03/05 12:37:10 grid powers: [3490 1427 3567]W
[site ] DEBUG 2025/03/05 12:37:10 grid currents: [15 6.34 15.4]A
[site ] DEBUG 2025/03/05 12:37:10 site power: 8484W
[lp-1 ] DEBUG 2025/03/05 12:37:10 charger status: C
[lp-1 ] INFO 2025/03/05 12:37:10 start charging ->
[lp-1 ] DEBUG 2025/03/05 12:37:10 wake-up timer: stop
[lp-1 ] DEBUG 2025/03/05 12:37:10 soc estimated: 90.00% (vehicle: 90.00%)
[lp-1 ] DEBUG 2025/03/05 12:37:10 vehicle soc: 90%
[lp-1 ] DEBUG 2025/03/05 12:37:10 vehicle range: 14km
[lp-1 ] DEBUG 2025/03/05 12:37:10 plan: charge 14s between 2025-03-05 13:00:00 +0100 CET until 2025-03-05 16:00:00 +0100 CET (power: 7360W, avg cost: 0.144)
[circuit-main] DEBUG 2025/03/05 12:37:10 validate power: 8203.6W + (0W -> 7360W) > 10000W capped at 1796.4W
[lp-1 ] DEBUG 2025/03/05 12:37:10 charger disable
```

mfuchs1984, 2025-03-05:
`charge power: 0W` dürfte das Problem sein. Dein charger meldet nie eine Ladeleistung. Dadurch muss evcc davon ausgehen, dass die Leistung am grid meter von was anderem konsumiert wird und schaltet den charger ab. Funktioniert charge power nie, oder braucht die Wallbox einfach sehr lange? Im zweiten Fall kannst du das Intervall noch höher setzen. In ersten bleibt nur die Wallbox reparieren, austauschen oder mit einem externen meter (z.B. noch ein Shelly) auszustatten.

mfuchs1984, 2025-03-06:
Wenn du das Lastmanagement ausschaltet, zeigt dass evcc dann irgendwann eine Ladeleistung an?

pocki80, 2025-03-06:
Ja, sobald der charger auf Status "lädt" wechselt, wird die freigegebene Current mal Phasen als Ladeleistung angegeben, das sind bei mir 2x16A also 7.4kW. 
Wenn ich aber den custom-Meter einbinde, stimmt zwar chargeCurrent, doch es bleibt charging=false und chargePower=0 und state=B

mfuchs1984, 2025-03-06:
@andig kannst du das in eine discussion umwandeln?