evcc-io/evcc #19446: Fehler beim Eingeben des Sponsor Tokens über UI

flumb, 2025-03-05:
### Describe the bug

Hallo zusammen, 

ich versuche den Sponsortoken über die Oberfläche einzugeben. Leere evcc.yaml. 
Wenn ich den Sponsortoken in die Textbox kopiere, dann bleibt der "Speichern" button inaktiv. 
Gebe ich den Sponsortoken manuell ein erhalte ich folgende Fehlermeldung: 
sponsortoken: rpc error: code = Unknown desc = token is malformed: could not JSON decode header: invalid character 'Ò' after object key

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
frisch aufgesetztes evcc mit leerer evcc.yaml
```

### Log details

```text
sponsortoken: rpc error: code = Unknown desc = token is malformed: could not JSON decode header: invalid character 'Ò' after object key
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

ChristophWeigert, 2025-03-05:
Wenn du den [Demo Key](sponsortoken: ) verwendest, einfach den Kommentar `# trial token, valid until 2025-03-10`  und `sponsortoken: ` entfernen. So hat es bei mir geklappt.

flumb, 2025-03-05:
Danke für die Antwort. Ich verwende einen echten eigenen Sponsortoken. Ich kopiere auch nur den Key ohne #trial token ....

naltatis, 2025-03-05:
Dass der Speichern Button nach einem Paste inaktiv bleibt sollte nicht so sein. Schau ich mir an.

Probier mal bitte folgendes:

- Token reinpasten
- Inhalt verändern (bspw. ein Zeichen einfügen und wieder löschen)
- Speichern

deadrabbit87, 2025-03-05:
Ich hatte das auch fest gestellt. 

Fügt man einen Umbruch nach dem Token ein, geht es. 

Läuft aber technisch vermutlich auf das gleiche hinaus...

closed 2025-03-05 by andig