evcc-io/evcc #19450: Wallbox without measurement: reduce chargePower by unknownImport

pocki80, 2025-03-05:
Hi. Ich verwende eine eebus-Wallbox (EVBox Livo), die leider keine realen Leistungsdaten angibt. evcc vermutet deshalb, dass die Wallbox immer mit der freigegebenen Leistung lädt, was über 85% SoC nicht stimmt.

Im Powerflow ist dieser Zustand erkennbar durch eine unbekannte Leistungsquelle, siehe Screenshot anbei: 
[image]

IN:
PV liefert 0.3kW (gemessen durch einen Meter, korrekt)
Grid: 4,8kW (gemessen von Shelly 3EM Pro, korrekt)
Unknown: 2.3kW

OUT:
Ladepunkt: 7.4kW (16a@2p freigegeben, lädt aber weniger)
Consumption: 0kW, ist aber vermutlich mehr

Tatsächlich lädt die Wallbox nicht mehr mit 7.4kW, sondern eher 7.4-2.3=5.1kW oder sogar noch weniger, da die Haus-Consumption unbekannt ist.

Gibt es ein Feature, um die Ladeleistung um den unbekannten-Import zu reduzieren? Das würde die geladene Leistungssumme deutlich richtiger machen, wenn auch nicht gänzlich korrekt (der unbekannte Hausverbrauch würde als Ladeleistung missinterpretiert werden).

RacerEdge, 2025-03-05:
Hallo,
so einen Zustand habe ich auch.
[image]

Akku wird auch nur noch mit 2kw entladen.
Ging letzten Sommer noch in der gleichen Konstellation.

Gruß

RacerEdge, 2025-03-05:
[image]

premultiply, 2025-03-05:
Dagegen hilft ein einfaches Mittel: externen Zähler vor der Wallbox einsetzen und mit dem Ladepunkt verknüpfen.
Modbus-Zähler wie z.B. der Eastron SDM72DMv2 kosten nicht die Welt.

RacerEdge, 2025-03-05:
Die Leistung der Wallbox wird ja korrekt angezeigt. Das Verhältnis des Bezugs wird falsch angezeigt.
Und im August lief es noch perfekt. Die anderen Werte der Huawei Sun2000 werden auch korrekt angezeigt und die maximale Entladeleistung liegt bei 4kw.

VolkerK62, 2025-03-05:
>Gibt es ein Feature, um die Ladeleistung um den unbekannten-Import zu reduzieren? Das würde die geladene Leistungssumme deutlich richtiger machen, wenn auch nicht gänzlich korrekt (der unbekannte Hausverbrauch würde als Ladeleistung missinterpretiert werden).

Als Software-Version ein Charge-Meter konfigurieren (**wobei ein richtiger Zähler absolut zu empfehlen ist**):
```
meters: 
- name: charge
  type: custom
  power:
    source: http
    uri: http://localhost/api/state
    jq: if .result.loadpoints[0].charging==true then .result.pvPower + .result.grid.power else 0 end
```
Dieses meter dann dem Loadpoint zuweisen `meter: charge`

Bei hohem Hausverbrauch kann die Leistung dann aber auch höher sein, als freigegeben.

naltatis, 2025-03-05:
@RacerEdge ich glaube dein Thema ist nicht das gleich, wie das von @pocki80

RacerEdge, 2025-03-05:
Stellt sich aber gleich dar.🥹
Wenn ich wüsste was das Problem ist, hätte ich es schon zur Lösung eingebracht.😀

VolkerK62, 2025-03-05:
>Die Leistung der Wallbox wird ja korrekt angezeigt.

Hier geht es darum, dass die Wallbox gar keine Leistung übermittlet.
Deshalb, völlig anderes Thema, auch wenn die Auswirkungen optisch sehr ähnlich sind.

>Und im August lief es noch perfekt

Im Oktober gab es Änderungen bei den Huawei Templates.
Evtl liegt da dein Problem. Mach doch dafür eine Diskussion auf.

andig, 2025-03-05:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

pocki80, 2025-03-07:
Ich habe das nochmals getestet und stelle folgendes fest:

Die Wallbox beginnt ohne custom-Meter ordnungsgemäß zu laden. 

Mit custom-Meter lädt sie auch, doch bleibt im
Evcc der Status "ladebereit", obwohl der custom-Meter eine Leistung ausspuckt. 

Was könnte evcc daran hindern, `charging` von false auf true zu ändern?

mfuchs1984, 2025-03-08:
Das liegt daran, dass in der eebus Implementierung bei einem externen meter dieses als Quelle für den Status herangezogen wird. Da dein Meter aber 0 liefert, wenn der Status mit "charging" bzw. StatusC ist, hast du damit einen deadlock gebaut.
https://github.com/evcc-io/evcc/blob/ac0963263ce6bd4045ba288183e7c3343b68f2ce/charger/eebus.go#L153

pocki80, 2025-03-08:
Verstehe, somit muss mein custom-Meter nicht nur `power` liefern, sondern auch `current` (am Besten 3-phasig)?
Werde ich mal ausprobieren, das ist machbar.

mfuchs1984, 2025-03-08:
Stehe gerade auf dem Schlauch, was würden die currents ändern?

pocki80, 2025-03-08:
Laut quellcode entscheidet chargeCurrent > minCurrent * 0.6 über  `charging = true/false`
Wenn mein custom-meter nur `chargePower` liefert, wird daraus aber kein `chargeCurrent` ermittelt, das bleibt also 0 und `charging` somit immer `false`.
Ich habe in meinem custom-Meter nun 3 `currents` mit definiert (errechnen sich aus dem `chargePower`) und siehe da: nun funktioniert es sauber :-)