evcc-io/evcc #19452: IDM Wärmepumpe Navigator 1.7

florianfiby, 2025-03-05:
### Describe the bug

Das IDM Template funktioniert mit Navigator 1.7 nicht, der Überschuss wird nicht korrekt übertragen:

[image]

[image]

Folgende Lösung in Homeassistant funktioniert. Eventuell gibt es unterschiede zu Navigator 2.0:

[image]

hub: idm_wp
unit: 1
address: 74
value: >-
  [ {{ unpack(pack(states('sensor.solarnet_leistung_vom_netz')|float / -1000,
  ">f"), ">H", offset=2) }},  {{ 
  unpack(pack(states('sensor.solarnet_leistung_vom_netz')|float / -1000, ">f"),
  ">H") }} ]

[image]

Die Adresse 74 ist in Navigator 1.7 nicht dokumentiert, funktioniert aber. 

[image]

### Steps to reproduce

1. IDM Template mit IDM Nav 1.7 per ModBus verbinden

### Configuration details

```yaml
- name: idm
    type: template
    template: idm
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 10.7.0.3 # Hostname
    port: 502 # Port
```

### Log details

```text
.
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

andig, 2025-03-05:
Könntest du mal bitte ein Modbus trace machen? Sieht aus wie Faktor 1Mio zuviel?

florianfiby, 2025-03-05:
Werde ich nachliefern sobald ich untertags wieder Zeit habe.

Ich bilde mir ein bei meiner Home Assistant Lösung auch über diesen Fehler gestolpert zu sein.
Die Home Assistant Automatisierung habe ich nach einigen Try & Errors zum Laufen gebracht.

[image]

andig, 2025-03-06:
Im Template sieht es richtig aus: `float32s`. Deshalb müssen wir schauen was da in Reg. 74 wirklich geschrieben wird. Da hilft nur ein Log.

florianfiby, 2025-03-06:
Werde versuchen es nächste Woche zu erfassen. Im Moment läuft wieder die HomeAssistant Lösung

mathiasmora, 2025-03-07:
@andig: Ich hab mit dem Navigator 2.0 ein ähnliches Thema. Ist das ggf. das gleiche Verhalten? Logs habe ich angehängt #19002

andig, 2025-03-07:
Bisher dachte ich, es wird falsch geschrieben. Laut #19002 ist das aber korrekt!

@florianfiby bitte aktuelle Version prüfen.
@mathiasmora bitte neue Issue. Bei Dir ist die Frage wo die 11kW Überschuss her kommen die geschrieben werden.

closed 2025-03-07 by andig

mathiasmora, 2025-03-11:
@andig: Ich hab das Problem mit dem PV-Überschuss jetzt genauer angeschaut. Ich denke da stimmt die Berechnung vom PV-Überschuss für die WP nicht. Ich habe das Ticket erstellt und genauer dokumentiert: #19669