evcc-io/evcc #19459: loadpoint overloaded despite set maxacpower

farcorben, 2025-03-05:
### Describe the bug

Hallo Jungs,

jetzt wo die Tage wieder länger werden und das PV output den MaxACPower eines Wechselrichters sprengt, habe ich folgende Beobachtung gemacht. Nach meinem Verständnis sollte der `maxacpower` Parameter genau solch ein Problem verhindern.. oder verstehe ich das Feature grundlegend falsch?

Für mich stellt es sich so dar als würde er die  `maxacpower` zwar berücksichtigen im log, denn immerhin berechnet er den `pv 1 excess DC: 851W` richtig ( pv power 6851 - MaxACPower 6000 = 851W).. aber er limitiert den `charge power` nicht auf die "MaxACPower 6000". 
Am ende führt dies dazu das 1024 Watt aus dem Netz zusätzlich hinzugezogen werden die eigentlich nicht nötig sein sollten.
Meiner Meinung nach wird ab dem Moment, wo "PV input > maxacpower" ist auch die `site power` fehlerhaft berechnet da ich negative Werte hier eigentlich nicht erwarte ^^
 
**Hinweis**: Mein Inverter kann 6600Watt leisten, die er hier auch ausnutzt. Ich habe Ihn aber aus testzwecken versucht auf 6000 mittels MaxACPower zu regulieren. Auch das einstellen auf `maxacpower: 6600` führt bei mir zu exakt dem gleichen Ergebnis. 

[image]

Vielen lieben Dank

### Steps to reproduce

1. PV input Leistung muss Wechselrichter AC Max Leistung übersteigen

### Configuration details

```yaml
site:
  title: ***
  residualPower: 100
  meters:
    grid: grid1
    pv:
    - pv
    battery:
    - battery

meters:
- type: template
  template: solax
  id: 1
  host: 192.168.1.27
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solax
  id: 1
  host: 192.168.1.27
  port: 502
  usage: pv
  modbus: tcpip
  name: pv
  maxacpower: 6000 # Watt
- type: template
  template: solax
  id: 1
  host: 192.168.1.27
  port: 502
  usage: battery
  modbus: tcpip
  name: battery
  capacity: "12"
  defaultmode: 5 # 0 self use # 5 # TOU

Version: `0.200.5`
```

### Log details

```text
[main  ] INFO 2025/03/05 12:41:19 evcc 0.200.5
[main  ] INFO 2025/03/05 12:41:19 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/05 12:41:19 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2025/03/05 12:41:24 listening at :7070
[main  ] ERROR 2025/03/05 12:41:58 creating vehicle db:13 failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
[lp-1  ] DEBUG 2025/03/05 12:41:58 set title: Solax EVC
[lp-1  ] DEBUG 2025/03/05 12:41:58 set priority: 0
[lp-1  ] DEBUG 2025/03/05 12:41:58 set smart cost limit: 0.0
[lp-1  ] DEBUG 2025/03/05 12:41:58 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:1100}}
[lp-1  ] DEBUG 2025/03/05 12:41:58 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2025/03/05 12:41:58 set session energy limit: 0
[lp-1  ] DEBUG 2025/03/05 12:41:58 set session soc limit: 0
[lp-1  ] DEBUG 2025/03/05 12:41:58 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0x638363c}
[lp-1  ] DEBUG 2025/03/05 12:41:58 set default mode: pv
[lp-1  ] DEBUG 2025/03/05 12:41:58 set phases: 3
[lp-1  ] DEBUG 2025/03/05 12:41:58 set max current: 9
[gsi   ] ERROR 2025/03/05 12:42:04 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/03/05 12:42:04 creating tariff co2 failed: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 429 (Too Many Requests)
[lp-1  ] DEBUG 2025/03/05 12:42:04 charge total import: 7836.200kWh
[site  ] WARN 2025/03/05 12:42:04 vehicle 'Kia EV6' does not support automatic detection
[site  ] WARN 2025/03/05 12:42:04 vehicle 'LeapMotor T03' does not support automatic detection
[site  ] INFO 2025/03/05 12:42:04 site config:
[site  ] INFO 2025/03/05 12:42:04   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/03/05 12:42:04     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/03/05 12:42:04     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/03/05 12:42:04     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2025/03/05 12:42:04   vehicles:
[site  ] INFO 2025/03/05 12:42:04     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/03/05 12:42:04     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/03/05 12:42:04   tariffs:
[site  ] INFO 2025/03/05 12:42:04     grid:      priceforecast
[site  ] INFO 2025/03/05 12:42:04     feed-in:   ✗
[site  ] INFO 2025/03/05 12:42:04     co2:       TariffType(0)
[site  ] INFO 2025/03/05 12:42:04     solar:     solar
[lp-1  ] INFO 2025/03/05 12:42:04 loadpoint 1:
[lp-1  ] INFO 2025/03/05 12:42:04   mode:        pv
[lp-1  ] INFO 2025/03/05 12:42:04   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/03/05 12:42:04   meters:      charge ✓
[lp-1  ] INFO 2025/03/05 12:42:04     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2025/03/05 12:42:04 set buffer soc: 90
[site  ] DEBUG 2025/03/05 12:42:04 set buffer start soc: 0
[site  ] DEBUG 2025/03/05 12:42:04 set priority soc: 65
[site  ] DEBUG 2025/03/05 12:42:04 set battery discharge control: true
[site  ] DEBUG 2025/03/05 12:42:04 set residual power: 105
[lp-1  ] DEBUG 2025/03/05 12:42:04 set smart cost limit: 0.0
[lp-1  ] DEBUG 2025/03/05 12:42:04 phase timer inactive
[lp-1  ] DEBUG 2025/03/05 12:42:04 pv timer inactive
[lp-1  ] DEBUG 2025/03/05 12:42:04 max charge current: 6A
[site  ] WARN 2025/03/05 12:42:04 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2025/03/05 12:42:04 ----

...

[site  ] DEBUG 2025/03/05 12:49:26 ----
[lp-1  ] DEBUG 2025/03/05 12:49:26 charge power: 7074W
[lp-1  ] DEBUG 2025/03/05 12:49:26 charge currents: [9.78 9.73 9.62]A
[site  ] DEBUG 2025/03/05 12:49:27 grid power: 969W
[site  ] DEBUG 2025/03/05 12:49:27 battery 1 power: -253W
[site  ] DEBUG 2025/03/05 12:49:27 battery 1 soc: 77%
[site  ] DEBUG 2025/03/05 12:49:27 pv 1 power: 6853W
[site  ] DEBUG 2025/03/05 12:49:27 pv 1 excess DC: 853W
[site  ] DEBUG 2025/03/05 12:49:27 site power: -32W
[lp-1  ] DEBUG 2025/03/05 12:49:27 charge voltages: [239 243 245]V
[lp-1  ] DEBUG 2025/03/05 12:49:27 detected connected phases: 3p
[lp-1  ] DEBUG 2025/03/05 12:49:27 detected active phases: 3p
[lp-1  ] DEBUG 2025/03/05 12:49:27 charge total import: 7837.000kWh
[lp-1  ] DEBUG 2025/03/05 12:49:28 charger status: C
[lp-1  ] DEBUG 2025/03/05 12:49:28 pv charge current: 9.83A = 9.79A + 0.0464A (-32W @ 3p)
[lp-1  ] DEBUG 2025/03/05 12:49:28 max charge current: 9.83A
[site  ] DEBUG 2025/03/05 12:49:46 ----
[lp-1  ] DEBUG 2025/03/05 12:49:46 charge power: 7078W
[lp-1  ] DEBUG 2025/03/05 12:49:46 charge currents: [9.81 9.73 9.64]A
[site  ] DEBUG 2025/03/05 12:49:47 grid power: 989W
[site  ] DEBUG 2025/03/05 12:49:47 battery 1 power: -257W
[site  ] DEBUG 2025/03/05 12:49:47 battery 1 soc: 77%
[site  ] DEBUG 2025/03/05 12:49:48 pv 1 power: 6854W
[site  ] DEBUG 2025/03/05 12:49:48 pv 1 excess DC: 854W
[site  ] DEBUG 2025/03/05 12:49:48 site power: -17W
[lp-1  ] DEBUG 2025/03/05 12:49:48 charge voltages: [239 243 245]V
[lp-1  ] DEBUG 2025/03/05 12:49:48 detected connected phases: 3p
[lp-1  ] DEBUG 2025/03/05 12:49:48 detected active phases: 3p
[lp-1  ] DEBUG 2025/03/05 12:49:48 charge total import: 7837.100kWh
[lp-1  ] DEBUG 2025/03/05 12:49:48 charger status: C
[lp-1  ] DEBUG 2025/03/05 12:49:48 pv charge current: 9.86A = 9.83A + 0.0246A (-17W @ 3p)
[lp-1  ] DEBUG 2025/03/05 12:49:48 max charge current: 9.86A

[site  ] DEBUG 2025/03/05 12:50:06 ----
[lp-1  ] DEBUG 2025/03/05 12:50:06 charge power: 7127W
[lp-1  ] DEBUG 2025/03/05 12:50:06 charge currents: [9.88 9.82 9.7]A
[site  ] DEBUG 2025/03/05 12:50:07 grid power: 1024W
[site  ] DEBUG 2025/03/05 12:50:07 battery 1 power: -252W
[site  ] DEBUG 2025/03/05 12:50:07 battery 1 soc: 77%
[site  ] DEBUG 2025/03/05 12:50:07 pv 1 power: 6851W
[site  ] DEBUG 2025/03/05 12:50:07 pv 1 excess DC: 851W
[site  ] DEBUG 2025/03/05 12:50:07 site power: 26W
[lp-1  ] DEBUG 2025/03/05 12:50:08 charge voltages: [238 243 245]V
[lp-1  ] DEBUG 2025/03/05 12:50:08 detected connected phases: 3p
[lp-1  ] DEBUG 2025/03/05 12:50:08 detected active phases: 3p
[lp-1  ] DEBUG 2025/03/05 12:50:08 charge total import: 7837.100kWh
[lp-1  ] DEBUG 2025/03/05 12:50:08 charger status: C
[lp-1  ] DEBUG 2025/03/05 12:50:08 pv charge current: 9.82A = 9.86A + -0.0377A (26W @ 3p)
[lp-1  ] DEBUG 2025/03/05 12:50:08 max charge current: 9.82A
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

farcorben, 2025-03-05:
Zuerst noch eine Entschuldigung, ich habe "Nightly build" abgehakt aber nicht getestet da mir die Bedingung mitten am Tag als ich die Daten gesammelt habe nicht bewusst war. Ich habe aber die neusten commits gelesen und außer "eventuell" #19383 keine Überschneidungen gefunden.

mfuchs1984, 2025-03-05:
https://github.com/evcc-io/evcc/pull/19383 behebt das Problem.

farcorben, 2025-03-05:
jetzt fühle ich mich ziemlich dumm ... ^^ 
Aber danke für die extrem schnelle Antwort..

mfuchs1984, 2025-03-05:
Dann kannst du das issue ja schließen :)

farcorben, 2025-03-05:
natürlich gern, und morgen mit Freude das Nigthly testen. 
Danke

Geri-G93, 2025-03-15:
moin
ich grab das nochmal aus
nutze 200.9
mit einem Huawei 10Ktl, der macht 10kw (max 10.5kw) AC, hat aber 15kwP DC dran hängen (süd+ west verteilung)
zu mittag wenn die Luna lädt kommt der auch auf DC 15kw, heut Mittag hat unser Auto durchweg mit 10.5kw geladen, also Netzbezug 

log würde ich morgen nachreichen

mfuchs1984, 2025-03-15:
@Geri-G93 wie hast du dein pv meter in evcc denn konfiguriert?