evcc-io/evcc #19461: Volvo Legacy not receiving information: error 403

OebeleStoker, 2025-03-05:
### Describe the bug

The Volvo Legacy module has worked for a couple of weeks, however all of a sudden it gives an error. I have tried re-installing via both UI config and YAML without success. 

### Steps to reproduce

1. Create entity of vehicle via UI config or add entity to YAML file
2.  Validate entity
...

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: volvo
    user: ...
    password: ...
    title: Volvo
    capacity: 9.6
```

### Log details

```text
[main ] ERROR 2025/03/05 21:13:01 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'volvo': cannot get vehicles: unexpected status: 403 (Forbidden)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.5

andig, 2025-03-06:
Please add trace log for evcc vehicle

OebeleStoker, 2025-03-06:
[volvo ] TRACE 2025/03/06 10:21:04 GET https://vocapi.wirelesscar.net/customerapi/rest/v3.0/customeraccounts
[volvo ] TRACE 2025/03/06 10:21:04 <html> <head><title>403 Forbidden</title></head> <body> <center><h1>403 Forbidden</h1></center> </body> </html>

andig, 2025-03-06:
Your user is not allowed.

OebeleStoker, 2025-03-06:
What do you mean? I has worked fine before

andig, 2025-03-06:
You'll have to ask Volvo- it's their server...

closed 2025-03-06 by andig

OebeleStoker, 2025-03-06:
How does evcc pull info from Volvo’s server? I am not that technical...

bitflower, 2025-03-09:
I have the same issue. Worked for over y ear now.

My Logs:
```
[main ] ERROR 2025/03/09 13:03:47 creating vehicle ev2 failed: cannot create vehicle type 'template': cannot create vehicle type 'volvo': cannot get vehicles: unexpected status: 403 (Forbidden)
```