evcc-io/evcc #19462: Cupra Born offline

Oeffly, 2025-03-06:
### Describe the bug

Soc Wert wird nicht mehr abgerufen auf der OpenWb

siehe auch Bug #18979 

### Steps to reproduce

leider wird der Soc Wert unserem Curp Born nicht mehr abgerufen....
Weitere Fahrzeuge gehen (Audi Q4, Vw ID.4)

### Configuration details

```yaml
EVCC-Soc Modul auf der OpenWB

Marke:Seat-Cupra
```

### Log details

```text
2025-03-06 06:00:10,880 - {modules.common.configurable_vehicle:56} - {DEBUG:fetch soc_ev6} - Vehicle Instance <class 'modules.vehicles.evcc.config.EVCCVehicleSoc'>
2025-03-06 06:00:10,881 - {modules.common.configurable_vehicle:57} - {DEBUG:fetch soc_ev6} - Calculated SoC-State CalculatedSocState(imported_start=8347621.09, manual_soc=None, soc_start=85.0)
2025-03-06 06:00:10,882 - {modules.common.configurable_vehicle:58} - {DEBUG:fetch soc_ev6} - Vehicle Update Data VehicleUpdateData(plug_state=False, charge_state=False, imported=None, battery_capacity=60000, efficiency=90, soc_from_cp=None, timestamp_soc_from_cp=None)
2025-03-06 06:00:10,883 - {modules.common.configurable_vehicle:59} - {DEBUG:fetch soc_ev6} - General Config GeneralVehicleConfig(use_soc_from_cp=False, request_interval_charging=300, request_interval_not_charging=3600, request_only_plugged=False)
2025-03-06 06:00:10,884 - {modules.common.component_context:26} - {DEBUG:fetch soc_ev6} - Update Komponente ['EVCC']
2025-03-06 06:00:10,885 - {modules.vehicles.evcc.api:66} - {DEBUG:fetch soc_ev6} - Fetching EVCC SOC
2025-03-06 06:00:10,889 - {modules.vehicles.evcc.api:79} - {DEBUG:fetch soc_ev6} - Vehicle id found in config: 20
2025-03-06 06:00:10,891 - {modules.vehicles.evcc.api:81} - {DEBUG:fetch soc_ev6} - Fetching SoC for vehicle id: 20
2025-03-06 06:00:10,999 - {modules.common.fault_state:49} - {ERROR:fetch soc_ev6} - EVCC: FaultState FaultStateLevel.ERROR, FaultStr <class 'grpc.RpcError'> (<_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "cannot create vehicle type 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"
	debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"cannot create vehicle type \'cupra\': cannot get vehicles: Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)", grpc_status:4, created_time:"2025-03-06T06:00:10.991633722+01:00"}"
>,), Traceback: 
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 86, in fetch_soc
    response = stub.SoC(
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1160, in __call__
    return _end_unary_response_blocking(state, call, False, None)
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1003, in _end_unary_response_blocking
    raise _InactiveRpcError(state)  # pytype: disable=not-instantiable
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "cannot create vehicle type 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"
	debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"cannot create vehicle type \'cupra\': cannot get vehicles: Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)", grpc_status:4, created_time:"2025-03-06T06:00:10.991633722+01:00"}"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 66, in update
    car_state = self._get_carstate_by_source(vehicle_update_data, source)
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 110, in _get_carstate_by_source
    return self.__component_updater(vehicle_update_data)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/soc.py", line 16, in updater
    return fetch_soc(vehicle_config.configuration, vehicle_update_data, vehicle)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 105, in fetch_soc
    raise grpc.RpcError(rpc_error)
grpc.RpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "cannot create vehicle type 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"
	debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"cannot create vehicle type \'cupra\': cannot get vehicles: Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)", grpc_status:4, created_time:"2025-03-06T06:00:10.991633722+01:00"}"
>
```

### What type of operating system or environment does evcc run on?

other

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

OpenWB Version 2.1.6-Patch.1

Oeffly, 2025-03-06:
Danke für die schnelle reaktion.... Wie meinst du nochmals versuchen?

2025-03-06 08:54:23,053 - {modules.common.configurable_vehicle:56} - {DEBUG:fetch soc_ev6} - Vehicle Instance <class 'modules.vehicles.evcc.config.EVCCVehicleSoc'>
2025-03-06 08:54:23,062 - {modules.common.configurable_vehicle:57} - {DEBUG:fetch soc_ev6} - Calculated SoC-State CalculatedSocState(imported_start=8347621.09, manual_soc=None, soc_start=85.0)
2025-03-06 08:54:23,063 - {modules.common.configurable_vehicle:58} - {DEBUG:fetch soc_ev6} - Vehicle Update Data VehicleUpdateData(plug_state=False, charge_state=False, imported=None, battery_capacity=60000, efficiency=90, soc_from_cp=None, timestamp_soc_from_cp=None)
2025-03-06 08:54:23,065 - {modules.common.configurable_vehicle:59} - {DEBUG:fetch soc_ev6} - General Config GeneralVehicleConfig(use_soc_from_cp=False, request_interval_charging=300, request_interval_not_charging=3600, request_only_plugged=False)
2025-03-06 08:54:23,068 - {modules.common.component_context:26} - {DEBUG:fetch soc_ev6} - Update Komponente ['EVCC']
2025-03-06 08:54:23,071 - {modules.vehicles.evcc.api:66} - {DEBUG:fetch soc_ev6} - Fetching EVCC SOC
2025-03-06 08:54:23,082 - {modules.vehicles.evcc.api:79} - {DEBUG:fetch soc_ev6} - Vehicle id found in config: 20
2025-03-06 08:54:23,093 - {modules.vehicles.evcc.api:81} - {DEBUG:fetch soc_ev6} - Fetching SoC for vehicle id: 20
2025-03-06 08:54:23,214 - {modules.common.fault_state:49} - {ERROR:fetch soc_ev6} - EVCC: FaultState FaultStateLevel.ERROR, FaultStr <class 'grpc.RpcError'> (<_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "cannot create vehicle type 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-03-06T08:54:23.204186958+01:00", grpc_status:4, grpc_message:"cannot create vehicle type \'cupra\': cannot get vehicles: Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}"
>,), Traceback: 
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 86, in fetch_soc
    response = stub.SoC(
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1160, in __call__
    return _end_unary_response_blocking(state, call, False, None)
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1003, in _end_unary_response_blocking
    raise _InactiveRpcError(state)  # pytype: disable=not-instantiable
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "cannot create vehicle type 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-03-06T08:54:23.204186958+01:00", grpc_status:4, grpc_message:"cannot create vehicle type \'cupra\': cannot get vehicles: Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 66, in update
    car_state = self._get_carstate_by_source(vehicle_update_data, source)
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 110, in _get_carstate_by_source
    return self.__component_updater(vehicle_update_data)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/soc.py", line 16, in updater
    return fetch_soc(vehicle_config.configuration, vehicle_update_data, vehicle)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 105, in fetch_soc
    raise grpc.RpcError(rpc_error)
grpc.RpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "cannot create vehicle type 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-03-06T08:54:23.204186958+01:00", grpc_status:4, grpc_message:"cannot create vehicle type \'cupra\': cannot get vehicles: Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/##-car-id-##/garage/vehicles\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}"
>

andig, 2025-03-06:
Seat antwortet nicht. Da kann evcc leider nix tun. Einfach mal abwarten, zur Not ein paar Tage :(

andig, 2025-03-06:
Mit den Token hat das nix zu tun. Ich kann die leider nur sagen, dass der Seat Server nicht antwortet. Grund unklar.

Oeffly, 2025-03-06:
okay.... das ganze seit gestern mittag.

Oeffly, 2025-03-07:
leider bis jetzt nichts neues .... habe seiit dem 5.3. keinen SOC mehr bekommen für denn Cupra. Habt ihr noch eine Idee?

Habe mal das SOC-Modul geweselt und alles neu eingetragen... nun hab ich ein anders Fehlerbild.

`2025-03-07 05:45:08,227 - {modules.common.configurable_vehicle:56} - {DEBUG:fetch soc_ev6} - Vehicle Instance <class 'modules.vehicles.evcc.config.EVCCVehicleSoc'>
2025-03-07 05:45:08,230 - {modules.common.configurable_vehicle:57} - {DEBUG:fetch soc_ev6} - Calculated SoC-State CalculatedSocState(imported_start=8347621.09, manual_soc=None, soc_start=85.0)
2025-03-07 05:45:08,231 - {modules.common.configurable_vehicle:58} - {DEBUG:fetch soc_ev6} - Vehicle Update Data VehicleUpdateData(plug_state=False, charge_state=False, imported=None, battery_capacity=60000, efficiency=90, soc_from_cp=None, timestamp_soc_from_cp=None)
2025-03-07 05:45:08,232 - {modules.common.configurable_vehicle:59} - {DEBUG:fetch soc_ev6} - General Config GeneralVehicleConfig(use_soc_from_cp=False, request_interval_charging=300, request_interval_not_charging=3600, request_only_plugged=False)
2025-03-07 05:45:08,235 - {modules.common.component_context:26} - {DEBUG:fetch soc_ev6} - Update Komponente ['EVCC']
2025-03-07 05:45:08,236 - {modules.vehicles.evcc.api:66} - {DEBUG:fetch soc_ev6} - Fetching EVCC SOC
2025-03-07 05:45:08,240 - {modules.vehicles.evcc.api:79} - {DEBUG:fetch soc_ev6} - Vehicle id found in config: 20
2025-03-07 05:45:08,241 - {modules.vehicles.evcc.api:81} - {DEBUG:fetch soc_ev6} - Fetching SoC for vehicle id: 20
2025-03-07 05:45:08,333 - {modules.common.fault_state:49} - {ERROR:fetch soc_ev6} - EVCC: FaultState FaultStateLevel.ERROR, FaultStr <class 'grpc.RpcError'> (<_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "cannot create vehicle type 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/c40c13b5-10a0-450f-95b0-d23f0393fbac/garage/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-03-07T05:45:08.328742598+01:00", grpc_status:4, grpc_message:"cannot create vehicle type \'cupra\': cannot get vehicles: Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/c40c13b5-10a0-450f-95b0-d23f0393fbac/garage/vehicles\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}"
>,), Traceback: ##-car-id-##
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 86, in fetch_soc
    response = stub.SoC(
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1160, in __call__
    return _end_unary_response_blocking(state, call, False, None)
  File "/home/openwb/.local/lib/python3.9/site-packages/grpc/_channel.py", line 1003, in _end_unary_response_blocking
    raise _InactiveRpcError(state)  # pytype: disable=not-instantiable
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "cannot create vehicle type 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/c40c13b5-10a0-450f-95b0-d23f0393fbac/garage/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-03-07T05:45:08.328742598+01:00", grpc_status:4, grpc_message:"cannot create vehicle type \'cupra\': cannot get vehicles: Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/c40c13b5-10a0-450f-95b0-d23f0393fbac/garage/vehicles\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}"
>

During handling of the above exception, another exception occurred:
##-car-id-##
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 66, in update
    car_state = self._get_carstate_by_source(vehicle_update_data, source)
  File "/var/www/html/openWB/packages/modules/common/configurable_vehicle.py", line 110, in _get_carstate_by_source
    return self.__component_updater(vehicle_update_data)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/soc.py", line 16, in updater
    return fetch_soc(vehicle_config.configuration, vehicle_update_data, vehicle)
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 105, in fetch_soc
    raise grpc.RpcError(rpc_error)
grpc.RpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "cannot create vehicle type 'cupra': cannot get vehicles: Get "https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/c40c13b5-10a0-450f-95b0-d23f0393fbac/garage/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-03-07T05:45:08.328742598+01:00", grpc_status:4, grpc_message:"cannot create vehicle type \'cupra\': cannot get vehicles: Get \"https://ola.prod.code.seat.cloud.vwgroup.com/v2/users/c40c13b5-10a0-450f-95b0-d23f0393fbac/garage/vehicles\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}"
>

#EVCC Modul gelöscht ... etwas gewartet und wieder eingerichtet..... #

2025-03-07 06:39:07,342 - {modules.common.configurable_vehicle:56} - {DEBUG:fetch soc_ev6} - Vehicle Instance <class 'modules.vehicles.evcc.config.EVCCVehicleSoc'>
[... cut ...]
  File "/var/www/html/openWB/packages/modules/vehicles/evcc/api.py", line 105, in fetch_soc
    raise grpc.RpcError(rpc_error)
grpc.RpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNKNOWN
	details = "cannot create vehicle type 'cupra': invalid character 'c' looking for beginning of object key string"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2025-03-07T06:39:07.467150664+01:00", grpc_status:2, grpc_message:"cannot create vehicle type \'cupra\': invalid character \'c\' looking for beginning of object key string"}"
>

`

andig, 2025-03-07:
Der Service ist für die letzten VW API-Änderungen aktualisiert.