evcc-io/evcc #19464: SHRDZM Smartmeter Kundenschnittstellen Modul

Grufti68, 2025-03-06:
Hallo,

wird der Smartmeter SHRDZM auch in irgendeiner weise Unterstütz?

Laut Datenblatt kann der Smartmeter MBUS und ist auch kompatibel mit Volkszaehler.org

Hier mal der Link:
[https://cms.shrdzm.com/produkt/smartmeter-modul/](url)

Oder ist es Möglich die Smartmeter Daten welche bereits über MQTT an IO Broker gesendet werden zu verwenden?

LG

andig, 2025-03-06:
Gibts ne API-Doku?

Grufti68, 2025-03-06:
kann ich dir leider nicht sagen da ich in diesem Thema "API" nicht so die Ahnung habe

andig, 2025-03-06:
Ohne können wir leider nichts tun

closed 2025-03-06 by andig

Grufti68, 2025-03-06:
Das heißt ich kann nur über Volkszaehler.org das ganze bewerkstelligen?

andig, 2025-09-25:
Per Template wären nur REST und MQTT möglich. MQTT erfordert immer eine weitere Komponente. REST mit cache Headern ist auch recht effizient.

premultiply, 2025-09-25:
> detailliertere Informationen zur gewünschten Schnittstelle bereitstellen

Ja, das wäre in dem Fall für REST-Abruf obligatorisch.

Idealerweise die Implementierung selbst als PR beitragen.

saghonfly, 2025-09-25:
> Per Template wären nur REST und MQTT möglich. MQTT erfordert immer eine weitere Komponente. REST mit cache Headern ist auch recht effizient.

...wo findet man das Template? 

BTW...kann jemand den Task in dem Fall wiedereröffnen...