evcc-io/evcc #19481: Hysterese für Heizstäbe (Beispiel MY-PV Elwa 2)

hbpv, 2025-03-06:
Problem:

Ist in evcc limitSoc Beispielsweise auf 60°C gestellt, wird schon bei 59,9° begonnen zu Laden, was über den im Zweifel zu zahllosen Taktungen des Heizstabes führt. Im Grunde haben wir mit evcc nur ein klassisches Hysterese-PLUS (LimitSoc), aber wie üblich bei Warmwasseraufbereitung, kein Hysterese-MINUS. My-PV hat das auf seiner Weboberfläche, dort sind z.B., was auch eine Fallback-Sicherheit nach oben ist, -3°/+3° eingestellt. Somit stoppt der Heizstab, egal was evcc übermittelt - alles gut. Aber die -3° kommen nie zum Zuge.

Lösung:

Für Heizstab-Templates, vllt. gleich mit MY-PV testen, einen limit-hysterese-soc-min und limit-hysterese-soc-max einführen. Im Falle eines Heizstabes dann nur "Laden", wenn hysterese-min unterschritten ist. Alternativ zum limitSoc einen limitSocMin einführen.

Wenn man es ganzheitlich machen will und unabhängig vom Setting von MY-PV arbeiten will, sollte man beide values einführen. Das wäre auch kein problem, da der max Wer (aktuell limitSoc) sowieso durch den Heizstab/Thor überschrieben wird, sollte das bei MY-PV festgelegte absolute Templimit erreicht sein:

limitHystereseSocMin
limitSoc
limitHystereseSocMax

Kontext:

Wenn man mit negativer Hysterese arbeitet, schon man den Heizstab vor zu hoher Taktung.

Beispiel:

limitHystereseSocMin = 57
limitSoc = 60
limitHystereseSocMax = 63

Beginne ansteuern bei 56,99, stoppe bei 60.

andig, 2025-03-06:
Ich würde vorschlagen, die Hysterese einfach der Elwa zu überlassen. Dafür muss in evcc die Zieltemperatur höher als die Hysterese gesetzt sein.

hbpv, 2025-03-06:
> Ich würde vorschlagen, die Hysterese einfach der Elwa zu überlassen. Dafür muss in evcc die Zieltemperatur höher als die Hysterese gesetzt sein.

Kann an der Uhrzeit liegen, aber verstehe ich das richtig?:

a) Elwa Setting "Temperatur max. - Solarbetrieb" = 60°
b) Elwa Setting Hysterese+ = 3°
c) Elwa Setting Hysterese- = 3°
d) evcc SOC = 65°

1. Temperatur des Wassers geht auf unter 65°: evcc will laden, Elwa nimmt den Befehl nicht an weil a)
2. Temparatur des Wassers geht auf unter 57°: evcc will laden (immernoch), Elwa nimmt den Befehl an, weil c), wird aber im 60 Sek. Fenster (Zeitablauf Ansteuerung) den Befehl wieder schließen, was ggf. nichtmal für 58° reicht...
3. Das Ganze beginnt von vorn...
4. evcc befindet sich damit permanent in WILL LADEN oder LADE Modus :-/

Oder? Also wäre das doch auch nicht wirklich die Lösung.

andig, 2025-03-06:
> wird aber im 60 Sek. Fenster (Zeitablauf Ansteuerung) den Befehl wieder schließen, was ggf. nichtmal für 58° reicht...

Wieso? Im Zweifel: einfach mal ausprobieren!

hbpv, 2025-03-06:
Es führt meiner Meinung nach zu:

charger logic error: disabled but charging

Evcc disabled nicht, da SOC quasi ja dann nie erreicht wird. Aktuelle Einstellung:

Max Soll MYPV: 61
Hysterese +-3
SOC evcc 65 (lp3)

Er bleibt in der UI auf heizen stehen.

`[site ] DEBUG 2025/03/06 16:18:01 ----
[lp-1 ] DEBUG 2025/03/06 16:18:01 charge power: 0W
[lp-2 ] DEBUG 2025/03/06 16:18:01 charge power: 0W
[lp-3 ] DEBUG 2025/03/06 16:18:02 charge power: 0W
[site ] DEBUG 2025/03/06 16:18:02 pv 1 power: 3800W
[site ] DEBUG 2025/03/06 16:18:02 grid power: -3410W
[site ] DEBUG 2025/03/06 16:18:02 battery 1 power: -10W
[site ] DEBUG 2025/03/06 16:18:04 battery 1 soc: 100%
[site ] DEBUG 2025/03/06 16:18:04 grid powers: [-1040 -1060 -1300]W
[site ] DEBUG 2025/03/06 16:18:05 grid currents: [-4.44 -4.44 -5.48]A
[site ] DEBUG 2025/03/06 16:18:05 site power: -3320W
[lp-3 ] DEBUG 2025/03/06 16:18:05 charger status: C
[lp-3 ] WARN 2025/03/06 16:18:05 charger logic error: disabled but charging
[lp-3 ] DEBUG 2025/03/06 16:18:05 pv timer elapse
[lp-3 ] DEBUG 2025/03/06 16:18:05 pv timer inactive
[lp-3 ] DEBUG 2025/03/06 16:18:05 pv charge current: 26.4A = 12A + 14.4A (-3320W @ 1p)
[lp-3 ] DEBUG 2025/03/06 16:18:05 pv timer reset
[lp-3 ] DEBUG 2025/03/06 16:18:05 pv timer inactive
[site ] DEBUG 2025/03/06 16:18:31 ----
[lp-1 ] DEBUG 2025/03/06 16:18:31 charge power: 0W
[lp-2 ] DEBUG 2025/03/06 16:18:31 charge power: 0W
[lp-3 ] DEBUG 2025/03/06 16:18:32 charge power: 0W
[site ] DEBUG 2025/03/06 16:18:32 pv 1 power: 3800W
[site ] DEBUG 2025/03/06 16:18:32 grid power: -3400W
[site ] DEBUG 2025/03/06 16:18:32 battery 1 power: -10W
[site ] DEBUG 2025/03/06 16:18:33 battery 1 soc: 100%
[site ] DEBUG 2025/03/06 16:18:34 grid powers: [-1040 -1060 -1300]W
[site ] DEBUG 2025/03/06 16:18:35 grid currents: [-4.44 -4.45 -5.49]A
[site ] DEBUG 2025/03/06 16:18:35 site power: -3310W
[lp-1 ] DEBUG 2025/03/06 16:18:35 charger status: A
[site ] DEBUG 2025/03/06 16:19:01 ----
[lp-1 ] DEBUG 2025/03/06 16:19:01 charge power: 0W
[lp-2 ] DEBUG 2025/03/06 16:19:01 charge power: 0W
[lp-3 ] DEBUG 2025/03/06 16:19:02 charge power: 0W
[site ] DEBUG 2025/03/06 16:19:02 pv 1 power: 3800W
[site ] DEBUG 2025/03/06 16:19:02 grid power: -3520W
[site ] DEBUG 2025/03/06 16:19:03 battery 1 power: -10W
[site ] DEBUG 2025/03/06 16:19:04 battery 1 soc: 100%
[site ] DEBUG 2025/03/06 16:19:04 grid powers: [-1090 -1060 -1370]W
[site ] DEBUG 2025/03/06 16:19:05 grid currents: [-4.66 -4.44 -5.78]A
[site ] DEBUG 2025/03/06 16:19:06 site power: -3430W
[lp-2 ] DEBUG 2025/03/06 16:19:06 charger status: A `

hbpv, 2025-03-06:
Du hast recht, Problem besteht dennoch ..
Brauchst du noch andere Logs?

Ich meine, die hysterese nach oben seitens MyPV ist ja nur ein Nachlauf - das nutzt uns hier nichts..

Der SOC höher stellen als Hysterese funktioniert nicht..

hbpv, 2025-03-06:
[evcc-20250306-163755-trace.log](https://github.com/user-attachments/files/19111239/evcc-20250306-163755-trace.log)

hbpv, 2025-03-07:
Moin @andig 
Es scheint wohl doch zu laufen. Die Temp ging die Nacht auf <58° Grad zurück (Ziel MYPV 61 mit Hysterese 3) und der Heizstab hat (weil Akkunutzung erlaubt) wieder geladen bis 61°.
Derzeit liegen wir bei 58.5°, es sollte also gleich nochmal losgehen.

Das einzige was bleibt ist der charger-logic error. Siehe anbei nochmal von eben. Wie bekommt man das gelöst? Ein gelbe Ausrufezeichen in der UI ist immer anstrengend, für einen Monk. :-)

[site ] DEBUG 2025/03/07 08:06:31 ----
[ACThor] TRACE 2025/03/07 08:06:31 GET http://192.168.8.201/data.jsn
[ACThor] TRACE 2025/03/07 08:06:32 { "device":"ACTHOR", "acthor9s":1, "fwversion":"a0021400", "psversion":111, "p9sversion":null, "fsetup":0, "screen_mode_flag":3, "power_system":null, "power_act":0, "power_solar_act":0, "power_grid_act":0, "power_ac9":null, "power_solar_ac9":null, "power_grid_ac9":null, "power1_solar":0, "power1_grid":0, "power2_solar":0, "power2_grid":0, "power3_solar":null, "power3_grid":null, "load_state":"1", "load_nom":3256, "rel1_out":"0", "pump_pwm":0, "temp1":591, "temp2":null, "temp3":null, "temp4":null, "boostactive":0, "legboostnext":"7", "date":"07.03.25", "loctime":"08:06:31", "unixtime":1741331191, "uptime":3556, "wp_flag":0, "wp_time1_ctr":0, "wp_time2_ctr":0, "wp_time3_ctr":0, "schicht_flag":0, "act_night_flag":0, "ctrlstate":"Modbus Read received", "blockactive":0, "error_state":0, "meter1_id":null, "meter1_ip":"null", "meter2_id":null, "meter2_ip":"null", "meter3_id":null, "meter3_ip":"null", "meter4_id":null, "meter4_ip":"null", "meter5_id":null, "meter5_ip":"null", "meter6_id":null, "meter6_ip":"null", "meter_ss":null, "meter_ssid":"null", "surplus":null, "m0sum":null, "m0l1":null, "m0l2":null, "m0l3":null, "m0bat":null, "m1sum":null, "m1l1":null, "m1l2":null, "m1l3":null, "m1devstate":null, "m2sum":null, "m2l1":null, "m2l2":null, "m2l3":null, "m2soc":null, "m2state":null, "m2devstate":null, "m3sum":null, "m3l1":null, "m3l2":null, "m3l3":null, "m3soc":null, "m3devstate":null, "m4sum":null, "m4l1":null, "m4l2":null, "m4l3":null, "m4devstate":null, "ecarstate":"null", "ecarboostctr":null, "mss2":"null", "mss3":"null", "mss4":"null", "mss5":"null", "mss6":"null", "mss7":"null", "mss8":"null", "mss9":"null", "mss10":"null", "mss11":"null", "volt_mains":237, "curr_mains":0, "volt_L2":null, "curr_L2":null, "volt_L3":null, "curr_L3":null, "volt_out":0, "freq":49994, "temp_ps":273, "fan_speed":0, "ps_state":0, "9s_state":null, "cur_ip":"192.168.8.201", "cur_sn":"255.255.255.0", "cur_gw":"192.168.8.1", "cur_dns":"8.8.8.8", "fwversionlatest":"a0021800", "psversionlatest":111, "p9sversionlatest":null, "upd_state":1, "upd_files_left":0, "ps_upd_state":0, "p9s_upd_state":null, "cloudstate":4, "debug_ip":"0.0.0.0" }
[lp-3 ] DEBUG 2025/03/07 08:06:32 charge power: 0W
[site ] DEBUG 2025/03/07 08:06:32 pv 1 power: 5000W
[site ] DEBUG 2025/03/07 08:06:32 grid power: 10W
[site ] DEBUG 2025/03/07 08:06:33 battery 1 power: -4900W
[site ] DEBUG 2025/03/07 08:06:33 grid powers: [140 160 -310]W
[site ] DEBUG 2025/03/07 08:06:34 battery 1 soc: 65%
[site ] DEBUG 2025/03/07 08:06:35 grid currents: [0.61 0.68 -1.31]A
[site ] DEBUG 2025/03/07 08:06:35 site power: -4790W
[ac-thor] TRACE 2025/03/07 08:06:35 modbus: send 13 ea 00 00 00 06 01 03 03 eb 00 01
[ac-thor] TRACE 2025/03/07 08:06:35 modbus: recv 13 ea 00 00 00 05 01 03 02 00 00
[lp-3 ] DEBUG 2025/03/07 08:06:35 charger status: B
[ac-thor] TRACE 2025/03/07 08:06:35 modbus: send 13 eb 00 00 00 06 01 03 03 e9 00 01
[ac-thor] TRACE 2025/03/07 08:06:35 modbus: recv 13 eb 00 00 00 05 01 03 02 02 4f
[ac-thor] TRACE 2025/03/07 08:06:35 modbus: send 13 ec 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/03/07 08:06:35 modbus: recv 13 ec 00 00 00 05 01 03 02 00 00
[lp-3 ] WARN 2025/03/07 08:06:35 charger out of sync: expected enabled, got disabled
[lp-3 ] DEBUG 2025/03/07 08:06:35 charger disable
[lp-3 ] DEBUG 2025/03/07 08:06:35 pv charge current: 20.8A = 0A + 20.8A (-4790W @ 1p)
[lp-3 ] DEBUG 2025/03/07 08:06:35 site power -4790W <= 0W enable threshold
[lp-3 ] DEBUG 2025/03/07 08:06:35 pv enable in 0s
[lp-3 ] DEBUG 2025/03/07 08:06:35 pv enable timer elapsed
[ac-thor] TRACE 2025/03/07 08:06:35 modbus: send 13 ed 00 00 00 09 01 10 03 e8 00 01 02 02 b2
[ac-thor] TRACE 2025/03/07 08:06:35 modbus: recv 13 ed 00 00 00 06 01 10 03 e8 00 01
[lp-3 ] DEBUG 2025/03/07 08:06:35 charger enable
[lp-3 ] DEBUG 2025/03/07 08:06:35 wake-up timer: start
[ac-thor] TRACE 2025/03/07 08:06:56 modbus: send 13 ee 00 00 00 06 01 03 03 e8 00 01
[ac-thor] TRACE 2025/03/07 08:06:56 modbus: recv 13 ee 00 00 00 05 01 03 02 00 00
[site ] DEBUG 2025/03/07 08:07:01 ----
[ACThor] TRACE 2025/03/07 08:07:01 GET http://192.168.8.201/data.jsn
[ACThor] TRACE 2025/03/07 08:07:02 { "device":"ACTHOR", "acthor9s":1, "fwversion":"a0021400", "psversion":111, "p9sversion":null, "fsetup":0, "screen_mode_flag":3, "power_system":null, "power_act":0, "power_solar_act":0, "power_grid_act":0, "power_ac9":null, "power_solar_ac9":null, "power_grid_ac9":null, "power1_solar":0, "power1_grid":0, "power2_solar":0, "power2_grid":0, "power3_solar":null, "power3_grid":null, "load_state":"1", "load_nom":3256, "rel1_out":"0", "pump_pwm":0, "temp1":591, "temp2":null, "temp3":null, "temp4":null, "boostactive":0, "legboostnext":"7", "date":"07.03.25", "loctime":"08:07:01", "unixtime":1741331221, "uptime":3556, "wp_flag":0, "wp_time1_ctr":0, "wp_time2_ctr":0, "wp_time3_ctr":0, "schicht_flag":0, "act_night_flag":0, "ctrlstate":"Modbus Read received", "blo [... cut ...] 0, "p9s_upd_state":null, "cloudstate":4, "debug_ip":"0.0.0.0" }
[lp-3 ] DEBUG 2025/03/07 08:08:32 charge power: 0W
[site ] DEBUG 2025/03/07 08:08:32 pv 1 power: 5100W
[site ] DEBUG 2025/03/07 08:08:34 grid power: 50W
[site ] DEBUG 2025/03/07 08:08:34 battery 1 power: -4520W
[site ] DEBUG 2025/03/07 08:08:36 battery 1 soc: 66%
[site ] DEBUG 2025/03/07 08:08:36 grid powers: [10 40 -0]W
[site ] DEBUG 2025/03/07 08:08:37 grid currents: [0 0.08 0]A
[site ] DEBUG 2025/03/07 08:08:37 site power: -4370W

hbpv, 2025-03-11:
Hi @andig kannst du dort ggf. noch schauen? Problem besteht weiterhin. LG

andig, 2025-03-20:
Das wird https://github.com/evcc-io/evcc/issues/19753

closed 2025-03-20 by andig

hbpv, 2025-06-18:
Hi @andig 
Ich muss das leider nochmal öffnen. Ich habe einen ELWA-2 mit evcc 204.2 und es kommt immer noch das Warning:

```
**[lp-2 ] WARN 2025/06/18 09:29:51 charger out of sync: expected enabled, got disabled**
[lp-2 ] DEBUG 2025/06/18 09:29:51 charger disable
[lp-2 ] DEBUG 2025/06/18 09:29:51 pv charge current: 20.2A = 0A + 20.2A (-4653W @ 1p)
[lp-2 ] DEBUG 2025/06/18 09:29:51 site power -4653W <= 0W enable threshold
[lp-2 ] DEBUG 2025/06/18 09:29:51 pv enable timer start: 1m0s
[lp-2 ] DEBUG 2025/06/18 09:29:51 pv enable in 1m0s
[lp-2 ] DEBUG 2025/06/18 09:30:20 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:30:50 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:30:51 charger status: B
[lp-2 ] DEBUG 2025/06/18 09:30:51 charger soc limit: 61%
[lp-2 ] DEBUG 2025/06/18 09:30:51 pv charge current: 21.8A = 0A + 21.8A (-5021W @ 1p)
[lp-2 ] DEBUG 2025/06/18 09:30:51 site power -5021W <= 0W enable threshold
[lp-2 ] DEBUG 2025/06/18 09:30:51 pv enable in 0s
[lp-2 ] DEBUG 2025/06/18 09:30:51 pv enable timer remaining: 0s
[lp-2 ] DEBUG 2025/06/18 09:31:20 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:31:50 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:31:51 charger status: B
[lp-2 ] DEBUG 2025/06/18 09:31:51 charger soc limit: 61%
[lp-2 ] DEBUG 2025/06/18 09:31:51 pv charge current: 20.5A = 0A + 20.5A (-4707W @ 1p)
[lp-2 ] DEBUG 2025/06/18 09:31:51 site power -4707W <= 0W enable threshold
[lp-2 ] DEBUG 2025/06/18 09:31:51 pv enable in 0s
[lp-2 ] DEBUG 2025/06/18 09:31:51 pv enable timer elapsed
[lp-2 ] DEBUG 2025/06/18 09:31:51 pv timer reset
[lp-2 ] DEBUG 2025/06/18 09:31:51 pv timer inactive
[lp-2 ] DEBUG 2025/06/18 09:31:51 charger enable
[lp-2 ] DEBUG 2025/06/18 09:31:51 wake-up timer: start
[lp-2 ] DEBUG 2025/06/18 09:32:20 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:32:50 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:32:51 charger status: B
[lp-2 ] DEBUG 2025/06/18 09:32:51 charger soc limit: 61%
[lp-2 ] DEBUG 2025/06/18 09:32:51 pv charge current: 20.1A = 0A + 20.1A (-4615W @ 1p)
[lp-2 ] DEBUG 2025/06/18 09:32:51 set charge current limit: 14A
[lp-2 ] DEBUG 2025/06/18 09:33:20 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:33:50 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:33:51 charger status: B
[lp-2 ] DEBUG 2025/06/18 09:33:51 charger soc limit: 61%
**[lp-2 ] WARN 2025/06/18 09:33:51 charger out of sync: expected enabled, got disabled**
[lp-2 ] DEBUG 2025/06/18 09:33:51 charger disable
[lp-2 ] DEBUG 2025/06/18 09:33:51 pv charge current: 20.2A = 0A + 20.2A (-4640W @ 1p)
[lp-2 ] DEBUG 2025/06/18 09:33:51 site power -4640W <= 0W enable threshold
[lp-2 ] DEBUG 2025/06/18 09:33:51 pv enable timer start: 1m0s
[lp-2 ] DEBUG 2025/06/18 09:33:51 pv enable in 1m0s
[lp-2 ] DEBUG 2025/06/18 09:34:20 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:34:50 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:34:51 charger status: B
[lp-2 ] DEBUG 2025/06/18 09:34:51 charger soc limit: 61%
[lp-2 ] DEBUG 2025/06/18 09:34:51 pv charge current: 19.9A = 0A + 19.9A (-4568W @ 1p)
[lp-2 ] DEBUG 2025/06/18 09:34:51 site power -4568W <= 0W enable threshold
[lp-2 ] DEBUG 2025/06/18 09:34:51 pv enable in 0s
[lp-2 ] DEBUG 2025/06/18 09:34:51 pv enable timer elapsed
[lp-2 ] DEBUG 2025/06/18 09:34:51 pv timer reset
[lp-2 ] DEBUG 2025/06/18 09:34:51 pv timer inactive
[lp-2 ] DEBUG 2025/06/18 09:34:51 set charge current limit: 4A
[lp-2 ] DEBUG 2025/06/18 09:34:51 charger enable
[lp-2 ] DEBUG 2025/06/18 09:34:51 wake-up timer: start
[lp-2 ] DEBUG 2025/06/18 09:35:20 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:35:50 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:35:54 charger status: B
[lp-2 ] DEBUG 2025/06/18 09:35:54 charger soc limit: 61%
**[lp-2 ] WARN 2025/06/18 09:35:54 charger out of sync: expected enabled, got disabled**
[lp-2 ] DEBUG 2025/06/18 09:35:54 charger disable
[lp-2 ] DEBUG 2025/06/18 09:35:54 pv charge current: 10.1A = 0A + 10.1A (-2316W @ 1p)
[lp-2 ] DEBUG 2025/06/18 09:35:54 site power -2316W <= 0W enable threshold
[lp-2 ] DEBUG 2025/06/18 09:35:54 pv enable timer start: 1m0s
[lp-2 ] DEBUG 2025/06/18 09:35:54 pv enable in 1m0s
[lp-2 ] DEBUG 2025/06/18 09:36:20 charge power: 0W
[... 11 lines cut ...]
[lp-2 ] DEBUG 2025/06/18 09:37:51 pv charge current: 20.2A = 0A + 20.2A (-4638W @ 1p)
[lp-2 ] DEBUG 2025/06/18 09:37:51 site power -4638W <= 0W enable threshold
[lp-2 ] DEBUG 2025/06/18 09:37:51 pv enable in 0s
[lp-2 ] DEBUG 2025/06/18 09:37:51 pv enable timer elapsed
[lp-2 ] DEBUG 2025/06/18 09:37:51 pv timer reset
[lp-2 ] DEBUG 2025/06/18 09:37:51 pv timer inactive
[lp-2 ] DEBUG 2025/06/18 09:37:51 charger enable
[lp-2 ] DEBUG 2025/06/18 09:37:51 wake-up timer: start
[lp-2 ] DEBUG 2025/06/18 09:38:20 charge power: 1174W
[lp-2 ] DEBUG 2025/06/18 09:38:50 charge power: 1181W
[lp-2 ] DEBUG 2025/06/18 09:38:51 charger status: C
[lp-2 ] INFO 2025/06/18 09:38:51 start charging ->
[lp-2 ] DEBUG 2025/06/18 09:38:51 wake-up timer: stop
[lp-2 ] DEBUG 2025/06/18 09:38:51 charger soc limit: 61%
[lp-2 ] DEBUG 2025/06/18 09:38:51 pv charge current: 26.1A = 5.13A + 21A (-4821W @ 1p)
[lp-2 ] DEBUG 2025/06/18 09:38:51 set charge current limit: 14A
[lp-2 ] DEBUG 2025/06/18 09:39:20 charge power: 3550W
[lp-2 ] DEBUG 2025/06/18 09:39:50 charge power: 0W
[lp-2 ] DEBUG 2025/06/18 09:39:51 charger status: B
[lp-2 ] INFO 2025/06/18 09:39:51 stop charging <-
[lp-2 ] DEBUG 2025/06/18 09:39:51 charger soc limit: 61%
**[lp-2 ] WARN 2025/06/18 09:39:51 charger out of sync: expected enabled, got disabled**
[lp-2 ] DEBUG 2025/06/18 09:39:51 charger disable
[lp-2 ] DEBUG 2025/06/18 09:39:51 limitSoc reached: 63.6% > 63%
```

andig, 2025-06-28:
Das Log ist ohne die Ausgabe von site nicht lesbar da die Abgrenzung der Zyklen fehlt

hbpv, 2025-06-30:
> Das Log ist ohne die Ausgabe von site nicht lesbar da die Abgrenzung der Zyklen fehlt

```
[site ] DEBUG 2025/06/30 15:13:12 ----
[heizstabmeter] TRACE 2025/06/30 15:13:12 GET http://192.168.178.92/data.jsn
[heizstabmeter] TRACE 2025/06/30 15:13:12 { "device":"AC ELWA 2", "fwversion":"e0001301", "psversion":"ep108", "coversion":"ec104", "fsetup":0, "p1_s":0, "p1_v":"null", "p2_s":240, "p2_v":"0001301", "p_co_s":240, "p_co_v":"ec104", "p_ps_s":240, "p_ps_v":"ep108", "power_system":null, "screen_mode_flag":3, "power_elwa2":0, "power_solar":0, "power_grid":0, "power1_solar":0, "power1_grid":0, "power2_solar":0, "power2_grid":0, "power3_solar":0, "power3_grid":0, "rel1_out":0, "rel_selv":0, "power_nominal":3000, "power_max":0, "temp1":581, "temp2":0, "boostactive":0, "legboostnext":"null", "date":"30.06.25", "loctime":"15:13:12", "unixtime":1751289192, "uptime":2, "ctrlstate":"Modbus Read received", "blockactive":0, "meter1_id":null, "meter1_ip":"null", "meter2_id":null, "meter2_ip":"null", "meter3_id":null, "meter3_ip":"null", "meter4_id":null, "meter4_ip":"null", "meter5_id":null, "meter5_ip":"null", "meter6_id":null, "meter6_ip":"null", "meter_ss":null, "meter_ssid":"null", "surplus":null, "m0sum":null, "m0l1":null, "m0l2":null, "m0l3":null, "m0bat":null, "m1sum":null, "m1l1":null, "m1l2":null, "m1l3":null, "m1devstate":null, "m2sum":null, "m2l1":null, "m2l2":null, "m2l3":null, "m2soc":null, "m2state":null, "m2devstate":null, "m3sum":null, "m3l1":null, "m3l2":null, "m3l3":null, "m3soc":null, "m3devstate":null, "m4sum":null, "m4l1":null, "m4l2":null, "m4l3":null, "m4devstate":null, "batblockstate":null, "ecarstate":"null", "ecarboostctr":null, "mss2":"null", "mss3":"null", "mss4":"null", "mss5":"null", "mss6":"null", "mss7":"null", "mss8":"null", "mss9":"null", "mss10":"null", "mss11":"null", "volt_mains":240, "volt_aux":5, "freq":50008, "temp_ps":351, "fan_speed":0, "ps_state":0, "ctrl_errors":0, "warnings":0, "fwversionlatest":"e0001401", "coversionlatest":"ec104", "psversionlatest":"ep108", "upd_state":1, "ps_upd_state":0, "co_upd_state":0, "cur_eth_mode":0, "wifi_signal":0, "wifi_list":[{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0}], "cur_ip":"192.168.178.92", "cur_sn":"255.255.255.0", "cur_gw":"192.168.178.1", "cur_dns":"192.168.178.1", "cloudstate":4, "debug_ip":"0.0.0.0" }
[lp-2 ] DEBUG 2025/06/30 15:13:12 charge power: 0W
[heizstabmeter] TRACE 2025/06/30 15:13:12 GET http://192.168.178.92/data.jsn
[heizstabmeter] TRACE 2025/06/30 15:13:12 { "device":"AC ELWA 2", "fwversion":"e0001301", "psversion":"ep108", "coversion":"ec104", "fsetup":0, "p1_s":0, "p1_v":"null", "p2_s":240, "p2_v":"0001301", "p_co_s":240, "p_co_v":"ec104", "p_ps_s":240, "p_ps_v":"ep108", "power_system":null, "screen_mode_flag":3, "power_elwa2":0, "power_solar":0, "power_grid":0, "power1_solar":0, "power1_grid":0, "power2_solar":0, "power2_grid":0, "power3_solar":0, "power3_grid":0, "rel1_out":0, "rel_selv":0, "power_nominal":3000, "power_max":0, "temp1":581, "temp2":0, "boostactive":0, "legboostnext":"null", "date":"30.06.25", "loctime":"15:13:12", "unixtime":1751289192, "uptime":2, "ctrlstate":"Modbus Read received", "blockactive":0, "meter1_id":null, "meter1_ip":"null", "meter2_id":null, "meter2_ip":"null", "meter3_id":null, "meter3_ip":"null", "meter4_id":null, "meter4_ip":"null", "meter5_id":null, "meter5_ip":"null", "meter6_id":null, "meter6_ip":"null", "meter_ss":null, "meter_ssid":"null", "surplus":null, "m0sum":null, "m0l1":null, "m0l2":null, "m0l3":null, "m0bat":null, "m1sum":null, "m1l1":null, "m1l2":null, "m1l3":null, "m1devstate":null, "m2sum":null, "m2l1":null, "m2l2":null, "m2l3":null, "m2soc":null, "m2state":null, "m2devstate":null, "m3sum":null, "m3l1":null, "m3l2":null, "m3l3":null, "m3soc":null, "m3devstate":null, "m4sum":null, "m4l1":null, "m4l2":null, "m4l3":null, "m4devstate":null, "batblockstate":null, "ecarstate":"null", "ecarboostctr":null, "mss2":"null", "mss3":"null", "mss4":"null", "mss5":"null", "mss6":"null", "mss7":"null", "mss8":"null", "mss9":"null", "mss10":"null", "mss11":"null", "volt_mains":240, "volt_aux":5, "freq":50008, "temp_ps":351, "fan_speed":0, "ps_state":0, "ctrl_errors":0, "warnings":0, "fwversionlatest":"e0001401", "coversionlatest":"ec104", "psversionlatest":"ep108", "upd_state":1, "ps_upd_state":0, "co_upd_state":0, "cur_eth_mode":0, "wifi_signal":0, "wifi_list":[{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0},{"ssid":"","signal":0}], "cur_ip":"192.168.178.92", "cur_sn":"255.255.255.0", "cur_gw":"192.168.178.1", "cur_dns":"192.168.178.1", "cloudstate":4, "debug_ip":"0.0.0.0" }
[site ] DEBUG 2025/06/30 15:13:12 aux 1 power: 0W
[site ] DEBUG 2025/06/30 15:13:12 grid power: -7215W
[site ] DEBUG 2025/06/30 15:13:12 battery 1 power: -19W
[site ] DEBUG 2025/06/30 15:13:12 battery 1 soc: 100%
[site ] DEBUG 2025/06/30 15:13:13 pv 1 power: 7561W
[... cut ...]
[lp-2 ] DEBUG 2025/06/30 15:18:43 set charge current limit: 4A
[ac-elwa-2] TRACE 2025/06/30 15:18:43 modbus: send 03 61 00 00 00 09 01 10 03 e8 00 01 02 03 98
[ac-elwa-2] TRACE 2025/06/30 15:18:43 modbus: recv 03 61 00 00 00 06 01 10 03 e8 00 01
[lp-2 ] DEBUG 2025/06/30 15:18:43 charger enable
[lp-2 ] DEBUG 2025/06/30 15:18:43 wake-up timer: start
[ac-elwa-2] TRACE 2025/06/30 15:18:49 modbus: send 03 62 00 00 00 06 01 03 03 e8 00 01
[ac-elwa-2] TRACE 2025/06/30 15:18:49 modbus: recv 03 62 00 00 00 05 01 03 02 00 00
```