evcc-io/evcc #19482: Skoda Enyaq Klimatisierung

l-mango-l, 2025-03-06:
**Is your feature request related to a problem? Please describe.**
Beim Skoda Enyaq funktioniert aktuell noch keine Klimatisierung nach https://docs.evcc.io/docs/features/vehicle#klimatisierung

**Describe the solution you'd like**
Nach dem Anstecken des Fahrzeugs wird bei der Skoda API auch foldendes abgerufen:
```
[enyaq ] TRACE 2025/03/06 09:43:24 GET https://mysmob.api.connect.skoda-auto.cz/api/v2/air-conditioning/TMBJB*
[enyaq ] TRACE 2025/03/06 09:43:25 {"state":"OFF","runningRequests":[],"targetTemperature":{"temperatureValue":16.0,"unitInCar":"CELSIUS"},"airConditioningAtUnlock":true,"windowHeatingEnabled":true,"steeringWheelPosition":"LEFT","seatHeatingActivated":{"frontLeft":true,"frontRight":false},"chargerConnectionState":"CONNECTED","chargerLockState":"UNLOCKED","windowHeatingState":{"front":"OFF","rear":"OFF","unspecified":"INVALID"},"timers":[{"id":1,"enabled":true,"time":"05:15","type":"RECURRING","selectedDays":["MONDAY","WEDNESDAY"]},{"id":2,"enabled":false,"time":"06:30","type":"RECURRING","selectedDays":["TUESDAY","FRIDAY"]}],"carCapturedTimestamp":"2025-03-06T06:16:38Z","errors":[]}
```

Danach wird für den SOC Poll nur noch folgendes abgerufen:
[enyaq ] TRACE 2025/03/06 13:35:34 GET https://mysmob.api.connect.skoda-auto.cz/api/v1/charging/TMBJB*

Bei SOC Poll auch die "air-conditioning" API abrufen und entsprechend Laden aktivieren.

**Describe alternatives you've considered**
In der API sind auch die geplanten Klimatisierungen enthalten. Ggf. können diese beim Anstecken des Fahrzeugs in die Ladeplanung mit aufgenommen werden.

**Additional context**

andig, 2025-03-06:
Wurde von @GrimmiMeloni in https://github.com/evcc-io/evcc/pull/14349 ausgebaut.

l-mango-l, 2025-03-06:
@GrimmiMeloni Warum wurde das Feature ausgebaut?

GrimmiMeloni, 2025-03-07:
Der im OP gezeigte Flow stammt aus der alten Skoda Connect API, die nicht mehr länger von Skoda unterstützt wird. 
In der neuen MySkoda API gibt es diese Information meines Wissens nach nicht mehr.
Daher habe ich bei der Umstellung den Climater entfernt.

Technisch funktioniert das ganze also nur für Enyaqs MJ <= 2022 die noch vor der Abschaltung in Skoda Connect registriert wurden. Für neuere Modelle funktioniert dieser Weg nicht mehr.

Wenn jemand einen Weg gefunden hat, wie der Klimatisierungszustand über MySkoda abgerufen werden kann, sollten wir es wieder einbauen.

GrimmiMeloni, 2025-03-07:
Es muß mich hier nochmal selbst korrigieren, denn es ist genau anders herum. Der Change damals war etwas "vielschichtig". 😓 

Erstmal: Die gezeigten Aufrufe sind bereits von der MySkoda API.

Also Stand ist:
Im Enyaq Template (aka MySkoda, aka aktuelle API) ist die gewünschte Funktionalität vorhanden. 
Wenn es nicht funktioniert, sollten wir nochmal genau draufschauen.

Hinweis zur Historie:
Im alten bzw. generischen Skoda Template war der Climater nie vorhanden. Der Remove den Andig gesehen hat, ist ein Revert, der dadurch Zustande kam, daß ich zunächst möglichst viel des Codes wiederverwenden wollte (siehe #14186). Es stellte sich aber im Nachhinein heraus, daß MySkoda in sein eigenes Package muß.

Es ist also andersherum. Ich habe effektiv in #14349 zumindest for Enyaq/MySkoda damals den Climater hinzugefügt.

l-mango-l, 2025-03-07:
@GrimmiMeloni Vielen Dank für deine schnelle Rückmeldung und auch gererell vielen Dank dir aund dem ganzen evcc-Team für eure super Arbeit. 
Falls ich dich irgendwie unterstützen kann, gib einfach Bescheid.

closed 2025-03-08 by andig

Giermann, 2025-03-11:
Hallo zusammen,
ich bin auf diesen Issue gekommen, als ich die Release-Notes gelesen habe.
In der umgesetzten Implementierung wird nur auf "HEATING" geprüft, daher mir kam gleich die Frage "und was ist mit COOLING?"...

Lt. MySkoda Python Library gibt es außerdem noch AUXILIARY, ON und VENTILATION:
https://github.com/skodaconnect/myskoda/blob/main/myskoda/models/air_conditioning.py#L28

Ich habe leider keinen Enyaq und kann es bei meinem Superb auch aktuell nicht testen. Aber in meinen Augen sollten diese 4 Stati auch berücksichtigt werden.

andig, 2025-03-11:
Gerne PR 👍🏻

Giermann, 2025-03-11:
Hmm... ich bin noch nicht ganz fit in Go - jetzt meckert Lint, dass ich mit ":=" keine neue Variable benannt habe, darüber scheint es aber für "err" okay zu sein...

PR #19523 (2025-03-08): Skoda: add VehicleClimater Interface
fix #19482