evcc-io/evcc #19489: Add runtime-only templates to simplify testing

andig, 2025-03-06:
Die sind tatsächlich mit eincompiliert. Die Idee ist aber spannend: zu Testzwecken wäre es vllt. gut, die auch aus einer Datei einbinden zu können? Das sollte sich machen lassen.

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/17625#discussioncomment-12415532_

andig, 2025-03-06:
S.o.

CiNcH83, 2025-03-06:
OK, dachte, die `{{ }}` enthalten Go-Programmcode, der kompiliert werden muss. Aber dann wird das auch nur zur Laufzeit interpretiert? (Ich habe leider keine Ahnung von Go...)

andig, 2025-03-06:
Das ist eine Go-spezifische Templatesprache und wird zur Laufzeit bzw. beim Start interpretiert. Die Templates selbst müssen aber (bis wir das hier gemacht haben) eincompiliert werden. Mit diesem Issue kann man die auch ohne Compiler testen.

closed 2025-03-07 by andig