evcc-io/evcc #19491: Fehler beim erstellen des Skoda Enyaq

jheinitz, 2025-03-06:
### Describe the bug

Moin!
Ich habe gerade das Upgrade von `0.200.5` auf `0.200.6` gemacht und nach dem Neustart konnte der **Enyaq** nicht mehr eingerichtet werden. Ich bekomme direkt zu Beginn den folgenden Fehler, den ich so noch nicht gesehen habe:
```
ERROR 2025/03/06 17:03:20 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': invalid character 'c' looking for beginning of object key string
```
Ich habe dann versucht wieder auf `0.200.5` zurück zu gehen, aber der Fehler bleibt bestehen. 

In der **MySkoda** App gibt es auch keine neuen Bedingungen, die ich akzeptieren muss. Hat noch jemand das Problem? Ich habe gerade bis kurz vor dem Upgrade noch erfolgreich die Sonne hier im Norden zum Laden genutzt. 

Logs mit `enyaq: trace` kann ich liefern. Ich weiß nur nicht so genau was ich da maskieren sollte.

Viele Grüße

Jens

### Steps to reproduce

1. Restart evcc
2. Check Log
3. Notice that vehicle cannot be created. See error in description
...

### Configuration details

```yaml
not changed
```

### Log details

```text
ERROR 2025/03/06 17:03:20 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': invalid character 'c' looking for beginning of object key string
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.6

zachelnet, 2025-03-06:
Kann ich bestätigen.

Thunderbolt, 2025-03-06:
Gleicher Fehler hier ebenfalls VW ID.7

zachelnet, 2025-03-06:
Anbei von mir den Tracelog

[evcc-20250306-165014-trace.log](https://github.com/user-attachments/files/19112369/evcc-20250306-165014-trace.log)

Chris45671, 2025-03-06:
Hier ein Skoda Enjaq und habe diesen Fehler in 200.5 gehabt und erst danach auf 200.6 aktualisiert. Denke daher dass der Fehler entweder am Server zu suchen ist oder es Änderungen gegeben hat.

zachelnet, 2025-03-06:
Sieht so aus, als hätte VW wieder etwas geändert.

In der Antwort von `https://identity.vwgroup.io/signin-service/v1/3c756d46-f1ba-4d78-9f9a-cff0d5292d51@apps_vw-dilab_com/login/authenticate?relayState=efdb5065c422c314f10d963e449fca7545428e43` ist Javascript-Code enthalten, der anscheinend von evcc falsch interpretiert wird.

Pnuts76, 2025-03-06:
Selbiger Fehler hier mit einem We Connect und einem ID.7:
`[main  ] ERROR 2025/03/06 18:03:46 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': invalid character 'c' looking for beginning of object key string`

VW Server sind "up and running" und bspw. über den ioBroker VW Connect-Adapter kommen die Daten noch an.

andig, 2025-03-06:
VW schraubt mal wieder...

MatzeG, 2025-03-06:
Bekomme gerade mit der Version 0.200.7 nen [main ] ERROR 2025/03/06 18:58:37 creating vehicle DienstWagen failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': 405 Method Not Allowed

wittmannchris, 2025-03-06:
> Bekomme gerade mit der Version 0.200.7 nen [main ] ERROR 2025/03/06 18:58:37 creating vehicle DienstWagen failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': 405 Method Not Allowed
> 
> Liegt das an VW ?

Selbiges bei mir. 

200.7 funzt auch nicht.,..

markogenzel, 2025-03-06:
> Bekomme gerade mit der Version 0.200.7 nen [main ] ERROR 2025/03/06 18:58:37 creating vehicle DienstWagen failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': 405 Method Not Allowed
> 
> Liegt das an VW ?

same here mit 200.7

Velandriel, 2025-03-06:
Jap auch kein VW unter 200.7

Pnuts76, 2025-03-06:
@andig :  reopen Kandidat?
Fehlermeldung:
`[main  ] ERROR 2025/03/06 19:27:47 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': 405 Method Not Allowed`

StefanSchoof, 2025-03-06:
Ich glaube wir brauchen hier keine me too mehr. Klickt einfach unten auf Subscribe und wartet ab.

andig, 2025-03-06:
Please try nightly in 20min. Retracted release. Sign-in seems to work but at least for Audi I don't get any data now :(

jheinitz, 2025-03-06:
I can confirm that `evcc vehicle` returns data from my Enyaq. Version of evcc is `0.200.6 (ec6fead5).

Thanks @andig for the really really quick solution. 

Cheers 

Jens

Doppellhelix, 2025-03-06:
Audi auch Probleme unter 200.6 & 200.7
`[main ] ERROR 2025/03/06 20:54:38 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'etron': 405 Method Not Allowed `

jheinitz, 2025-03-06:
> Wird es denn auch in der normalen Version bald behoben? Mit der Nightly weiß ich nicht Bescheid, hab immer bisher nur die normale Version geupdatet... 

Ich gehe mal davon aus, dass @andig kurzfristig eine neue Version veröffentlichen wird. Noch scheint es ja Probleme mit Fahrzeugen von Audi zu geben.

RogerHartmann, 2025-03-07:
ich hänge mich hier mal dran...

bei mir beim Cupra ebenfalls nach Update auf 0.200.7.
habe dann ein downgrade gemacht:
0.200.5
[main ] ERROR 2025/03/07 07:05:22 creating vehicle Cupra_Born failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': invalid character 'c' looking for beginning of object key string

0.200.6
[main ] ERROR 2025/03/07 07:06:49 creating vehicle Cupra_Born failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': invalid character 'c' looking for beginning of object key string

und jetzt ist irgendwie die 0.200.7 verschwunden

Falls es eine executable für Linux x64 gibt, kann ich die gerne ausprobieren, mit Docker gehts aber bei mir nicht (ist direkt in debian via apt installiert)

andig, 2025-03-07:
https://github.com/evcc-io/evcc/releases/tag/0.200.8 Ist released. Wenn Skoda läuft machen wir für Audi ein neues Thema.

closed 2025-03-07 by andig

chrischtili, 2025-03-07:
Für Homeassistant gibts aber momentan nur 0.200.6 ?! Das System versucht jedoch 0.200.7 zu installieren, welches nicht verfügbar ist.

Kugelfang666, 2025-03-07:
> Für Homeassistant gibts aber momentan nur 0.200.6 ?! Das System versucht jedoch 0.200.7 zu installieren, welches nicht verfügbar ist.

Einmal auf check update dann gibt es auch 200.8. 

Hat bei mir funktioniert und ich kann bestätigen das mein ID7 nun wieder verbunden ist!

Besten Dank!

Pebberman, 2025-03-07:
Test mit Skoda Enyaq: läuft! Danke 🙏

lurchi38, 2025-03-07:
Mit 200.8 ist das Problem bei meinen ID.4 behoben.
@andig Vielen Dank für die schnelle Hilfe!