evcc-io/evcc #19499: Ladewerte Huawei Sun2000 falsch.

RacerEdge, 2025-03-06:
### Describe the bug

Hallo zusammen,

wie von anidg empfohlen, eröffne ich ein separates Thema.
Seit es jetzt wieder Solarstrom gibt, zeigt die app völlig falsche Werte an.
Letzten August hat das noch problemlos funktioniert.

ich habe PV-Laden ab 90% bis dann auf 80% im evcc eingestellt.

Das Auto lädt auch sehr langsam, unabhängig vom max. Strom.
Den habe ich eher niedriger, da ich nur 5kw auf dem Dach habe.

Woran kann das liegen?

Danke für die Mühe im Voraus.

Grüsse aus dem Süden der Republik.

### Steps to reproduce

1. warten bis über 90% und Auto verbunden.
2.
3.
...

### Configuration details

```yaml
Starte Ladung bei Batterieladung über 90% bis die Batterie dann 80% hat.
```

### Log details

```text
[lp-1  ] INFO 2025/03/05 12:15:18 vehicle updated: unknown -> ZOE
[lp-1  ] INFO 2025/03/05 12:15:20 car connected
[lp-1  ] INFO 2025/03/05 12:15:20 start charging ->
[lp-1  ] WARN 2025/03/05 12:15:21 charger logic error: current mismatch (got 7.1A measured, expected 6A)
[lp-1  ] INFO 2025/03/05 12:16:55 stop charging <-
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/05 12:27:15 evcc 0.200.5
[main  ] INFO 2025/03/05 12:27:15 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/05 12:27:15 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/03/05 12:27:25 connecting evcc-1875099867 at tcp://192.168.13.5:1883
[main  ] INFO 2025/03/05 12:27:25 listening at :7070
[site  ] INFO 2025/03/05 12:27:28 site config:
[site  ] INFO 2025/03/05 12:27:28   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/03/05 12:27:28     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/03/05 12:27:28     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/03/05 12:27:28     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/03/05 12:27:28   vehicles:
[site  ] INFO 2025/03/05 12:27:28     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/03/05 12:27:28   tariffs:
[site  ] INFO 2025/03/05 12:27:28     grid:      ✗
[site  ] INFO 2025/03/05 12:27:28     feed-in:   ✗
[site  ] INFO 2025/03/05 12:27:28     co2:       ✗
[site  ] INFO 2025/03/05 12:27:28     solar:     solar
[lp-1  ] INFO 2025/03/05 12:27:28 loadpoint 1:
[lp-1  ] INFO 2025/03/05 12:27:28   mode:        pv
[lp-1  ] INFO 2025/03/05 12:27:28   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2025/03/05 12:27:28   meters:      charge ✓
[lp-1  ] INFO 2025/03/05 12:27:28     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2025/03/05 12:27:28 vehicle updated: unknown -> ZOE
[lp-1  ] INFO 2025/03/05 12:27:30 car connected
[lp-1  ] INFO 2025/03/05 12:28:33 start charging ->
[lp-1  ] WARN 2025/03/05 12:31:39 charger logic error: current mismatch (got 7.1A measured, expected 6A)
[lp-1  ] WARN 2025/03/05 12:34:15 charger logic error: current mismatch (got 7.1A measured, expected 6A)
[lp-1  ] INFO 2025/03/05 12:35:16 stop charging <-
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/05 12:45:32 evcc 0.200.5
[main  ] INFO 2025/03/05 12:45:32 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/05 12:45:32 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/03/05 12:45:32 connecting evcc-1158444698 at tcp://192.168.13.5:1883
[main  ] INFO 2025/03/05 12:45:32 listening at :7070
[site  ] INFO 2025/03/05 12:45:35 site config:
[site  ] INFO 2025/03/05 12:45:35   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/03/05 12:45:35     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/03/05 12:45:35     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/03/05 12:45:35     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/03/05 12:45:35   vehicles:
[site  ] INFO 2025/03/05 12:45:35     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/03/05 12:45:35   tariffs:
[site  ] INFO 2025/03/05 12:45:35     grid:      ✗
[site  ] INFO 2025/03/05 12:45:35     feed-in:   ✗
[site  ] INFO 2025/03/05 12:45:35     co2:       ✗
[site  ] INFO 2025/03/05 12:45:35     solar:     solar
[lp-1  ] INFO 2025/03/05 12:45:35 loadpoint 1:
[lp-1  ] INFO 2025/03/05 12:45:35   mode:        pv
[lp-1  ] INFO 2025/03/05 12:45:35   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2025/03/05 12:45:35   meters:      charge ✓
[lp-1  ] INFO 2025/03/05 12:45:35     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2025/03/05 12:45:35 vehicle updated: unknown -> ZOE
[lp-1  ] INFO 2025/03/05 12:45:37 car connected
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/05 12:48:07 evcc 0.200.5
[main  ] INFO 2025/03/05 12:48:07 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/05 12:48:07 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/03/05 12:48:08 connecting evcc-343261183 at tcp://192.168.13.5:1883
[main  ] INFO 2025/03/05 12:48:08 listening at :7070
[site  ] INFO 2025/03/05 12:48:11 site config:
[site  ] INFO 2025/03/05 12:48:11   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/03/05 12:48:11     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/03/05 12:48:11     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/03/05 12:48:11     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/03/05 12:48:11   vehicles:
[site  ] INFO 2025/03/05 12:48:11     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/03/05 12:48:11   tariffs:
[site  ] INFO 2025/03/05 12:48:11     grid:      ✗
[site  ] INFO 2025/03/05 12:48:11     feed-in:   ✗
[site  ] INFO 2025/03/05 12:48:11     co2:       ✗
[site  ] INFO 2025/03/05 12:48:11     solar:     solar
[lp-1  ] INFO 2025/03/05 12:48:11 loadpoint 1:
[lp-1  ] INFO 2025/03/05 12:48:11   mode:        pv
[lp-1  ] INFO 2025/03/05 12:48:11   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2025/03/05 12:48:11   meters:      charge ✓
[lp-1  ] INFO 2025/03/05 12:48:11     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2025/03/05 12:48:11 vehicle updated: unknown -> ZOE
[lp-1  ] INFO 2025/03/05 12:48:13 car connected
[lp-1  ] INFO 2025/03/05 13:34:44 start charging ->
[lp-1  ] INFO 2025/03/05 14:15:02 stop charging <-
[lp-1  ] INFO 2025/03/05 14:15:33 start charging ->
[lp-1  ] INFO 2025/03/05 14:18:08 stop charging <-
[lp-1  ] INFO 2025/03/05 15:30:59 start charging ->
[lp-1  ] INFO 2025/03/05 16:16:20 stop charging <-
[lp-1  ] INFO 2025/03/06 08:49:29 car disconnected
[lp-1  ] INFO 2025/03/06 12:20:48 car connected
[main  ] INFO 2025/03/06 12:48:11 new version available: 0.200.6
[lp-1  ] INFO 2025/03/06 13:39:51 start charging ->
[lp-1  ] INFO 2025/03/06 15:22:09 stop charging <-
[lp-1  ] INFO 2025/03/06 15:22:09 car disconnected
[main  ] INFO 2025/03/06 18:48:11 new version available: 0.200.7
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

200.5

andig, 2025-03-06:
Config fehlt

RacerEdge, 2025-03-06:
Was für eine Konfiguration?

RacerEdge, 2025-03-06:
Habe ich doch geschrieben!

RacerEdge, 2025-03-06:
Ab 90% heimakku wird geladen bis er auf 80% ist!

RacerEdge, 2025-03-06:
Was fehlt ??