evcc-io/evcc #19505: heating devices: Hautec heatpump (Keba-Controller)

ttjaden, 2025-03-06:
**Is your feature request related to a problem? Please describe.**

New heating device (additional to IDM, Vaillant, Lambda, Stiebel Eltron, ... )

**Describe the solution you'd like**

Device Template for Hautec heatpumps over modbus/tcp. Several other heatpumps (like M-Tec) could also use this with slightly adaption, because they also use a controller made by Keba.

My wish on functionality: When energy is cheap, set the heatpump into "boost mode" which means new set temperature on buffer or hot water storage.

**Additional context**

Result could be

```
chargers:
  - name: my_heating
    type: template
    template: hautec
    modbus: tcpip
    host: 192.0.2.2 # ip address of heatpump
    port: 502 # Port
    zone: # Boost zone id for hot water tank or buffer tank (optional)
    setpoint: # Boost temperature setpoint (optional)
```

These holdingregisters are given

-  hot water temperature
-- holding register: 401
-- Unsigned 16 bit (Big Endian)
-- value in °C
-  hot water set temperature
-- holding register: 402
-- Unsigned 16 bit (Big Endian)
-- value in °C
-  buffer storage temperature
-- holding register: 601
-- Unsigned 16 bit (Big Endian)
-- value in °C
-  buffer storage set temperature
-- holding register: 604
-- Unsigned 16 bit (Big Endian)
-- value in °C
-  electrical power consumption of heatpump
-- holding register: 740
-- Unsigned 16 bit (Big Endian)
-- value in kW
-  thermal power generation of heatpump
-- holding register: 743
-- Unsigned 16 bit (Big Endian)
-- value in kW

ttjaden, 2025-03-06:
tried to start to write a yaml

```
template: hautec
products:
  - brand: Hautec
group: heating
requirements:
  # evcc: ["sponsorship"]
params:
  - name: modbus
    choice: ["tcpip"]
  - name: tempsource
    type: choice
    choice: ["", "warmwater", "buffer"]
    description:
      de: "Temperaturquelle"
      en: "Temperature source"
  - name: phases
    type: int
    advanced: true
render: |
  type: heatpump
  phases: {{ .phases }}
  power:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      address: 740 # electrical power consumption of heatpump in kW
      type: holding
      decode: uint16
    scale: 1000
  {{- if .tempsource }}
  temp:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      address: {{ if eq .tempsource "warmwater" -}} 401 {{ else }} 601 {{- end }} # 401 hot water, 601 buffer temp in °C
      type: holding
      decode: uint16
  {{- end }}
```

- don't know if  "scale: 1000" is right to convert kW to W
- don't know how to integrate the writable holding register 402 and 604 to define set temperature values

andig, 2025-03-07:
Please open PR, easier to fiscuss the code.

closed 2025-03-07 by andig