evcc-io/evcc #19506: Wrong version on DockerHub

zigad, 2025-03-06:
### Describe the bug

Hey @andig ,

I have updated the app via docker and I saw that there is now a wrong latest version on docker hub.

I think it was accidentally triggered with this action run: https://github.com/evcc-io/evcc/actions/runs/13704865618

Can you guys update latest tag to version 0.200.6?

[image]

As I can see latest tag has different Digest than 0.200.6

[image]

As I can see from the url the tag was already deleted: https://github.com/evcc-io/evcc/releases/tag/0.200.7

### Steps to reproduce

N/A

### Configuration details

```yaml
N/A
```

### Log details

```text
[main ] INFO 2025/03/06 21:08:42 evcc 0.200.7
[main ] INFO 2025/03/06 21:08:42 using config file: /etc/evcc.yaml
[db ] INFO 2025/03/06 21:08:42 using sqlite database: /root/.evcc/evcc.db
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

latest, should be 0.200.6 but is 0.200.7

StefanSchoof, 2025-03-06:
0.200.7 was retracted: https://github.com/evcc-io/evcc/issues/19491#issuecomment-2704680083

christophwen, 2025-03-06:
I have the same problem with the HA addon of evcc:

[image]

zigad, 2025-03-07:
> 0.200.7 was retracted: [#19491 (comment)](https://github.com/evcc-io/evcc/issues/19491#issuecomment-2704680083)

That's fine but I think the latest container should also be changed back to 0.200.6

zigad, 2025-03-07:
This is also affecting this #19509 and other users.

andig, 2025-03-07:
See https://github.com/evcc-io/evcc/releases/tag/0.200.8

closed 2025-03-07 by andig

asmcc, 2025-03-07:
Yes, but update from HA is stil not possible, since HA awaiting the 0.200.7 and can not update directly from 0.200.6 to 0.200.8. Do you have a workaround for it?
I think, that was not a good idea with retracting of 0.200.7. At least from HA point of view.