evcc-io/evcc #19509: Update from 0.200.5 to 0.200.7 in HomeAsstiant not working

timostr, 2025-03-06:
### Describe the bug

HomeAsstiant has shown available update.
After starting the update following error message popped up:

Error updating evcc: Can't install evcc/evcc:0.200.7: 404 Client Error for http+docker://localhost/v1.47/images/create?tag=0.200.7&fromImage=evcc%2Fevcc&platform=linux%2Farm64: Not Found ("manifest for evcc/evcc:0.200.7 not found: manifest unknown: manifest unknown"

HomeAsstiant Version: 
Core 2025.3.0
Supervisor 2025.03.2
Operating System 14.2
Frontend 20250305.0

### Steps to reproduce

1.Press update evcc
2.
3.
...

### Configuration details

```yaml
...
```

### Log details

```text
Not related to this problem
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

timostr, 2025-03-06:
Seems to be already known problem

Thomas-Doc, 2025-03-06:
me too ...
cant restore previous version, too
"manifest" is missing

MaStr, 2025-03-07:
And my Telegraf pulled 0.200.7 Docker image, but there are no release information.

closed 2025-03-07 by andig

asmcc, 2025-03-07:
Yes, but update from HA is stil not possible, since HA awaiting the 0.200.7 and can not update directly from 0.200.6 to 0.200.8. Do you have a workaround for it?
I think, that was not a good idea with retracting of 0.200.7. At least from HA point of view.

klinkym, 2025-03-07:
Version 0.200.8 is now available in HA. Update from 0.200.5 -> 0.200.8 works

asmcc, 2025-03-07:
But that is not possible, if you have 0.200.6 or if HA found yesterday 0.200.7. In my case it want to update to 0.200.7 and I can not find, how I can directly switch to 0.200.8. Restart of HA does not help. HA stil awaiting 0.200.7 for update and fails

klinkym, 2025-03-07:
You have to wait. In my case, HA also waited for an update to version 0.200.7. Now HA has recognized that there is a newer version and I was able to update directly to 0.200.8. Just wait a few more minutes.

asmcc, 2025-03-07:
Yes, after restart and awaiting of a few minutes I could update directly to 0.200.8