evcc-io/evcc #19510: EVCC als HA-AddOn nach Fehler nicht mehr zu installieren

jove01, 2025-03-06:
### Describe the bug

Ich habe beim Addon 0.200.5 aus Versehen  aud die Updatw-Info gedrückt. EVCC wollte dann  die neue Version laden.
Im AddOn Bildschirm habe ich aus Panik deinstalliwert, Dann eine 1-Tag alte Sicherung des Addons, dann des kompletten HA und schließlich HA ohne EVCC-Addon zurück geladen. Immer dieser Fehler:

`Add-on konnte nicht installiert werden

Can't install evcc/evcc:0.200.7: 404 Client Error for http+docker://localhost/v1.47/images/create?tag=0.200.7&fromImage=evcc%2Fevcc&platform=linux%2Famd64: Not Found ("manifest for evcc/evcc:0.200.7 not found: manifest unknown: manifest unknown")`

Ich hatte gerade EVCC komplett auf UI umgestellt.

Ich benötige dringend Eure Hilfe. wie ich mein HA retten und am Besten auch das EVCC-Backup wieder einspielen kann.

Meine dringende  Bitte um  Hilfe und eine Bitte, damit nicht noch anderen diesen Fehler begehen, im HA-Umfeld bizte den Ginweis auf eine neue Version ausblenden.

Vielen Dank

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
oben beschrieben
```

### Log details

```text
nicht vorhanden
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

.200.5 auf .200.7

hbpv, 2025-03-06:
https://github.com/evcc-io/evcc/issues/19491#issuecomment-2704680083

jove01, 2025-03-06:
Vielen Dank für die schnelle Reaktion.
Mir scheint, das hier abewr ein anderes Problem vorliegt. 

Bei mir geht es darum, dass ich durch meinen Fehler gar nicht in der Lage bin, dass AddOn zu installieren und nicht nur um 1 VW-Fahrzeug.

PS: Ich bin gerade dabei, die komplette Installation in Nightly zu bringen. (Ein Restart dauer dabei aktuell ewig).
Ich möchte dringend wieder in die stabille Version.

Dabke

christophwen, 2025-03-06:
@jove01 das ist schon dasselbe Grund-Problem. 
Es wird versucht die Version 0.200.7 (als Neuste) zu ziehen, aber diese wurde zurückgezogen und dadurch nicht erreicht.

jove01, 2025-03-06:
also warten und dann versuchen neu zu installieren und dann das backup von .200.5 einspielen.

Vielen Dank

Thomas-Doc, 2025-03-06:
Gleiches Problem hier. 0.200.7 hat zum Desaster geführt. Jetzt lässt sich gar nichts mehr installieren. Auch nicht die frühere Version.
Werde morgen versuchen, ein komplettes Backup einzuspielen.
Ansonsten funktioniert aber alles (in Homeassistat OS).

pergolafabio, 2025-03-07:
My HA is still promoting that .7 is avaible, can the config be edited back to .6 so the update goes away?

andig, 2025-03-07:
See https://github.com/evcc-io/evcc/releases/tag/0.200.8

closed 2025-03-07 by andig