evcc-io/evcc #19515: VW ID.4 "Fahrzeug nicht erreichbar" nach Neustart evcc [0.200.5]

lurchi38, 2025-03-07:
### Describe the bug

Nach einem Neustart von evcc erscheint im UI beim Fahrzeug eine durchgestrichene Wolke mit dem Hinweis "Fahrzeug nicht erreichbar". Es werden keine Informationen zum Fahrzeug mehr angezeigt.
Im Protokoll steht eine Fehlermeldung, siehe unten 

### Steps to reproduce

1. Restart evcc
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: VW_ID4
    type: template
    template: vw
    user: xxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: WVGxxxxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    title: ID.4 # Wird in der Benutzeroberfläche angezeigt (optional)
    capacity: 77 # Akkukapazität in kWh (optional)
    #timeout: 10s # optional
    #icon: car # Icon in der Benutzeroberfläche (optional)
    #phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
    #mode: # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
    #minCurrent: 6 # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    #maxCurrent: 16 # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    #identifiers: # Kann meist erst später eingetragen werden, siehe: https://docs.evcc.io/docs/features/vehicle (optional)
    #priority: # Priorität des Ladepunktes oder Fahrzeugs in Relation zu anderen Ladepunkten oder Fahrzeugen für die Zuweisung von PV-Energie (optional)
    #cache: 15m # Zeitintervall nach dem Daten erneut vom Fahrzeug abgefragt werden (optional)
```

### Log details

```text
[main  ] ERROR 2025/03/07 02:10:33 creating vehicle VW_ID4 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': invalid character 'c' looking for beginning of object key string
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

OJack0815, 2025-03-07:
Ich habe das gleiche Problem bei v0.200.6
Gestern war die Verbindung ok bei v0.200.5 und auch nach Update auf v0.200.6

runnel22, 2025-03-07:
Ich denke es ist der gleiche Fehler wie hier #19491

closed 2025-03-07 by andig

runnel22, 2025-03-07:
Das Update 0.200.8 ist verfügbar. 
Mein ID4 wird wieder erkannt

OJack0815, 2025-03-07:
Mein ID.7 funktioniert auch wieder mit 0.200.8
Danke!