evcc-io/evcc #19521: Update von 200.5 auf 200.8 Timeout Fehler

JAK0721, 2025-03-07:
### Describe the bug

Hi, mit dem Update ist evcc gar nicht mehr gestartet mit einem Fehler grid meter. Erst als ich meine RCT Anlage über   das UI mit der IP Adresse neu konfiguriert habe, startet evcc wieder. Allerding bekommen ich jetzt timeout Fehler
[site ] DEBUG 2025/03/07 08:19:08 battery 1 power: -662W
[site ] DEBUG 2025/03/07 08:19:08 battery 1 soc: 20%
[site ] ERROR 2025/03/07 08:19:18 pv 1 power: timeout
[site ] DEBUG 2025/03/07 08:19:18 battery has priority at soc 20% (< 70%)
[site ] DEBUG 2025/03/07 08:19:18 site power: 100W
[lp-1 ] DEBUG 2025/03/07 08:19:18 charge voltages: [236 20 12]V
[lp-1 ] DEBUG 2025/03/07 08:19:18 detected connected phases: 1p
[lp-1 ] DEBUG 2025/03/07 08:19:18 charge total import: 3388.534kWh
[lp-1 ] DEBUG 2025/03/07 08:19:19 charger status: B
[lp-1 ] DEBUG 2025/03/07 08:19:19 pv charge current: 0A = 0A + -0.435A (100W @ 1p)
[site ] DEBUG 2025/03/07 08:19:19 ----
[lp-1 ] DEBUG 2025/03/07 08:19:19 charge power: 0W
[lp-1 ] DEBUG 2025/03/07 08:19:19 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/07 08:19:19 grid power: -1W
[site ] DEBUG 2025/03/07 08:19:19 battery 1 power: -669W
[site ] DEBUG 2025/03/07 08:19:19 pv 1 power: 1158W
[site ] ERROR 2025/03/07 08:19:29 battery 1 soc: timeout
[site ] DEBUG 2025/03/07 08:19:29 battery has priority at soc 0% (< 70%)
[site ] DEBUG 2025/03/07 08:19:29 site power: 99W
[lp-1 ] DEBUG 2025/03/07 08:19:29 charge voltages: [235 20 1

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2025/03/07 08:24:02 evcc 0.200.8
[main  ] INFO 2025/03/07 08:24:02 using config file: /etc/evcc.yaml
[db    ] ERROR 2025/03/07 08:24:09 persist: attempt to write a readonly database (8)
[db    ] ERROR 2025/03/07 08:24:09 persist: attempt to write a readonly database (8)
[db    ] ERROR 2025/03/07 08:24:09 persist: attempt to write a readonly database (8)

Konfiguration (/etc/evcc.yaml):

network:
  schema: http
  host: 192.168.0.240 # .local suffix announces the hostname on MDNS
  port: 7070

log: error
levels:
  #site: trace
  #lp-1: trace
  #daheim: trace
  #hyundai: trace
  #main: trace
  #cache: trace
  #evcc: trace
  influx: trace

interval: 10s # control cycle interval

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

sponsortoken: *****

meters:
- name: grid
  type: template
  template: rct-power
  usage: grid
  host: 192.168.0.225
#  template: volkszaehler-http
#  usage: grid
#  url: http://192.168.0.240/middleware/data # Beispiel: http://zaehler.network.local:8080/api/data # Optional
#  uuid: ee24ade0-1ae3-11ed-a4ff-a56046a7b2fe
- name: pv
  type: template
  template: rct-power
  usage: pv
  host: 192.168.0.225
- name: battery
  type: template
  template: rct-power
  usage: battery
  host: 192.168.0.225

chargers:
- type: template
  template: heidelberg
  id: 2
  device: /dev/ttyUSB0
  baudrate: 19200
  comset: 8E1
  modbus: rs485serial
  name: wallbox

vehicles:
- name: Mercedes
  type: mercedes
  title: Mercedes E300e
  user: *****
  region: EMEA # MB me Region (EMEA, APAC, NORAM)
  vin: *****
  capacity: 25.4 # Akkukapazität in kWh (optional)
  tokens:
    access: *****
    refresh: *****

loadpoints:
- title: Garage
  charger: wallbox
  vehicle: Mercedes
  mode: PV
  phases: 3
#  mincurrent: 6
#  maxcurrent: 16
#  resetOnDisconnect: true
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true

site:
  title: My home
  meters:
    grid: grid
    pv: pv
    battery: battery
  residualPower: 400

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.2688 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.0944 # [currency]/kWh

influx:
  url: http://localhost:8086
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  #token: *****
  token: *****
  org: evcc

Version: `0.200.8`
```

### Log details

```text
[site ] DEBUG 2025/03/07 08:19:08 battery 1 power: -662W
[site ] DEBUG 2025/03/07 08:19:08 battery 1 soc: 20%
[site ] ERROR 2025/03/07 08:19:18 pv 1 power: timeout
[site ] DEBUG 2025/03/07 08:19:18 battery has priority at soc 20% (< 70%)
[site ] DEBUG 2025/03/07 08:19:18 site power: 100W
[lp-1 ] DEBUG 2025/03/07 08:19:18 charge voltages: [236 20 12]V
[lp-1 ] DEBUG 2025/03/07 08:19:18 detected connected phases: 1p
[lp-1 ] DEBUG 2025/03/07 08:19:18 charge total import: 3388.534kWh
[lp-1 ] DEBUG 2025/03/07 08:19:19 charger status: B
[lp-1 ] DEBUG 2025/03/07 08:19:19 pv charge current: 0A = 0A + -0.435A (100W @ 1p)
[site ] DEBUG 2025/03/07 08:19:19 ----
[lp-1 ] DEBUG 2025/03/07 08:19:19 charge power: 0W
[lp-1 ] DEBUG 2025/03/07 08:19:19 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/07 08:19:19 grid power: -1W
[site ] DEBUG 2025/03/07 08:19:19 battery 1 power: -669W
[site ] DEBUG 2025/03/07 08:19:19 pv 1 power: 1158W
[site ] ERROR 2025/03/07 08:19:29 battery 1 soc: timeout
[site ] DEBUG 2025/03/07 08:19:29 battery has priority at soc 0% (< 70%)
[site ] DEBUG 2025/03/07 08:19:29 site power: 99W
[lp-1 ] DEBUG 2025/03/07 08:19:29 charge voltages: [235 20 1
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.8

JAK0721, 2025-03-07:
pi@raspberrypi:~ $ evcc migrate
[main  ] INFO 2025/03/07 08:27:26 evcc 0.200.8
[main  ] INFO 2025/03/07 08:27:26 using config file: /etc/evcc.yaml
[main  ] ERROR 2025/03/07 08:27:26 cannot save settings: attempt to write a readonly database (8)

goebelmeier, 2025-03-07:
Ich würde sagen, dass da generell was mit deinem Raspberry irgendwie im Argen ist. Erster Versuch: Raspberry ein mal rebooten

TobiasHuber1980, 2025-03-07:
[image]

spiel dich mal mit dem Cache

TobiasHuber1980, 2025-03-07:
Und die time out Fehler habe ich auch, sind leider normal bei rct...

JAK0721, 2025-03-07:
@TobiasHuber1980,  vor dem Update auf die 200.8 hatte ich keinerlei Problem mit Timeouts. Lediglig vereinzelt, wenn die RCT Power App parallel an war.

TobiasHuber1980, 2025-03-07:
Erst mal das da fixen!

```

[db    ] ERROR 2025/03/07 08:24:09 persist: attempt to write a readonly database (8)
[db    ] ERROR 2025/03/07 08:24:09 persist: attempt to write a readonly database (8)
[db    ] ERROR 2025/03/07 08:24:09 persist: attempt to write a readonly database (8)

```

JAK0721, 2025-03-07:
@TobiasHuber1980, danke erstmal für Deine Mühe und die Hinweise. 
Stehe allerdings ein wenig im Wald. Vor dem update von der Version 200.5 auf die 200.8 lief alles fehlerfrei. Erwartet die 200.8 nun eine Konfiguration über das UI? Was sagt mir der Fehler [main ] ERROR 2025/03/07 08:27:26 cannot save settings: attempt to write a readonly database (8)