evcc-io/evcc #19524: OpenMeteo Konfiguration Fehler

ThomasMeyer1970, 2025-03-07:
### Describe the bug

Die Konfiguration von OpenMeteo Solar Forecast führt zu folgenden Fehler im Log und der Forecast wird nicht angezeigt: 

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/07 09:03:33 evcc 0.200.8
[main  ] INFO 2025/03/07 09:03:33 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/07 09:03:33 using sqlite database: /data/evcc.db
[main  ] INFO 2025/03/07 09:03:33 listening at :7070
**[tariff] ERROR 2025/03/07 09:03:36 invalid character 'U' looking for beginning of value
[main  ] ERROR 2025/03/07 09:03:36 creating tariff solar failed: cannot create tariff type 'template': cannot create tariff type 'custom': invalid character 'U' looking for beginning of value**
[site  ] INFO 2025/03/07 09:03:37 site config:
[site  ] INFO 2025/03/07 09:03:37   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2025/03/07 09:03:37     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/03/07 09:03:37     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/03/07 09:03:37   vehicles:
[site  ] INFO 2025/03/07 09:03:37     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/03/07 09:03:37   tariffs:
[site  ] INFO 2025/03/07 09:03:37     grid:      priceforecast
[site  ] INFO 2025/03/07 09:03:37     feed-in:   ✗
[site  ] INFO 2025/03/07 09:03:37     co2:       ✗
[site  ] INFO 2025/03/07 09:03:37     solar:     TariffType(0)
[lp-1  ] INFO 2025/03/07 09:03:37 loadpoint 1:
[lp-1  ] INFO 2025/03/07 09:03:37   mode:        minpv
[lp-1  ] INFO 2025/03/07 09:03:37   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/03/07 09:03:37   meters:      charge ✓
[lp-1  ] INFO 2025/03/07 09:03:37     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] INFO 2025/03/07 09:03:38 car connected
[lp-1  ] INFO 2025/03/07 09:03:38 start charging ->

### Steps to reproduce

1.Konfiguration über UI
2.Add On neu starten
3.
...

### Configuration details

```yaml
grid:
  type: template
  template: awattar
  region: AT 
  charges: 0.0144
  tax: 0.2
solar:
  type: template
  template: open-meteo
  lat: 48.18935454240716
  lon: 15.812713801860811
  dec: 42 
  kwp: 5.4
  az: 187 
  ac: 6000 
  dm: 0
  de: 0.35
  efficiency: 93
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/07 09:03:33 evcc 0.200.8
[main  ] INFO 2025/03/07 09:03:33 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/07 09:03:33 using sqlite database: /data/evcc.db
[main  ] INFO 2025/03/07 09:03:33 listening at :7070
**[tariff] ERROR 2025/03/07 09:03:36 invalid character 'U' looking for beginning of value
[main  ] ERROR 2025/03/07 09:03:36 creating tariff solar failed: cannot create tariff type 'template': cannot create tariff type 'custom': invalid character 'U' looking for beginning of value**
[site  ] INFO 2025/03/07 09:03:37 site config:
[site  ] INFO 2025/03/07 09:03:37   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2025/03/07 09:03:37     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/03/07 09:03:37     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/03/07 09:03:37   vehicles:
[site  ] INFO 2025/03/07 09:03:37     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/03/07 09:03:37   tariffs:
[site  ] INFO 2025/03/07 09:03:37     grid:      priceforecast
[site  ] INFO 2025/03/07 09:03:37     feed-in:   ✗
[site  ] INFO 2025/03/07 09:03:37     co2:       ✗
[site  ] INFO 2025/03/07 09:03:37     solar:     TariffType(0)
[lp-1  ] INFO 2025/03/07 09:03:37 loadpoint 1:
[lp-1  ] INFO 2025/03/07 09:03:37   mode:        minpv
[lp-1  ] INFO 2025/03/07 09:03:37   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/03/07 09:03:37   meters:      charge ✓
[lp-1  ] INFO 2025/03/07 09:03:37     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] INFO 2025/03/07 09:03:38 car connected
[lp-1  ] INFO 2025/03/07 09:03:38 start charging ->
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.8

andig, 2025-03-07:
/cc @thecem

thecem, 2025-03-07:
es gibt kein `az: 187`! das muss dann `az: 7 `

thecem, 2025-03-07:
es ist ein süd azimuth kein geo azimuth north

thecem, 2025-03-07:
https://docs.evcc.io/docs/tariffs#openmeteo

closed 2025-03-07 by andig

ThomasMeyer1970, 2025-03-07:
Danke, habe ich übersehen.   Die HA Open-Meteo Solar Forecast Integration will es mit 187.

thecem, 2025-03-07:
Richtig, aber die Meteo API nutzt den Wetter bzw. Sonnen Azimuth.