evcc-io/evcc #19530: manifest for evcc/evcc:0.200.7 not found: manifest unknown: manifest unknown

mmierke, 2025-03-07:
### Describe the bug

Can't install evcc/evcc:0.200.7: 404 Client Error for http+docker://localhost/v1.47/images/create?tag=0.200.7&fromImage=evcc%2Fevcc&platform=linux%2Famd64: Not Found ("manifest for evcc/evcc:0.200.7 not found: manifest unknown: manifest unknown")

### Steps to reproduce

1. Try to update or install 0.200.7 addon in HA
2.
3.
...

### Configuration details

```yaml
HA 2025.3.0
```

### Log details

```text
Failed to install add-on
Can't install evcc/evcc:0.200.7: 404 Client Error for http+docker://localhost/v1.47/images/create?tag=0.200.7&fromImage=evcc%2Fevcc&platform=linux%2Famd64: Not Found ("manifest for evcc/evcc:0.200.7 not found: manifest unknown: manifest unknown")
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-03-07:
https://github.com/evcc-io/evcc/releases/tag/0.200.8

closed 2025-03-07 by andig

mmierke, 2025-03-07:
Aas irgendeinem Grund hat HA bei mir noch die 0.200.7 angezeigt. Ich hab das Repository neu verknüpft, um das Problem zu lösen.