evcc-io/evcc #19537: CPU usage is very high when no loadpoint and charger is defined

marcelGoerentz, 2025-03-07:
### Describe the bug

As the title says, when there is no charger and loadpoint defined the cpu usage goes up to 100%.

I know this is a very special case, but if some user is playing around with the software to check if it fits his needs he might stumble over this bug.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

meters:
- type: template
  template: lg-ess-home-8-10
  usage: grid
  host: xxx
  password: xxx
  name: grid
- type: template
  template: lg-ess-home-8-10
  usage: pv
  host: xxx
  password: xxx
  name: pv
- type: template
  template: lg-ess-home-8-10
  usage: battery
  host: xxx
  password: xxx  name: battery
  capacity: 9.8

mqtt:
  broker: localhost:1883
  topic: evcc
  clientid: evcc-server

site:
  title: My home
  meters:
    grid: grid
    pv:
    - pv
    battery:
    - battery

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.31 # EUR/kWh
  feedin:
    type: fixed
    price: 0.08 # EUR/kWh
```

### Log details

```text
*No Log necessary*
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.6

andig, 2025-03-07:
Thats for the report, will take a look. I guess the site busy-loops over the nonexistent loadpoints.

andig, 2025-03-07:
Closed in 869e5e6bad91f3af2790da1d02f50c6c416f0417

closed 2025-03-07 by andig