evcc-io/evcc #19552: Add vehicle climater to OVMS

stumbaumr, 2025-03-07:
**Is your feature request related to a problem? Please describe.**
The Smart ForFour is using the battery in the morning when heating up. Should rather use energy from the wallbox.

**Describe the solution you'd like**
Enable vehicleClimater for OVMS.

**Additional context**
You should probably use "v.e.hvac" = yes to identify if PreHeat/PreCool is active and then enable the wallbox.

andig, 2025-03-07:
Happy to take a PR

andig, 2025-03-08:
/cc @PanNwt

PanNwt, 2025-03-08:
@stumbaumr: Ich kann gern unterstützen, da wir aber Anfang des letzten Jahres den Smart mit dem OVMS Modul verkauft haben, kann ich es nicht selbst testen.

Tailer67, 2025-03-11:
Ich kann auch gerne unterstützen bzw. testen, habe einen e-UP mit OVMS Modul.

Klimatisierung an:
`curl 'https://dexters-web.de/api/ovms/cmd'   --data-urlencode 'fn.vehicleid=XX-XX999E'   --data-urlencode 'fn.carpass=XXXXXXXX'   --data-urlencode 'fn.cmd=climatecontrol on'
Climate Control 
on
`
Klimatisierung aus
`curl 'https://dexters-web.de/api/ovms/cmd'   --data-urlencode 'fn.vehicleid=XX-XX999E'   --data-urlencode 'fn.carpass=XXXXXXXX'   --data-urlencode 'fn.cmd=climatecontrol off'
Climate Control 
off
`

andig, 2025-03-11:
Wir wollen nur wissen, ob sie sie an ist

Tailer67, 2025-03-11:
Achso, das geht so

`curl 'https://dexters-web.de/api/ovms/cmd' --data-urlencode 'fn.vehicleid=XX-XX999E' --data-urlencode 'fn.carpass=XXXXXXXX' --data-urlencode 'fn.cmd=me li v.e.hvac'

v.e.hvac                                 yes` <-- Klimatisierung ist an

PanNwt, 2025-03-11:
Evcc verwendet die OVMS Server V2 HTTP Api. Davon werden die `charge` und `status` Endpunkte abgefragt und diese enthalten beide das Property `cooldown_active`. Nun ist die Frage, ob dies das Property ist, was wir suchen und welchen Datentyp das Feld hat.

Tailer67, 2025-03-12:
Wie sieht genau die Abfrage von evcc aus?

`https://dexters-web.de/api/call?fn.name=ovms/export&fn.vehicleid=XX-XX999E&fn.carpass=XXXXXXXX&fn.format=csv&fn.types=D,S&fn.last=1 `

Als Antwort bekomme ich
```
#FILE,d.csv
rec_time,0,doors1,doors2,lockunlock,tpem,tmotor,tbattery,trip,odometer,speed,parktimer,ambient,doors3,staletemps,staleambient,vehicle12v,doors4,vehicle12v_ref,doors5,tcharger,vehicle12v_current,cabin_temp
2025-03-12 17:18:27,0,0,8,4,13,17.25,7.17188,67,330130,0,77385,4,0,0,0,12.67,0,12.6,0,8,0,13
```
```
#FILE,s.csv
rec_time,0,soc,units,linevoltage,chargecurrent,chargestate,chargemode,idealrange,estimatedrange,chargelimit,chargeduration,chargeb4,chargekwh,chargesubstate,chargestateN,chargemodeN,chargetimer,chargestarttime,chargetimerstale,cac100,charge_etr_full,charge_etr_limit,charge_limit_range,charge_limit_soc,cooldown_active,cooldown_tbattery,cooldown_timelimit,charge_estimate,charge_etr_range,charge_etr_soc,idealrange_max,chargetype,chargepower,battvoltage,soh,chargepower_input,charger_efficiency,battcurrent,batt_rangespeed,charge_kwh_grid,charge_kwh_grid_total,batt_capacity,charge_timestamp
2025-03-12 17:18:27,0,67.0,K,0,0.00,stopped,standard,91,69,16,0,0,8,14,21,0,0,0,0,42.88,34,0,0,85,-1,0,0,0,0,11,137,0,0.00,383.25,85.76,0.00,0.00,0.00,-0.0
```
Das darin enthaltene Property `cooldown_active` steht immer auf `-1`, es zeigt also nicht eine laufende Klimatisierung an.

Allerdings ändert sich das Property `staleambient` von `0 nach 1,` wenn die Klimatisierung aktiviert wird.
Nach Beendigung der Klimatisierung wechselt das Proberty wieder auf 0, allerdings erst nach 5 Minuten. Vmtl. läuft erst ein Timer in OVMS ab, eine direkte Abfrage mit

`curl 'https://dexters-web.de/api/ovms/cmd'   --data-urlencode 'fn.vehicleid=XX-XX999E'   --data-urlencode 'fn.carpass=XXXXXXXX'   --data-urlencode 'fn.cmd=climatecontrol off'`

lieferte sofort das erwartete Ergebnis `Climate Control off`.

PanNwt, 2025-03-12:
Danke schon einmal für deine Unterstützung.

Kannst du mal bitte die Werte über [den Simple Client](https://dexters-web.de/f/simpleclient.htm) prüfen? Diese verwendet die HTTP Api nach [der OVMS Beschreibung](https://docs.openvehicles.com/en/latest/protocol_httpapi/index.html#), die auch evcc nutzt. Alles was dort sichtbar ist, kann evcc auch auslesen.

Es wäre interessant, ob dort auch der Wert `staleambient` beim zurück die Verzögerung hat.

Tailer67, 2025-03-13:
Testing über den Simple Client hat folgendes ergeben:

```
Klimatisierung eingeschaltet 
{ "m_msgtime_s": "2025-03-13 05:55:00", "staleambient": "1"}
```

`Klimatisierung ausgeschaltet um 05:56:00`
{ "m_msgtime_s": "2025-03-13 05:58:30", "staleambient": "0"}

`staleambient` ist eindeutig ein Indikator für eine laufende Klimatisierung.

und das Delay ist abhängig vom eingestelltem Update Intervall des OVMS zum V2 Server.

[image]

PanNwt, 2025-03-13:
Danke Dir fürs Testen. Werde in den nächsten Tagen einen Merge Request dafür stellen.

closed 2025-03-15 by andig

stumbaumr, 2025-03-17:
Verflixt - eine Woche kuckt man nicht rein und schon ist alles fertig und verfügbar!!!

Schönen Dank für die gute Arbeit!!

FlNDUS, 2025-03-25:
Leider funktioniert für mich staleambient nicht als marker, der Wert wechselt unabhängig ab und zu.
Siehe Issue: https://github.com/evcc-io/evcc/issues/20112