evcc-io/evcc #19562: Solar Formats Erweiterung Anzeige

StevieC121176, 2025-03-08:
Wäre es möglich die Solar Focast Anzeige noch um die Erzeugung bis zum aktuellen Zeitpunkt zu erweitern? Dann hätte man eine schönen grafischen Bezug zum adjusted Forcast.
Evtl. so in der Art:

[image]

andig, 2025-03-08:
Du meinst die Ist-Erzeugung im Tagesverlauf? Dafür gibts im allgemeinen Fall leider keine Daten.

StevieC121176, 2025-03-08:
genau das meine ich. 
Ich hatte 

> track daily PV generation (energyDemand.solarActual- generated until now)

aus #18867 so verstanden, das die tägliche Erzeugung bis zum aktuellen Zeitpunkt zur Berechnung des adjusted Formats gespeichert wird.

andig, 2025-03-08:
Wir akkumulieren Erzeugung, aber unabhängig von irgendeiner Zeitscheibe und einfach als Summe. Fürs erste geht das also nicht, wir schauen uns das aber später wieder an. Da wird noch Einiges kommen ;)

closed 2025-03-08 by andig