evcc-io/evcc #19566: Fehlermeldung: accumulated solar yield: json: unsupported value: NaN

zinsius, 2025-03-08:
### Describe the bug

Hallo,
ich bekomme permanent die Fehlermeldung unsupported value: NaN

Wo kann der Fehler liegen?

Danke
Jörg

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
    schema: http
    host: 192.168.11.250 # .local suffix announces the hostname on MDNS
    port: 7070

# Sponsortoken

sponsortoken: XXXX

# log: debug
log: debug
levels:
    site: debug
    lp-1: debug
    lp-2: debug
    cache: error
    db: error
    ocpp: debug

# unique installation id
plant: XXXX

interval: 30s # control cycle interval

tariffs:
    currency: EUR
    grid:
        type: fixed
        price: 0.2895
    feedin:
        type: fixed
        price: 0.2443
 #   co2:
 #       type: grünstromindex
  #      zip: 86502 # PLZ mit führender Null mit in "" setzen

chargers:
- name: DUOSIDA
  type: template
  template: ocpp
  remotestart: true

meters:
- name: grid1
  type: custom ##Leistungswert aus solarview EVU-Zähler
  power:
    source: mqtt
    topic: homeassistant/sensor/solarview/D0/DPAC
    scale: -1
  energy:
    source: mqtt
    topic: homeassistant/sensor/solarview/D0/GKT0
    scale: 1
- name: pv1
  type: custom
  power:    ##Leistungswert aus solarview Umrichter
    source: mqtt
    topic: hmd/sensor/Viessmann-Gridbox/Photovoltaic/state
    scale: 1
  energy:   ##Gesamtertrag aus solarview Umrichter
    source: mqtt
    topic: viessmann/pvertrag
    scale: 1
- name: pv2
  type: custom
  power:    ##Leistungswert aus solarview Umrichter
    source: mqtt
    topic: inverter/sensor/power_in_total/state
    scale: 1
  energy:   ##Gesamtertrag aus solarview Umrichter
    source: mqtt
    topic: inverter/sensor/cumulated_energy_today/state
    scale: 1
- name: battery1
  type: custom
  power:    ##Leistungswert aus Viessmann WR
    source: mqtt
    topic: hmd/sensor/Viessmann-Gridbox/Battery-1-Power/state
    scale: 1
  soc:
    source: mqtt
    topic: hmd/sensor/Viessmann-Gridbox/Battery-1-Level/state
  capacity: 5.0
    
loadpoints:
- title: Garage
  charger: DUOSIDA
  mode: off
  minCurrent: 6 # minimum charge current (default 6A)
  maxCurrent: 16 # maximum charge current (default 16A)
  phases: 3

#  vehicle: Mini, Smart
  
 
mqtt:
    broker: 192.168.11.250:1883
    topic: evcc
    user: XXXX
    password: XXXX
    
site:
  title: AtHome
  residualpower: 100
  meters:
    grid: grid1
    pv:
    - pv1
    - pv2
    battery: battery1

messaging:
  events:
    start: # charge start event
      title: Charge of {{.vehicleTitle}} started
      msg: |
        Wallbox {{.title}} started charging {{.vehicleTitle}} in {{ toString .mode | upper }} mode.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    stop: # charge stop event
      title: Charge of {{.vehicleTitle}} finished
      msg: |
        Wallbox {{.title}} finished charging {{.vehicleTitle}} 
        with {{round (divf .chargedEnergy 1000) 2 }} kWh in {{.chargeDuration}}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    connect: # vehicle connect event
      title: "{{.vehicleTitle}} connected on wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} connected on wallbox {{.title}} at {{round (divf .pvPower 1000) 2 }} kW PV.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    disconnect: # vehicle connected event
      title: "{{.vehicleTitle}} disconnected of wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} disconnected of wallbox {{.title}} after {{.connectedDuration}}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
```

### Log details

```text
Log:
lp-1 ] DEBUG 2025/03/08 09:16:44 charge power: 4591W
[site ] DEBUG 2025/03/08 09:16:44 grid power: -121W
[site ] DEBUG 2025/03/08 09:16:44 pv 2 power: 4889W
[site ] DEBUG 2025/03/08 09:16:44 battery 1 power: -1933W
[site ] DEBUG 2025/03/08 09:16:44 battery 1 soc: 40%
[site ] DEBUG 2025/03/08 09:16:44 pv 1 power: 2765W
[site ] DEBUG 2025/03/08 09:16:44 pv power: 7654W
[site ] ERROR 2025/03/08 09:16:44 accumulated solar yield: json: unsupported value: NaN
[site ] DEBUG 2025/03/08 09:16:44 battery has priority at soc 40% (< 100%)
[site ] DEBUG 2025/03/08 09:16:44 site power: -21W
[lp-1 ] DEBUG 2025/03/08 09:16:44 charger status: C
[lp-1 ] DEBUG 2025/03/08 09:16:44 soc estimated: 35.58% (vehicle: 34.00%)
[lp-1 ] DEBUG 2025/03/08 09:16:44 vehicle soc: 36%
[lp-1 ] DEBUG 2025/03/08 09:16:44 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/03/08 09:16:44 vehicle range: 46km
[lp-1 ] DEBUG 2025/03/08 09:16:44 pv charge current: 6.94A = 6.91A + 0.0304A (-21W @ 3p)
[lp-1 ] DEBUG 2025/03/08 09:16:44 max charge current: 6.94A
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.200.8

andig, 2025-03-08:
Das ist erstmal nicht schlimm, dürfte aber auch nicht passieren. Falls Du debuggen kannst- könntest Du mal schauen, wieso da NaN drin steht?

zinsius, 2025-03-08:
Danke, ich probiere es mal.

andig, 2025-03-08:
Das wäre grossartig. Du landest in site.go:554. Die Frage ist warum in `pvEnergy` anscheinend ein `NaN` drin steht und wo das her kommt.

andig, 2025-03-11:
@zinsius ich mache mal vorsichtig zu bis du einen Reproducer findest.

closed 2025-03-11 by andig

nelli81, 2025-03-18:
Hallo

Ich bekomme dieselbe Fehlermeldung. Ich finde die Meldung aufgrund der Häufigkeit doch rech störend. 

Kann ich bei der Fehlersuche irgendwie helfen? Falls ja, wie?

andig, 2025-03-18:
Wir brauchen ein Logfile vom auftreten dieser Werte.

andig, 2025-03-18:
Interessant wäre auch mal die Ausgabe von

    evcc settings get solarAccYield
    evcc settings get solarAccForecast

nelli81, 2025-03-18:
> Wir brauchen ein Logfile vom auftreten dieser Werte.

Habe das LogLevel von Site mal auf Trace gestellt und evcc neu gestartet. Wenn ich den Fehler erhalte, poste ich das Ergebnis

jgantenberg, 2025-03-18:
da bei mir der Fehler auch auftaucht, mische ich mich mal ein:
``` 
[ocpp ] TRACE 2025/03/18 09:25:07 recv 0311107101121150094: [3, "2938712695", {"status": "Accepted"}]
[ocpp ] TRACE 2025/03/18 09:25:07 recv 0311107101121150094: [2, "0e756b50-67d9-2de7-f3bf-bd23113bac65", "MeterValues", {"connectorId": 1, "meterValue": [{"timestamp": "2025-03-18T08:25:11.000Z", "sampledValue": [{"value": "0.0", "context": "Trigger", "measurand": "Power.Active.Import", "location": "Outlet", "unit": "W"}, {"value": "29.16", "context": "Trigger", "measurand": "Energy.Active.Import.Register", "location": "Outlet", "unit": "kWh"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L1", "location": "Outlet", "unit": "A"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L2", "location": "Outlet", "unit": "A"}, {"value": "0.00", "context": "Trigger", "measurand": "Current.Import", "phase": "L3", "location": "Outlet", "unit": "A"}, {"value": "238.88", "context": "Trigger", "measurand": "Voltage", "phase": "L1", "location": "Outlet", "unit": "V"}, {"value": "1.51", "context": "Trigger", "measurand": "Voltage", "phase": "L2", "location": "Outlet", "unit": "V"}, {"value": "0.24", "context": "Trigger", "measurand": "Voltage", "phase": "L3", "location": "Outlet", "unit": "V"}]}]}]
[ocpp ] TRACE 2025/03/18 09:25:07 send 0311107101121150094: [3,"0e756b50-67d9-2de7-f3bf-bd23113bac65",{}]
[mqtt ] TRACE 2025/03/18 09:25:09 recv esp-solar/sensor/growatt_battery_soc/state: '68'
[mqtt ] TRACE 2025/03/18 09:25:09 recv esp-solar/sensor/growatt_battery_discharge_total/state: '2755.6'
[mqtt ] TRACE 2025/03/18 09:25:09 recv esp-solar/sensor/growatt_local_load_total/state: '9987.6'
[mqtt ] TRACE 2025/03/18 09:25:12 recv esp-wb/sensor/sdm630_current_l1/state: '0.000'
[mqtt ] TRACE 2025/03/18 09:25:12 recv esp-wb/sensor/sdm630_current_l2/state: '0.000'
[mqtt ] TRACE 2025/03/18 09:25:12 recv esp-wb/sensor/sdm630_current_l3/state: '0.000'
[mqtt ] TRACE 2025/03/18 09:25:12 recv esp-wb/sensor/sdm630_total_power/state: '0.00'
[mqtt ] TRACE 2025/03/18 09:25:12 recv esp-wb/sensor/sdm630_import_active_energy/state: '1774.37'
[mqtt ] TRACE 2025/03/18 09:25:12 recv esp-solar/sensor/growatt_pv_power/state: '3543'
[mqtt ] TRACE 2025/03/18 09:25:12 recv esp-solar/sensor/growatt_combined_power_from_grid/state: '0'
[site ] DEBUG 2025/03/18 09:25:14 ----
[lp-1 ] DEBUG 2025/03/18 09:25:14 charge power: 0W
[lp-1 ] DEBUG 2025/03/18 09:25:14 charge currents: [0 0 0]A
[influx] TRACE 2025/03/18 09:25:14 write chargePower=map[value:0] (map[loadpoint:Garage])
[site ] DEBUG 2025/03/18 09:25:14 grid power: 0W
[influx] TRACE 2025/03/18 09:25:14 write chargeCurrents=map[l1:0 l2:0 l3:0] (map[loadpoint:Garage])
[influx] TRACE 2025/03/18 09:25:14 write gridPower=map[value:0] (map[])
[influx] TRACE 2025/03/18 09:25:14 write gridEnergy=map[value:9987.6] (map[])
[site ] DEBUG 2025/03/18 09:25:14 pv 1 power: 3543W
[influx] TRACE 2025/03/18 09:25:14 write pvPower=map[value:3543] (map[])
[influx] TRACE 2025/03/18 09:25:14 write pvEnergy=map[value:10639.9] (map[])
[influx] TRACE 2025/03/18 09:25:14 write pvPower=map[value:3543] (map[id:1])
[influx] TRACE 2025/03/18 09:25:14 write pvEnergy=map[value:10639.9] (map[id:1])
[site ] ERROR 2025/03/18 09:25:14 accumulated solar yield: json: unsupported value: NaN
[site ] DEBUG 2025/03/18 09:25:14 battery 1 power: -2850W
[site ] DEBUG 2025/03/18 09:25:14 battery 1 soc: 68%
[influx] TRACE 2025/03/18 09:25:14 write batteryCapacity=map[value:1] (map[])
[influx] TRACE 2025/03/18 09:25:14 write batterySoc=map[value:68] (map[])
[site ] DEBUG 2025/03/18 09:25:14 battery has priority at soc 68% (< 80%)
[site ] DEBUG 2025/03/18 09:25:14 site power: 500W
[influx] TRACE 2025/03/18 09:25:14 write batteryPower=map[value:-2850] (map[])
[influx] TRACE 2025/03/18 09:25:14 write batteryEnergy=map[value:2755.6] (map[])
[influx] TRACE 2025/03/18 09:25:14 write batteryPower=map[value:-2850] (map[id:1])
[influx] TRACE 2025/03/18 09:25:14 write batteryEnergy=map[value:2755.6] (map[id:1])
[lp-1 ] DEBUG 2025/03/18 09:25:14 charge total import: 1774.370kWh
[influx] TRACE 2025/03/18 09:25:14 write batteryCapacity=map[value:0] (map[id:1])
[influx] TRACE 2025/03/18 09:25:14 write batterySoc=map[value:68] (map[id:1])

```

mitten drin taucht es immer wieder mal auf.

zinsius, 2025-03-18:
Ich werde es auch versuchen, wenn mein EVCC wieder startet. Ich muss bei einem Neustart öfters meine Wallbox einmal kurz Stromlos machen, sonst findet EVCC die Box nicht. Startet dann natürlich auch nicht.

jgantenberg, 2025-03-18:
> Ich werde es auch versuchen, wenn mein EVCC wieder startet. Ich muss bei einem Neustart öfters meine Wallbox einmal kurz Stromlos machen, sonst findet EVCC die Box nicht. Startet dann natürlich auch nicht.

nutzt du das OCPP Binding in evcc? gib dem Charger mal mehr Zeit für den Connect. Meine Wallbox braucht auch deutlich mehr als die default eingestellten 15 Sekunden. Ich habe 2 min eingestellt, seitdem startet evcc eigentlich immer.

zinsius, 2025-03-18:
Ich gehe mal davon aus, dass es an meiner Wallbox liegt. Ist halt China (Entratek = DUOSIDA).

charger: [DUOSIDA] cannot create charger 'DUOSIDA': cannot create charger type 'template': cannot create charger type 'ocpp': timeout

jgantenberg, 2025-03-18:
ok, ich habe eine ähnliche WB vom deutschen Distributor, die heißt dann Entratek. Meine config sieht so aus:
``` 
chargers:
  - name: EntratekWallbox
    type: template
    template: ocpp
    connecttimeout: 2m
    timeout: 2m
    metervalues: -SoC,Power.Offered,Current.Offered

```

Damit startet die Kommunikation eigentlich immer.

andig, 2025-03-18:
Bleiben wir bitte beim Thema. Danke.

andig, 2025-03-21:
Ab morgen gibts im nightly besseres Logging. Falls die Meldung nochmal auftaucht bitte Fehler aus dem Logfile- diesmal gibts über mehrere Zeilen Details , woher/ welcher WR das mglw. verursacht.

jgantenberg, 2025-03-25:
Ich habe das nightly bisher nicht installiert, allerdings habe ich mir die MQTT Items angesehen. Ich habe zwei Werte die infrage kommen könnten, der eine ist 0, der andere 0.0. Der Fehler taucht gerade auf. Kann es mit dem float oder double Parser zu tun haben?
Ich würde gerne weiter debuggen, muss aber los...

jgantenberg, 2025-03-26:
Der Fehler taucht wieder auf:
Ich weiß jetzt auch, wie ich den Fehler reproduzieren kann: der WR wird mit ESPhome auf einem ESP32 ausgelesen und postet de Werte auf den MQTT Broker. Die Werte sind definitiv da, nur für den Moment, wenn der ESP rebootet wird sind noch keine Werte da und da kann es passieren, dass mal ein NaN ist. Seltsam ist, dass evcc danach die Meldung in jedem Zyklus ausgibt, obwohl längst gültige Werte vorhanden sind und auch in evcc angezeigt werden, als wenn der Fehlerstatus erhalten bleibt. 

```
Mar 26 14:18:39 evcc evcc[60]: [site  ] ERROR 2025/03/26 14:18:39 accumulated solar yield: json: unsupported value: NaN
Mar 26 14:18:39 evcc evcc[60]: [site  ] ERROR 2025/03/26 14:18:39 !! pv1: &{clock:0x54c1e50 updated:{wall:13974983605533739084 ext:3769179113689 loc:0x54b3cd8} meter:0x740cbb8 Accumulated:NaN}
```

jgantenberg, 2025-03-26:
im state steht 'nan'

andig, 2025-03-26:
We can do better, see https://github.com/evcc-io/evcc/pull/20137

jgantenberg, 2025-03-27:
getestet, funktioniert. Sehr gut, danke