evcc-io/evcc #19582: OCPP: VCHRGD Seven Pro (failed configuring availability: timeout)

Swiftnesses, 2025-03-08:
### Describe the bug

Hello!

Trying to set this up via OCPP for my VCHRGD Seven Pro (1.6 compliant) charger.

Trace logs included.

Any suggestions to get this working?

### Steps to reproduce

Simple YAML and start.

### Configuration details

```yaml
## minimal configuration example

log: trace

site:
  title: The Hyde

# trial token, valid until 2025-03-10
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3NDE2MDA4MDAsImlhdCI6MTc0MDMwNDgwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.nhueXl5zz7tdoTNEekJru7-pYhR2aY5IS9jfP6i6ZNw

chargers:
  - name: VP000458
    type: template
    template: ocpp
    # stationid: VP000458 # Station ID of the charging point. Only required if multiple OCPP charging stations are set up to assign them correctly. A single OCPP charging station can also be automatically assigned. Note: In exceptional cases, it may be necessary to manually append this ID to the OCPP URL of the charging station in the form ws://<evcc-address>:8887/<stationid>. Most charging stations automatically add the ID internally. (optional)
    # connector: 1 # Connector number for charging stations with multiple connectors. Counting starts at 1 for the first connector. (optional)
    # remotestart: # Only enable this option if there is no way to initiate transactions from the charger side! This is only the case if e.g. no RFID reader is available and charging processes would have to be released individually via app. Normally, the charger should always be configured at the device so that either an RFID card is used for activation or the charger is set to "Autostart", "Free Charging" or similar. First check the documentation and configuration possibilities of the charger, ask the manufacturer if necessary! (optional)
    # idtag: ***** # This option is only required in exceptional cases if the charger requires a specific token for accepting external transactions. (optional)
    # connecttimeout: 5m # optional
    # meterinterval: 10s # optional
    # metervalues: Energy.Active.Import.Register,Power.Active.Import,SoC,Current.Offered,Power.Offered,Current.Import,Voltage # optional
```

### Log details

```text
[cache ] TRACE 2025/03/08 12:55:46 vehicles: map[]
[site  ] WARN 2025/03/08 12:55:46 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[main  ] TRACE 2025/03/08 12:55:46 {"url":"https://api.github.com/repos/evcc-io/evcc/releases/204288833","assets_url":"https://api.github.com/repos/evcc-io/evcc/releases/204288833/assets","upload_url":"https://uploads.github.com/repos/evcc-io/evcc/releases/204288833/assets{?name,label}","html_url":"https://github.com/evcc-io/evcc/releases/tag/0.200.8","id":204288833,"author":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"node_id":"RE_kwDODX4bUs4MLTNB","tag_name":"0.200.8","target_commitish":"master","name":"0.200.8","draft":false,"prerelease":false,"created_at":"2025-03-06T18:49:44Z","published_at":"2025-03-07T06:41:37Z","assets":[{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/235372804","id":235372804,"node_id":"RA_kwDODX4bUs4OB4EE","name":"checksums.txt","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"content_type":"text/plain; charset=utf-8","state":"uploaded","size":754,"download_count":8,"created_at":"2025-03-07T06:53:58Z","updated_at":"2025-03-07T06:53:58Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.200.8/checksums.txt"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/235372800","id":235372800,"node_id":"RA_kwDODX4bUs4OB4EA","name":"evcc_0.200.8_amd64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"content_type":"application/vnd.debian.binary-package","state":"uploaded","size":23563368,"download_count":12,"created_at":"2025-03-07T06:53:57Z","updated_at":"2025-03-07T06:53:58Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.200.8/evcc_0.200.8_amd64.deb"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/235372798","id":235372798,"node_id":"RA_kwDODX4bUs4OB4D-","name":"evcc_0.200.8_arm64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"content_type":"application/vnd.debian.binary-package","state":"uploaded","size":21845696,"download_count":2,"created_at":"2025-03-07T06:53:56Z","updated_at":"2025-03-07T06:53:58Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.200.8/evcc_0.200.8_arm64.deb"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/235372799","id":235372799,"node_id":"RA_kwDODX4bUs4OB4D_","name":"evcc_0.200.8_armhf.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig [... cut ...] eriodic","measurand":"Voltage","phase":"L1","unit":"V"}]}]}]
[ocpp  ] TRACE 2025/03/08 12:56:37 send VP000458: [3,"135",{}]
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

Swiftnesses, 2025-03-08:
Logs:

```
[ocpp  ] TRACE 2025/03/08 12:58:30 send VP000458: [2,"620802018","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/08 12:58:31 recv VP000458: [3,"620802018",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/03/08 12:58:31 recv VP000458: [2,"147","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-03-08T11:58:29.521Z","sampledValue":[{"value":"0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},{"value":"0","context":"Other","measurand":"Current.Offered","unit":"A"},{"value":"0.00","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"243.2","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"}]}]}]
[ocpp  ] TRACE 2025/03/08 12:58:31 send VP000458: [3,"147",{}]
[ocpp  ] TRACE 2025/03/08 12:58:40 send VP000458: [2,"2009682669","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/08 12:58:41 recv VP000458: [3,"2009682669",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/03/08 12:58:41 recv VP000458: [2,"148","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-03-08T11:58:40.101Z","sampledValue":[{"value":"0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},{"value":"0","context":"Other","measurand":"Current.Offered","unit":"A"},{"value":"0.00","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"242.9","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"}]}]}]
[ocpp  ] TRACE 2025/03/08 12:58:41 send VP000458: [3,"148",{}]
[ocpp  ] TRACE 2025/03/08 12:58:50 send VP000458: [2,"2172209949","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/08 12:58:51 recv VP000458: [3,"2172209949",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/03/08 12:58:51 recv VP000458: [2,"149","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-03-08T11:58:50.131Z","sampledValue":[{"value":"0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},{"value":"0","context":"Other","measurand":"Current.Offered","unit":"A"},{"value":"0.00","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"242.6","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"}]}]}]
[ocpp  ] TRACE 2025/03/08 12:58:51 send VP000458: [3,"149",{}]
[ocpp  ] TRACE 2025/03/08 12:59:00 send VP000458: [2,"2809334895","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/08 12:59:01 recv VP000458: [3,"2809334895",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/03/08 12:59:01 recv VP000458: [2,"150","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-03-08T11:58:59.497Z","sampledValue":[{"value":"0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},{"value":"0","context":"Other","measurand":"Current.Offered","unit":"A"},{"value":"0.00","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"242.7","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"}]}]}]
[ocpp  ] TRACE 2025/03/08 12:59:01 send VP000458: [3,"150",{}]
[ocpp  ] TRACE 2025/03/08 12:59:10 send VP000458: [2,"4110843645","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/08 12:59:13 recv VP000458: [3,"4110843645",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/03/08 12:59:13 recv VP000458: [2,"151","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-03-08T11:59:12.233Z","sampledValue":[{"value":"0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},{"value":"0","context":"Other","measurand":"Current.Offered","unit":"A"},{"value":"0.00","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"242.3","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"}]}]}]
[ocpp  ] TRACE 2025/03/08 12:59:13 send VP000458: [3,"151",{}]
[ocpp  ] TRACE 2025/03/08 12:59:22 send VP000458: [2,"3046070474","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/03/08 12:59:25 recv VP000458: [3,"3046070474",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/03/08 12:59:25 recv VP000458: [2,"152","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-03-08T11:59:23.804Z","sampledValue":[{"value":"0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},{"value":"0","context":"Other","measurand":"Current.Offered","unit":"A"},{"value":"0.00","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"243.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"}]}]}]
[ocpp  ] TRACE 2025/03/08 12:59:25 send VP000458: [2,"938780211","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[... cut ...]
[ocpp  ] TRACE 2025/03/08 13:17:38 recv VP000458: [2,"253","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2025-03-08T12:17:36.362Z","sampledValue":[{"value":"0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"},{"value":"0","context":"Other","measurand":"Current.Offered","unit":"A"},{"value":"0.00","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"244.2","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"}]}]}]
[ocpp  ] TRACE 2025/03/08 13:17:38 send VP000458: [3,"253",{}]

```

Swiftnesses, 2025-03-08:
Seems that message might be nothing, and I stupidly didn't add the chargepoint to the config!

andig, 2025-03-08:
Maybe you could care to explain your problem instead of us finding one for you?

closed 2025-03-08 by andig

Swiftnesses, 2025-03-08:
@andig yes, it was user error it seems.

Q: I have times when I have cheap grid elec (no solar) - I've added zones to the tariff, how can I make it so the car only charges during these cheap times?