evcc-io/evcc #19596: Ich habe die Version 200.8 (nach downgrade auf 200.6 gleiches Verhalten) und seit zwei Tagen lädt das Auto nicht mit vollem Überschuss.

Ph1975, 2025-03-08:
Ich habe die Version 200.8 (nach downgrade auf 200.6 gleiches Verhalten) und seit zwei Tagen lädt das Auto nicht mit vollem Überschuss. 
Es ist mehr als Min Leistung. Hausakku ist auch voll. Wenn ich auf „schnell“ gehe lädt es mit maximalen 11 kw. Wenn ich im pv Modus im Auto Die Leistung erhöhe dann regelt der wieder runter und speist ein. 
[evcc-20250308-133009-debug.log](https://github.com/user-attachments/files/19143659/evcc-20250308-133009-debug.log)
[image]
[image]
[image]
[image]
[image]
[image]

_Originally posted by @MasterXDave in https://github.com/evcc-io/evcc/discussions/19588_

andig, 2025-03-08:
Ebenso config. Bitte keine Screenshots.

closed 2025-03-08 by andig

Ph1975, 2025-03-08:
In meinen TraceLogs sind Tokens z.B. von Tibber enthalten. Gibts nen einfachen Weg die vor dem Posten zu anonymisieren?

mfuchs1984, 2025-03-09:
In der ursprünglichen Diskussion https://github.com/evcc-io/evcc/discussions/19588#discussioncomment-12435414 hat sich herausgestellt, dass bei MasterXDave maxacpower falsch eingestellt war. Hast du das auch mal geprüft @Ph1975 ?

Ph1975, 2025-03-09:
Ich kann trace logs von meiner Situation zur Verfügung stellen plus obigen screenshot vorgestern  muss aber erst die Tokens und Passwörter entfernen. Ich habe leider keine debug logs.

Leider ist bei der Sonne das Auto einfach voll geladen, so dass ich das heute wohl nicht reproduzieren kann.

mfuchs1984, 2025-03-09:
Deine Konfiguration kannst du jederzeit teilen, da musst du nicht auf Sonne warten.

mfuchs1984, 2025-03-09:
>Was ich aber in jedem Fall gemacht habe ist für meinen SE Wechselrichter waren ebenso die AC Leistung bei 5k eingestellt er schafft aber laut Datenblatt 10k. Das habe ich aktualisiert.

Das dürfte das Problem lösen.

Ph1975, 2025-03-09:
also hier die Daten. Das Problem besteht weiterhin nach meiner Einschätzung:

[image]

[evcc-20250309-104402-debug.log](https://github.com/user-attachments/files/19148773/evcc-20250309-104402-debug.log)

mfuchs1984, 2025-03-09:
Du hast immernoch 5000 Watt eingestellt 

```
[site  ] DEBUG 2025/03/09 10:43:53 pv 1 power: 6427W
[site  ] DEBUG 2025/03/09 10:43:53 pv 1 excess DC: 1427W
```

Ph1975, 2025-03-09:
hm. Danke dafür! Hab es nun an beiden Konfig-Einstellungen geändert, davor war es nur unter "Netzzähler"  auf 10k eingestellt. Warum gibts die Einstellung zweimal?