evcc-io/evcc #19597: kein PV-Überschussladen SUNGROW SH10RT, Wallbox SUNGROW AC011E-01, OCPP-Anbindung

shellfisch, 2025-03-08:
### Describe the bug

PV-Überschussladen zeigt leider nicht die gewünschten Ergebnisse. Trotz genügend Strom vom Dach (10 kW) wird in der Schalterstellung PV konstant mit mehr als 10 kW ins Auto geladen bei gleichzeitigem Strombezug und Batteriebezug für die Autobatterie. Ich erwarte eher, dass der Ladepunkt runter geregelt wird, um den Strombezug zu minimieren bzw. ganz auszuschalten.
Es ändert sich z.B. auch nichts, wenn auf Min-PV bzw. Schnell eingestellt wird.

[evcc-20250308-153302-debug.log](https://github.com/user-attachments/files/19144276/evcc-20250308-153302-debug.log)

[evcc-20250308-153217-trace.log](https://github.com/user-attachments/files/19144277/evcc-20250308-153217-trace.log)

### Steps to reproduce

1. Neustart evcc bei angestecktem Auto
2.evcc startet für mich ohne erkennbaren Fehler
3.---
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  ocpp: trace

# unique installation id
plant: xxxxxxxxx
interval: 30s # control cycle interval

sponsortoken: xxxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

# mqtt message broker
mqtt:
  broker: 192.168.178.87:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user:
  password:

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.42  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 10s
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.42  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s
  name: pv2
#  maxAcPower: 10000
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.178.42  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 10s
  name: battery3

chargers:
- type: template
  template: ocpp 
  stationid: A229280xxxxx  
  name: wallbox5
  metervalues: -Current.Offered
  remotestart: true
  

vehicles:
- type: template
  name: Smart
  template: smart-hello 
  title: 'Smart #1'  
  user: xxxxx  
  password: xxxxx  
  capacity: 66
  icon: car  
#  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  vehicle: Smart # Referenz auf Standardfahrzeug
  mode: pv

tariffs:
  grid:
    type: tibber
    token: "xxxxxx" # access token

influx:
  url: http://192.168.178.86:8086
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  token: xxxxxxxxOAQzx4vhmONhre1ITBojUVGYRxsqKW60BQYA6fd4sBgOI4Gg==
  org: xxxx

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
```

### Log details

```text
siehe attached files
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.8

jove01, 2025-03-08:
Ich habe ein Verhalten festgestellt, dass möglicherweise hiermit zusammenhängt.

Ich habe ebenfalls den Sungrow  SH10RT mit dem Speicher SBR096.

Gestartet wurde das Laden mit niedrigem Tibber-Preis parallel zu der heutigen hohen PV-Leistung.
Meine Erwartung war, dass bis zur eingestellten Grenze 70% die PV-Leistung in die Batterie geht und nicht zum Laden verwendet wird. Aber der Speicher SoC blieb konstant.

[image]

Habe ich hier ein Verständnisproblem, dann bitte sorry.

mfuchs1984, 2025-03-08:
`[lp-1  ] DEBUG 2025/03/08 15:29:49 smart cost active: 0.25`

Das eingestellte Preislimit wurde unterschritten, deshalb mit maximaler Geschwindigkeit geladen.

mfuchs1984, 2025-03-08:
`#  maxAcPower: 10000` falls du mehr DC als AC Leistung hast, würde ich das wieder aktivieren, der Parameter funktioniert seit dem letzten Release nun.

shellfisch, 2025-03-09:
Ich konnte jetzt keinen für mich sichtbaren Unterschied mit aktiviertem maxACPower feststellen, d.h. auch damit bezieht er bei voller Sonneneinstrahlung bei 10 kW immer noch Netzstrom, statt den Netzstrom runter zu regeln und reinen PV-Strom zu konsumieren. 

Ich kann auch gerne noch zu dieser situation debug- und trace-logs schicken.

mfuchs1984, 2025-03-09:
Hast du denn dein Preislimit deaktiviert?

premultiply, 2025-03-09:
Kein Fehler für mich erkennbar. Preislimit aktiv.
Ich mach hier mal zu.

shellfisch, 2025-03-09:
Herzlichen Dank für den Hinweis, war mein Fehler, danke