evcc-io/evcc #19598: Solax Werte seit v0.200.2 falsch

slayerofmadness, 2025-03-08:
### Describe the bug

Seit dem Update auf v0.200.2 werden die Werte per Modbus falsch angezeigt . Die Werte sind totaler "Müll". Es werden selbst Nachts Erzeugerwerte von 50 - 60 kw gemeldet. Ein Betrieb ist so gar nicht mehr möglich :( . Modbus wird per LAN ausgelesen. Bis zur v0.200.1 ist alles super. Ein Downgrade stellt die richtige Übermittlung auch wieder her.

### Steps to reproduce

1. Update auf v0.200.2 und ff
2. Siehe Log Details
3. Keine sinnvolle Funktion von EVCC mehr :(

### Configuration details

```yaml
meters:
- type: template
  template: solax
  id: 1
  host: 192.168.178.76
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solax
  id: 1
  host: 192.168.178.76
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: solax
  id: 1
  host: 192.168.178.76
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
```

### Log details

```text
@slayerofmadness
slayerofmadness
6 hours ago
Author
Sponsor
Falsche Auslesung:

[lp-1 ] DEBUG 2025/03/08 10:20:15 charge power: 0W
[site ] DEBUG 2025/03/08 10:20:15 grid power: -2929W
[site ] DEBUG 2025/03/08 10:20:15 battery 1 power: 6W
[site ] DEBUG 2025/03/08 10:20:15 battery 1 soc: 99%
[site ] DEBUG 2025/03/08 10:20:15 pv 1 power: 64165W
[site ] DEBUG 2025/03/08 10:20:15 site power: -2923W
[lp-1 ] DEBUG 2025/03/08 10:20:15 charger status: A
[site ] DEBUG 2025/03/08 10:20:15 solar forecast: accumulated 0.4kWh, produced 5.1kWh, scale 14.3

Richtige Auslesung:

[lp-1 ] DEBUG 2025/03/08 10:19:54 charge power: 0W
[site ] DEBUG 2025/03/08 10:19:54 grid power: -2885W
[site ] DEBUG 2025/03/08 10:19:54 battery 1 power: 5W
[site ] DEBUG 2025/03/08 10:19:54 battery 1 soc: 99%
[site ] DEBUG 2025/03/08 10:19:54 pv 1 power: 4715W
[site ] DEBUG 2025/03/08 10:19:54 site power: -2880W
[lp-1 ] DEBUG 2025/03/08 10:19:54 charger status: A
[site ] DEBUG 2025/03/08 10:19:54 solar forecast: accumulated 0.3kWh, produced 4.8kWh, scale 14.4

Auslesezeit ist alle 20 Sekunden. Ungefähr jeder 10. Wert stimmt.

Fehler ist bei pv1
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.8

andig, 2025-03-08:
Klingt nach https://github.com/evcc-io/evcc/pull/18761. Was sagt

    evcc meter --log trace,db:error

/cc @spn91

slayerofmadness, 2025-03-08:
Die Ausgabe ist:

evcc meter --log trace,db:error
[main  ] INFO 2025/03/08 17:25:44 evcc 0.200.8
[main  ] INFO 2025/03/08 17:25:44 using config file: /etc/evcc.yaml
[main  ] FATAL 2025/03/08 17:25:44 
Found systemd service database at "/var/lib/evcc/evcc.db", evcc has been invoked with no explicit database path.
Running the same config with multiple databases can lead to expiring vehicle tokens.

If you want to use the existing service database run the following command:

evcc --database /var/lib/evcc/evcc.db

If you want to create a new user-space database run the following command:

evcc --database ~/.evcc/evcc.db

If you know what you're doing, you can skip the database check with the --ignore-db flag.

slayerofmadness, 2025-03-08:
Ich habe jetzt aktuell den Solax Meter raus genommen, die 1s refresh time war nur wegen der Fehlersuche

andig, 2025-03-08:
Bitte richtigen Befehl nehmen ;)

slayerofmadness, 2025-03-08:
Besser? :D 

[main  ] INFO 2025/03/08 17:35:24 evcc 0.200.8
[main  ] INFO 2025/03/08 17:35:24 using config file: /etc/evcc.yaml
[eebus ] INFO 2025/03/08 17:35:24 Local SKI: db5a2ef8aaebbd87ef3333df327d49158307f85c
[eebus ] DEBUG 2025/03/08 17:35:24 starting websocket server on :4712
[eebus ] ERROR 2025/03/08 17:35:24 websocket server error: listen tcp :4712: bind: address already in use
[eebus ] DEBUG 2025/03/08 17:35:24 mdns: announce
[eebus ] DEBUG 2025/03/08 17:35:24 mdns: using zeroconf
[discgy] TRACE 2025/03/08 17:35:24 GET https://api.inexogy.com/public/v1/meters
[discgy] TRACE 2025/03/08 17:35:24 GET https://api.inexogy.com/public/v1/meters
[discgy] TRACE 2025/03/08 17:35:24 [{"meterId":"a1bd6748ca7046268e27556170e29026","manufacturerId":"EMH","serialNumber":"10587423","fullSerialNumber":"***","printedFullSerialNumber":"1EMH0010587423","location":{"street":"Josef-Holzheu-Str.","streetNumber":"14","zip":"86932","city":"Lengenfeld","country":"DE"},"administrationNumber":"DE00002786932R1D00000001218888000","type":"EMH","measurementType":"ELECTRICITY","loadProfileType":"SLP","scalingFactor":1,"currentScalingFactor":1,"voltageScalingFactor":1,"internalMeters":1,"firstMeasurementTime":1617021816388,"lastMeasurementTime":1741455323390,"submeter":false},{"meterId":"8df54964152343c0b3c1a51702d30900","manufacturerId":"EMH","serialNumber":"10587424","fullSerialNumber":"1EMH0110587424","printedFullSerialNumber":"1EMH0010587424","location":{"street":"Josef-Holzheu-Str.","streetNumber":"14","zip":"86932","city":"Lengenfeld","country":"DE"},"administrationNumber":"DE00002786932R1D00000001218845000","type":"EMH","measurementType":"ELECTRICITY","loadProfileType":"SLP","scalingFactor":1,"currentScalingFactor":1,"voltageScalingFactor":1,"internalMeters":1,"firstMeasurementTime":1617021843260,"lastMeasurementTime":1741455322462,"submeter":false}]
[discgy] TRACE 2025/03/08 17:35:24 [{"meterId":"a1bd6748ca7046268e27556170e29026","manufacturerId":"EMH","serialNumber":"10587423","fullSerialNumber":"***","printedFullSerialNumber":"1EMH0010587423","location":{"street":"Josef-Holzheu-Str.","streetNumber":"14","zip":"86932","city":"Lengenfeld","country":"DE"},"administrationNumber":"DE00002786932R1D00000001218888000","type":"EMH","measurementType":"ELECTRICITY","loadProfileType":"SLP","scalingFactor":1,"currentScalingFactor":1,"voltageScalingFactor":1,"internalMeters":1,"firstMeasurementTime":1617021816388,"lastMeasurementTime":1741455323390,"submeter":false},{"meterId":"8df54964152343c0b3c1a51702d30900","manufacturerId":"EMH","serialNumber":"10587424","fullSerialNumber":"1EMH0110587424","printedFullSerialNumber":"1EMH0010587424","location":{"street":"Josef-Holzheu-Str.","streetNumber":"14","zip":"86932","city":"Lengenfeld","country":"DE"},"administrationNumber":"DE00002786932R1D00000001218845000","type":"EMH","measurementType":"ELECTRICITY","loadProfileType":"SLP","scalingFactor":1,"currentScalingFactor":1,"voltageScalingFactor":1,"internalMeters":1,"firstMeasurementTime":1617021843260,"lastMeasurementTime":1741455322462,"submeter":false}]
db:2
----
[db:2  ] TRACE 2025/03/08 17:35:24 modbus: send 00 01 00 00 00 06 01 04 00 16 00 01
[db:2  ] TRACE 2025/03/08 17:35:24 modbus: recv 00 01 00 00 00 05 01 04 02 f8 76
[db:2  ] TRACE 2025/03/08 17:35:24 modbus: send 00 02 00 00 00 06 01 04 00 1c 00 01
[db:2  ] TRACE 2025/03/08 17:35:24 modbus: recv 00 02 00 00 00 05 01 04 02 00 3e
Power:        1930W
Soc:          62%
Capacity:     6.5kWh
Controllable: true

db:3
----
[db:3  ] TRACE 2025/03/08 17:35:24 modbus: send 00 03 00 00 00 06 01 04 00 0a 00 01
[db:3  ] TRACE 2025/03/08 17:35:24 modbus: recv 00 03 00 00 00 05 01 04 02 00 14
[db:3  ] TRACE 2025/03/08 17:35:24 modbus: send 00 04 00 00 00 06 01 04 00 0b 00 01
[eebus ] DEBUG 2025/03/08 17:35:24 mdns: new - ski: 35ac774561189c420f1ded66e5bc571eb4e421d5 name: Elli-Wallbox-2145A1RTFC brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2145A1RTFC register: false host: wallbox-2145A1RTFC.local. port: 4712 addresses: [192.168.2.66]
[db:3  ] TRACE 2025/03/08 17:35:24 modbus: recv 00 04 00 00 00 05 01 04 02 00 15
[db:3  ] TRACE 2025/03/08 17:35:24 modbus: send 00 05 00 00 00 06 01 04 01 24 00 01
[db:3  ] TRACE 2025/03/08 17:35:24 modbus: recv 00 05 00 00 00 05 01 04 02 75 f1
[db:3  ] TRACE 2025/03/08 17:35:24 modbus: send 00 06 00 00 00 06 01 04 00 94 00 02
[db:3  ] TRACE 2025/03/08 17:35:25 modbus: recv 00 06 00 00 00 07 01 04 04 00 00 00 00
Power:  30234W
Energy: 0.0kWh

db:4
----
[db:4  ] TRACE 2025/03/08 17:35:25 modbus: send 00 07 00 00 00 06 01 04 00 16 00 01
[db:4  ] TRACE 2025/03/08 17:35:25 modbus: recv 00 07 00 00 00 05 01 04 02 f8 75
[db:4  ] TRACE 2025/03/08 17:35:25 modbus: send 00 08 00 00 00 06 01 04 00 1c 00 01
[db:4  ] TRACE 2025/03/08 17:35:25 modbus: recv 00 08 00 00 00 05 01 04 02 00 3e
Power:        1931W
Soc:          62%
Capacity:     6.5kWh
Controllable: true

db:8
----
[db:8  ] TRACE 2025/03/08 17:35:25 modbus: send 00 09 00 00 00 06 01 04 00 16 00 01
[db:8  ] TRACE 2025/03/08 17:35:25 modbus: recv 00 09 00 00 00 05 01 04 02 f8 75
[db:8  ] TRACE 2025/03/08 17:35:25 modbus: send 00 0a 00 00 00 06 01 04 00 1c 00 01
[db:8  ] TRACE 2025/03/08 17:35:25 modbus: recv 00 0a 00 00 00 05 01 04 02 00 3e
Power:        1931W
Soc:          62%
Capacity:     5.9kWh
Controllable: true

db:5
----
[discgy] TRACE 2025/03/08 17:35:25 GET https://api.inexogy.com/public/v1/last_reading?meterId=a1bd6748ca7046268e27556170e29026
[... cut ...] nergy":119186294000000}}
Power:  37W
Energy: 11918.6kWh

db:9
----
[discgy] TRACE 2025/03/08 17:35:25 GET https://api.inexogy.com/public/v1/last_reading?meterId=a1bd6748ca7046268e27556170e29026
[discgy] TRACE 2025/03/08 17:35:25 {"time":1741455323390,"values":{"energyOut":98772251000000,"energy2":0,"energy1":119186294000000,"energyOut1":98772251000000,"power":36900,"energyOut2":0,"energy":119186294000000}}
Power:  37W
Energy: 11918.6kWh

[eebus ] DEBUG 2025/03/08 17:35:25 mdns: stop announcement

andig, 2025-03-08:
Habs mal lesbar formatiert :)

slayerofmadness, 2025-03-08:
Evtl. gibt der output von dem Solax Meter (den ich wie bereits oben geschrieben) aktuell wegen Fehler deaktiviert habe noch Sinn für die Fehlersuche :):

[db:10 ] TRACE 2025/03/08 18:23:29 modbus: send 00 01 00 00 00 06 01 04 00 0a 00 01
[eebus ] DEBUG 2025/03/08 18:23:30 mdns: new - ski: 35ac774561189c420f1ded66e5bc571eb4e421d5 name: Elli-Wallbox-2145A1RTFC brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2145A1RTFC register: false host: wallbox-2145A1RTFC.local. port: 4712 addresses: [192.168.2.66]
[db:10 ] TRACE 2025/03/08 18:23:30 modbus: recv 00 01 00 00 00 05 01 04 02 00 00
[db:10 ] TRACE 2025/03/08 18:23:30 modbus: send 00 02 00 00 00 06 01 04 00 0b 00 01
[db:10 ] TRACE 2025/03/08 18:23:30 modbus: recv 00 02 00 00 00 05 01 04 02 00 00
[db:10 ] TRACE 2025/03/08 18:23:30 modbus: send 00 03 00 00 00 06 01 04 01 24 00 01
[db:10 ] TRACE 2025/03/08 18:23:30 modbus: recv 00 03 00 00 00 05 01 04 02 d0 f9
[db:10 ] TRACE 2025/03/08 18:23:30 modbus: send 00 04 00 00 00 06 01 04 00 94 00 02
[db:10 ] TRACE 2025/03/08 18:23:30 modbus: recv 00 04 00 00 00 07 01 04 04 00 00 00 00
Power:  53497W
Energy: 0.0kWh

spn91, 2025-03-09:
@andig ein revert würde bedeuten, dass bei mir dann „Müll“ angezeigt wird. Also eher keine Option. Ich schätze, dass der verwendete Inverter von @slayerofmadness auf der Modbus Adresse vom 3. MPPT einen Blödsinn liefert. Man müsste einen check einbauen, ob ein Inverter mit 2, oder 3 MPPT verwendet wird. 

Mit der neuen Generation die Solax heuer vorgestellt hat, bekommen auch die kleineren Inverter einen 3. MPPT, somit wird der „need“ nach einer Lösung nur größer. 

@slayerofmadness fordere von Solax mal die Modbus Dokumentation für Deinen Wechselrichter Typ an, damit wir sehen was sich auf der Adresse bei dir verbirgt.

andig, 2025-03-09:
@spn91 ich tendiere zu revert. Die Iplementierung für eine potenziell große Anzahl User kaputt zu machen ist keine Option. Wir können aber gerne einen zweiten Anlauf machen. 

@slayerofmadness um welchen WR handelt es sich denn exakt? Könntest Du dazu Remotezugang per SSH oder VPN für @spn91 einrichten so dass er beide WR vergleichen könnte?

slayerofmadness, 2025-03-09:
@andig es handelt sich um den SolaX Power X-Hybrid SK-SU5000 E Gen2. Eine SSH oder VPN Inspektion wäre grundsätzlich möglich. Ich persönlich stehe der Änderung aktuell auch kritisch gegenüber, viele werden noch nicht upgedatet haben und so noch nichts von diesem Problem wissen.

@spn91 ich kann gerne mal bei Solax anfragen. Auch wenn ich keinerlei Erfahrungen diesbezüglich habe. Ich verstehe auch deinen "Druck" dahinter jetzt endlich ordentliche Daten deines WR zu haben, aber für mich klingts schon so n bisschen nach... bei mir läufts jetzt sollen die anderen doch schauen wo sie bleiben. Eine Funktionalität aller Solax wäre mein persönliches Ziel für dieses Thema 😃

slayerofmadness, 2025-03-09:
[SolaxPower External Communication Protocol X1&X3 Hybrid -G4  ModbusRTU V3.14- public version.pdf](https://github.com/user-attachments/files/19148576/SolaxPower.External.Communication.Protocol.X1.X3.Hybrid.-G4.ModbusRTU.V3.14-.public.version.pdf)
Ich hatte mir die schon mal besorgt, da ich werte bei HA vergleichen wollte :)

andig, 2025-03-09:
Erstmal reverted. @spn91 wir können das asap wieder korrigieren. Siehst Du in der Modbus Doku neue Erkenntnisse?

spn91, 2025-03-09:
In der Modbus Doku kann ich leider keine Adresse mit 0x0124 (MPPT 3) finden, das Dokument wurde aber Ende 2021 das letzte Mal aktualisiert. Eventuell gibt es bereits eine neue Version davon.

Aber letztendlich ist es vermutlich egal, der Inverter liefert auf dieser Adresse blöderweise einen Wert zurück, ich würde also einen Check einbauen, der die 3. Adresse nicht abfragt wenn es kein Hybrid Gen5 Inverter mit 3 MPPTs ist.

Ich hab nur noch keine Idee ob sich sowas aktuell umsetzen lässt. Wäre natürlich hilfreich wenn es bereits ein Template gibt wo ähnliches gemacht wurde.

An-Wi, 2025-03-09:
Mal ganz blöd gefragt: Warum nicht einfach 2 Templates in evcc integrieren?
neben „solax“ bspw. „solax-mppt3“

Zur Frage der Unterscheidung der WR-Typen würde sich die Register 0x000E-0x0014 anbieten (laut verlinktet Doku) die den „ModuleName“ mit 14 Zeichen enthalten - wahrscheinlich das jeweilige Modell im Klartext.
Ich vermute, dass diese Register in alle Solax-WR-Modellen identisch belegt sind (so zumindest meine Erfahrung bisher in diesem Bereich).
Damit könnten die neuen 3-MPPT-Wechselrichter punktuell aufgenommen werden, für die das Auslesen von 3 Trackern notwendig ist. Für alle anderen WRs wird die vorherige Methodik verwendet.

andig, 2025-03-10:
@spn das ist doch einfach ein if statement und eine Option im Implate (mppt3: true).

@An-Wi gibts ein Mapping für Modulnamen und Mppt3? Ansonsten: kein zweites Template. Warum pflegen wenn die zu 99% identisch sind?!

An-Wi, 2025-03-10:
@andig 
Kann deinen Einwand zum 2. Template gut nachvollziehen. 
Die Option "mppt3" im Template ist genial. Damit kann das jeder einstellen, wie er das braucht. _Default_ sollte "mppt3: false" sein, um abwärtskompatibel zu bleiben.

Eine Auflistung der möglichen "ModuleName"s habe ich auf die Schnelle im Netz nicht gefunden. Auch in den verschiedenen Modbus-Dokus gibt es keinen Listen-Anhang mit möglichen Modellbezeichnungen. Das könntest du momentan nur als Ausgabe im "Debug"-Log einbauen und @spn91 und @slayerofmadness müssten dir dann ihre Logs schicken. Somit hättest du zumindest einen ersten Anhaltspunkt zum Mapping von 2 verschiedenen WR-Typen.

Die Template-Option mit einen entsprechenden Hinweis in der evcc-Doku finde ich dagegen viel charmanter, weil flexibler und zukunftssicherer. Die Hersteller sind bei den Modellbezeichnungen nie sehr konsequent und wechseln schnell mal das Schema, was für die MPPT3-Erkennung dann immer ein Nachpflegeaufwand nach sich ziehen würde.

spn91, 2025-03-10:
https://github.com/wills106/homeassistant-solax-modbus/blob/b213facca00c836a580c602e9eaec58b2c266394/custom_components/solax_modbus/plugin_solax.py

Im HA Addon findet die Zuordnung automatisch statt, das könnte man als Orientierung nutzen. 
Der Einfachheit halber, würde ich das aber tatsächlich als Option in der Config auswählbar machen. Ich werde das mal versuchen umzusetzen.

PR #19659 (2025-03-11): Solax: add mppt3 & active battery control
Reverts d194ce2783c70f0f4310aa7d0c2ee9d9ecb00f5d
Fixes #19598
Replace #18761

/cc @spn91
closed 2025-03-11 by andig