evcc-io/evcc #19600: Connect Felicity FLA48500 Battery

tomswager, 2025-03-08:
Hi. 
Is there a possibility to connect the battery Felicity FLA48500?

Best Tom 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

andig, 2025-03-09:
An api description is required

tomswager, 2025-03-10:
unfortunately i do not have any api description.
but thank you

andig, 2025-03-10:
I'm sorry but then there's nothing we can do. Feel free to ask the vendor to provide one!

closed 2025-03-10 by andig