evcc-io/evcc #19610: Growatt inverter incorrect readings

baraoresmungao, 2025-03-09:
### Describe the bug

Cannot read from my Growatt inverter 5000TL3-XH using the experimental configuration via UI.
System was reading corretly with a configuration via evcc.yaml, but now it does not work with the expermental UI config.
Using growatt-hybrid-tlxh it always reads both power and energy at 0, using the growatt-hybrid it reads always 1.7kW power and 0 kWh energy.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
I am not using evcc.yaml, using the UI to configure all.
```

### Log details

```text
Using growatt-hybrid-tlxh
[db:11 ] TRACE 2025/03/09 09:41:59 modbus: recv 01 db 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:41:59 modbus: send 01 dc 00 00 00 06 01 04 0b ed 00 02
[db:11 ] TRACE 2025/03/09 09:41:59 modbus: recv 01 dc 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: send 01 df 00 00 00 06 01 04 0b bd 00 02
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: recv 01 df 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: send 01 e0 00 00 00 06 01 04 0b c1 00 02
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: recv 01 e0 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: send 01 e1 00 00 00 06 01 04 0b c5 00 02
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: recv 01 e1 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: send 01 e2 00 00 00 06 01 04 0b c9 00 02
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: recv 01 e2 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: send 01 e3 00 00 00 06 01 04 0b ed 00 02
[db:11 ] TRACE 2025/03/09 09:42:02 modbus: recv 01 e3 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: send 01 ea 00 00 00 06 01 04 0b bd 00 02
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: recv 01 ea 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: send 01 eb 00 00 00 06 01 04 0b c1 00 02
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: recv 01 eb 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: send 01 ec 00 00 00 06 01 04 0b c5 00 02
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: recv 01 ec 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: send 01 ed 00 00 00 06 01 04 0b c9 00 02
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: recv 01 ed 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: send 01 ee 00 00 00 06 01 04 0b ed 00 02
[db:11 ] TRACE 2025/03/09 09:42:04 modbus: recv 01 ee 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: send 01 f1 00 00 00 06 01 04 0b bd 00 02
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: recv 01 f1 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: send 01 f2 00 00 00 06 01 04 0b c1 00 02
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: recv 01 f2 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: send 01 f3 00 00 00 06 01 04 0b c5 00 02
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: recv 01 f3 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: send 01 f4 00 00 00 06 01 04 0b c9 00 02
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: recv 01 f4 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: send 01 f5 00 00 00 06 01 04 0b ed 00 02
[db:11 ] TRACE 2025/03/09 09:42:07 modbus: recv 01 f5 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: send 01 fc 00 00 00 06 01 04 0b bd 00 02
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: recv 01 fc 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: send 01 fd 00 00 00 06 01 04 0b c1 00 02
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: recv 01 fd 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: send 01 fe 00 00 00 06 01 04 0b c5 00 02
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: recv 01 fe 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: send 01 ff 00 00 00 06 01 04 0b c9 00 02
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: recv 01 ff 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: send 02 00 00 00 00 06 01 04 0b ed 00 02
[db:11 ] TRACE 2025/03/09 09:42:09 modbus: recv 02 00 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:14 modbus: send 02 05 00 00 00 06 01 04 0b bd 00 02
[db:11 ] TRACE 2025/03/09 09:42:14 modbus: recv 02 05 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/03/09 09:42:14 modbus: send 02 06 00 00 00 06 01 04 0b c1 00 02
[db:11 ] TRACE 2025/03/09 09:42:14 modbus: recv 02 06 00 00 00 07 01 04 04 00 00 00 00
[... 106 lines cut ...]
[shelly] TRACE 2025/03/09 09:45:00 {"id":0,"on":false,"src":"evcc","method":"EMData.GetStatus"} -- {"id":0,"a_total_act_energy":488193.01,"a_total_act_ret_energy":12082.73,"b_total_act_energy":568134.28,"b_total_act_ret_energy":11831.89,"c_total_act_energy":403803.18,"c_total_act_ret_energy":38058.99,"total_act":1460130.48, "total_act_ret":61973.61}
[site ] DEBUG 2025/03/09 09:45:00 site power: 1424W
[etrel ] TRACE 2025/03/09 09:45:00 modbus: send 00 c6 00 00 00 06 01 04 00 00 00 01
[etrel ] TRACE 2025/03/09 09:45:00 modbus: recv 00 c6 00 00 00 05 01 04 02 00 02
[lp-1 ] DEBUG 2025/03/09 09:45:00 charger status: A
[etrel ] TRACE 2025/03/09 09:45:00 modbus: send 00 c7 00 00 00 06 01 04 00 04 00 02
[etrel ] TRACE 2025/03/09 09:45:01 modbus: recv 00 c7 00 00 00 07 01 04 04 00 00 00 00
[circuit-main] TRACE 2025/03/09 09:45:01 validate current: 5.23A + (0A -> 0A) <= 20A ok
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.8

andig, 2025-03-09:
Can oyu please compare with the working version from yaml? Show

    evcc meter —log trace

for working and non-working. We‘re only interested in the correct template.

baraoresmungao, 2025-03-09:
evcc does not work, I am using the homeassistant addon.

andig, 2025-03-09:
Closing until requested information available.

closed 2025-03-09 by andig

baraoresmungao, 2025-03-09:
I fixed the problem, it was a incorrect IP address.
Now with a fixed IP still I am having issues connecting

[site ] ERROR 2025/03/09 21:14:59 pv 1 power: read failed: dial tcp 192.168.1.192:502: connect: connection refused
[site ] DEBUG 2025/03/09 21:14:59 site power: 1409W
[lp-1 ] DEBUG 2025/03/09 21:14:59 charger status: A
[site ] DEBUG 2025/03/09 21:15:04 ----
[lp-1 ] DEBUG 2025/03/09 21:15:04 charge power: 0W
[lp-1 ] DEBUG 2025/03/09 21:15:04 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/03/09 21:15:04 power: 1406.3W
[circuit-main] DEBUG 2025/03/09 21:15:04 current: 3.56A
[site ] DEBUG 2025/03/09 21:15:04 grid power: 1406W
[site ] DEBUG 2025/03/09 21:15:04 grid powers: [250 791 365]W
[site ] DEBUG 2025/03/09 21:15:04 grid currents: [1.43 3.56 1.58]A
[site ] ERROR 2025/03/09 21:15:05 pv 1 power: read failed: dial tcp 192.168.1.192:502: connect: connection refused
[site ] DEBUG 2025/03/09 21:15:05 site power: 1406W
[lp-1 ] DEBUG 2025/03/09 21:15:05 charger status: A
[site ] DEBUG 2025/03/09 21:15:09 ----
[lp-1 ] DEBUG 2025/03/09 21:15:09 charge power: 0W
[lp-1 ] DEBUG 2025/03/09 21:15:09 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/03/09 21:15:09 power: 997.55W
[circuit-main] DEBUG 2025/03/09 21:15:09 current: 2.13A
[site ] DEBUG 2025/03/09 21:15:09 grid power: 998W
[site ] DEBUG 2025/03/09 21:15:09 grid powers: [247 386 365]W
[site ] DEBUG 2025/03/09 21:15:09 grid currents: [1.43 2.13 1.58]A
**[site ] ERROR 2025/03/09 21:15:10 pv 1 power: read failed: dial tcp 192.168.1.192:502: connect: connection refused**
[site ] DEBUG 2025/03/09 21:15:10 site power: 998W
[lp-1 ] DEBUG 2025/03/09 21:15:10 charger status: A