evcc-io/evcc #19611: Problem mit führenden Leerzeichen in Influx URL

Joe-md, 2025-03-09:
### Describe the bug

Einen sonnigen Sonntag Morgen!

Ich hatte das Problem dass nach Eintragung des Influx Dienstes in die UI mein Docker Container sich nicht mehr starten ließ.
Ich konnte das auf einen führendes Leerzeichen in der Influx URL zurückführen. Nachdem ich das Leerzeichen aus der DB gelöscht habe. startete der Container wieder problemlos.

EVCC Version: 0.200.8

VG aus dem Norden - Hannes

### Steps to reproduce

1 Anlegen einer Influx Config in der neuen UI-Konfig
2 Eintragen aller Daten
3 Einfügen eines Leerzeichen in der URL vor dem "http"
4 Neustart von evcc

```
Fehlerhafter Datensatz:
{"url":" http://192.168.x.x:8086","database":"EVCC","token":"MeinToken","org":"MeineOrg","user":"","password":"","insecure":true}

Korrigierter Datensatz:
{"url":" http://192.168.x.x:8086","database":"EVCC","token":"MeinToken","org":"MeineOrg","user":"","password":"","insecure":true}
```

### Configuration details

```yaml
/app # evcc -v
evcc version 0.200.8
/app # evcc dump --cfg
[main  ] INFO 2025/03/09 10:31:14 evcc 0.200.8
[main  ] INFO 2025/03/09 10:31:14 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/09 10:31:14 using sqlite database: /root/.evcc/evcc.db
[lp-2  ] WARN 2025/03/09 10:31:18 poll mode '{connected 10m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-3  ] WARN 2025/03/09 10:31:18 poll mode '{connected 10m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-4  ] WARN 2025/03/09 10:31:18 poll mode '{connected 10m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[gsi   ] ERROR 2025/03/09 10:31:22 unexpected status: 500 (Internal Server Error)
[main  ] ERROR 2025/03/09 10:31:22 creating tariff co2 failed: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': unexpected status: 500 (Internal Server Error)
[lp-1  ] DEBUG 2025/03/09 10:31:22 charge total import: 2239.812kWh
[lp-2  ] DEBUG 2025/03/09 10:31:22 charge total import: 388.604kWh
[lp-3  ] DEBUG 2025/03/09 10:31:22 charge total import: 246.908kWh
[lp-4  ] DEBUG 2025/03/09 10:31:23 charge total import: 368.726kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

# sponsor token, valid 1095 Tage ab 16.09.2024
sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: solax
  id: 1
  host: 192.168.x.x
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solax
  id: 1
  host: 192.168.x.x
  port: 502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: solax
  id: 2
  host: 192.168.x.x
  port: 502
  usage: battery
  modbus: tcpip
  capacity: 10.4
  name: battery1

chargers:
- type: template
  template: go-e-v3
  host: 192.168.x.x
  name: wallbox5
- type: template
  template: shelly
  name: ShellyHWR1
  heating: true
  host: 192.168.x.x # IP-Adresse oder Hostname
  user: *****
  password: *****
  channel: 0 # Optional
- type: template
  template: shelly
  name: ShellyHWR2
  heating: true
  host: 192.168.x.x # IP-Adresse oder Hostname
  user: *****
  password: *****
  channel: 0 # Optional
- type: template
  template: shelly
  name: ShellyHWR3
  heating: true
  host: 192.168.x.x # IP-Adresse oder Hostname
  user: *****
  password: *****
  channel: 0 # Optional

vehicles:
- type: template
  template: vw
  title: ID-4
  icon: car
  user: *****
  password: *****
  vin: *****
  capacity: 77
  phases: 3
  cache: 15m
  mode: pv
  minCurrent: 2
  maxCurrent: 16
  timeout: 10s
  name: ev4
- name: WWSpeicher_HWR
  type: custom
  title: WWSpeicher # display name for UI
  icon: waterheater
  capacity: 73 # battery capacity (kWh)
  phases: 1
  soc:
    source: http
    uri: http://192.168.x.x/rpc/Shelly.GetStatus
    jq: .["temperature:100"].tC

loadpoints:
- title: go-e-Carport
  charger: wallbox5
  vehicle: ev4
  mode: pv
  priority: 5
- title: WWShelly1 # display name for UI
  charger: ShellyHWR1 # charger
  vehicle: WWSpeicher_HWR
  mode: pv
  priority: 1
  mincurrent: 10 # 9A220V=1980W minimum charge current (default 6A)
  maxcurrent: 10 # 10A220V=2200W maximum charge current (default 16A)
  phases: 1 # ev phases (default 3)
  soc:
    poll:
      mode: connected
      interval: 10m
- title: WWShelly2 # display name for UI
  charger: ShellyHWR2 # charger
  vehicle: WWSpeicher_HWR
  mode: pv
  priority: 1
  mincurrent: 10 # 9A220V=1980W minimum charge current (default 6A)
  maxcurrent: 10 # 10A220V=2200W maximum charge current (default 16A)
  phases: 1 # ev phases (default 3)
  soc:
    poll:
      mode: connected
      interval: 10m
- title: WWShelly3 # display name for UI
  charger: ShellyHWR3 # charger
  vehicle: WWSpeicher_HWR
  mode: pv
  priority: 1
  mincurrent: 10 # 9A220V=1980W minimum charge current (default 6A)
  maxcurrent: 10 # 10A220V=2200W maximum charge current (default 16A)
  phases: 1 # ev phases (default 3)
  soc:
    poll:
      mode: connected
      interval: 10m
site:
  title: Kirchweg 11
  meters:
    grid: grid1
    pv:
    - pv1
    battery:
    - battery1
  residualPower: 100

Version: `0.200.8`
```

### Log details

```text
Konnte ich aufgrund der laufenden Restarts im Docker Container nicht finden. Konnte mich nicht anmelden...
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.8

andig, 2025-03-09:
@naltatis im UI vor API trimmen?

silent-pc, 2025-03-10:
Es hilft bei evcc Werte in Hochkommas ('WERT')zu setzen.
Auch wenn das in der Dokumentation so nicht vorgegeben wird funktioniert es dann.

Beim Grünstromindex passiert z. B. folgendes:
Gibt man hier:
ZIP = 07768
an, dann wird die führende Null der PLZ duch den Code entfernt und die Anfrage schlägt fehl.

Ändert man das so ab:
ZIP = '07768'
dann wird die korrekte PLZ 07768 übergeben und alles klappt so wie es soll.

naltatis, 2025-03-11:
@silent-pc im YAML Bereich (`evcc.yaml` und bspw. Tarife Yaml Editor) ist das korrekt. Hier gehts aber um um die Influx-DB UI. Die ist schon auf ein ordentliches Formular umgebaut. Da sollte kein spezielles Escaping notwendig sein.

> im UI vor API trimmen?

Ja, würd ich um UI sehen. An einigen Stellen (Sponsor Token pasten, Identifiers am Vehicle) machen wir das auch schon. Ich schau mal ob wir das einfach für alle Textfelder einbauen können.