evcc-io/evcc #19612: MQTT: Messages werden nicht mehr empfangen

ThomasWetzel, 2025-03-09:
### Describe the bug

Seit der Version 0.200.8 werden keine Nachrichten mehr von konfigurierten MQTT-Topics empfangen. In der Version 0.200.4 lief die Kommunikation per MQTT noch.

Die MQTT-Connection wird zwar hergestellt, es werden aber keine Nachrichten mehr empfangen.

Die Nachrichten können von einer anderen Anwendung vom Topic empfangen werden:

[image]

### Steps to reproduce

1. Konfigurieren von MQTT-Endpoint
2. Senden von neuem Ladestand
3. Überprüfen von Logs, ob Nachrichten ankommen
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

log: trace
levels:
  cache: error

plant: xxx

interval: 30s

meters:
  - type: template
    template: sma-home-manager
    usage: grid
    host: 10.2.201.120
    name: sunny

  - name: solar-se-pv
    type: template
    template: sma-hybrid
    usage: pv
    modbus: tcpip
    id: 3
    host: 10.2.201.106
    port: 502

  - name: solar-se-battery
    type: template
    template: sma-hybrid
    usage: battery
    modbus: tcpip
    id: 3
    host: 10.2.201.106
    port: 502

  - name: solar-tripower
    type: template
    template: sma-inverter-modbus
    usage: pv
    modbus: tcpip
    id: 3
    host: 10.2.201.6
    port: 502

loadpoints:
  - title: Kia
    charger: openwb-2
    vehicle: kia
    mode: pv

  - title: Superb
    charger: openwb-1
    vehicle: superb
    mode: pv

vehicles:
  - name: kia
    type: custom
    capacity: 64
    soc:
      source: mqtt
      topic: evcc/niro/soc
    range:
      source: mqtt
      topic: evcc/niro/range

  - name: superb
    type: custom
    capacity: 25
    soc:
      source: mqtt
      topic: evcc/superb/soc
    range:
      source: mqtt
      topic: evcc/superb/range

chargers:
  - name: openwb-2
    type: template
    template: openwb-2.0
    host: 10.2.201.11
    connector: 2
    modbus: tcpip
    id: 1
    port: 1502

  - name: openwb-1
    type: template
    template: openwb-2.0
    host: 10.2.201.11
    connector: 1
    modbus: tcpip
    id: 1
    port: 1502

site:
  title: Zu Hause
  meters:
    grid: sunny
    pv:
      - solar-se-pv
      - solar-tripower
    battery:
      - solar-se-battery

mqtt:
  broker: 192.168.2.218:1883
  topic: evcc
  clientid: evcc
  user: xxxx
  password: xxxx

tariffs:
  grid:
    type: template
    template: tibber
    token: xxxx
```

### Log details

```text
[main ] INFO 2025/03/09 11:47:31 evcc 0.200.8
[main ] INFO 2025/03/09 11:47:31 using config file: /etc/evcc.yaml
[db ] INFO 2025/03/09 11:47:31 using sqlite database: /root/.evcc/evcc.db
[mqtt ] INFO 2025/03/09 11:47:31 connecting evcc at tcp://192.168.2.218:1883
[mqtt ] DEBUG 2025/03/09 11:47:31 tcp://192.168.2.218:1883 connected
[main ] INFO 2025/03/09 11:47:31 listening at :7070
[lp-1 ] DEBUG 2025/03/09 11:47:32 charge total import: 4574.750kWh
[lp-2 ] DEBUG 2025/03/09 11:47:32 charge total import: 4941.305kWh
[site ] WARN 2025/03/09 11:47:34 vehicle 'Kia' does not support automatic detection
[site ] WARN 2025/03/09 11:47:34 vehicle 'Superb' does not support automatic detection
[site ] INFO 2025/03/09 11:47:34 site config:
[site ] INFO 2025/03/09 11:47:34 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/03/09 11:47:34 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2025/03/09 11:47:34 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2025/03/09 11:47:34 pv 2: power ✓ energy ✓ currents ✗
[site ] INFO 2025/03/09 11:47:34 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2025/03/09 11:47:34 vehicles:
[site ] INFO 2025/03/09 11:47:34 vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[site ] INFO 2025/03/09 11:47:34 vehicle 2: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[site ] INFO 2025/03/09 11:47:34 tariffs:
[site ] INFO 2025/03/09 11:47:34 grid: priceforecast
[site ] INFO 2025/03/09 11:47:34 feed-in: ✗
[site ] INFO 2025/03/09 11:47:34 co2: ✗
[site ] INFO 2025/03/09 11:47:34 solar: solar
[lp-1 ] INFO 2025/03/09 11:47:34 loadpoint 1:
[lp-1 ] INFO 2025/03/09 11:47:34 mode: pv
[lp-1 ] INFO 2025/03/09 11:47:34 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✓
[lp-1 ] INFO 2025/03/09 11:47:34 meters: charge ✓
[lp-1 ] INFO 2025/03/09 11:47:34 charge: power ✓ energy ✓ currents ✓
[lp-2 ] INFO 2025/03/09 11:47:34 loadpoint 2:
[lp-2 ] INFO 2025/03/09 11:47:34 mode: pv
[lp-2 ] INFO 2025/03/09 11:47:34 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✓
[lp-2 ] INFO 2025/03/09 11:47:34 meters: charge ✓
[lp-2 ] INFO 2025/03/09 11:47:34 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2025/03/09 11:47:34 set priority soc: 50
[lp-1 ] DEBUG 2025/03/09 11:47:34 phase timer inactive
[lp-1 ] DEBUG 2025/03/09 11:47:34 pv timer inactive
[lp-1 ] INFO 2025/03/09 11:47:34 vehicle updated: unknown -> Kia
[lp-2 ] DEBUG 2025/03/09 11:47:34 phase timer inactive
[lp-2 ] DEBUG 2025/03/09 11:47:34 pv timer inactive
[lp-2 ] INFO 2025/03/09 11:47:34 vehicle updated: unknown -> Superb
[site ] DEBUG 2025/03/09 11:47:34 ----
[lp-1 ] DEBUG 2025/03/09 11:47:34 charge power: 0W
[lp-1 ] DEBUG 2025/03/09 11:47:34 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/03/09 11:47:34 charge power: 0W
[lp-2 ] DEBUG 2025/03/09 11:47:34 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/09 11:47:34 grid power: -12325W
[site ] DEBUG 2025/03/09 11:47:34 grid powers: [-4050 -4090 -4186]W
[site ] DEBUG 2025/03/09 11:47:34 grid currents: [-17.9 -17.9 -18.3]A
[site ] DEBUG 2025/03/09 11:47:34 pv 2 power: 10273W
[site ] DEBUG 2025/03/09 11:47:34 pv 1 power: 3402W
[site ] DEBUG 2025/03/09 11:47:34 battery 1 power: 0W
[site ] DEBUG 2025/03/09 11:47:34 pv power: 13675W
[site ] DEBUG 2025/03/09 11:47:34 battery 1 soc: 100%
[site ] DEBUG 2025/03/09 11:47:34 site power: -12325W
[lp-1 ] DEBUG 2025/03/09 11:47:34 charge total import: 4574.750kWh
[lp-1 ] DEBUG 2025/03/09 11:47:34 charger status: B
[lp-1 ] DEBUG 2025/03/09 11:47:34 charge start energy: 4574.750kWh
[lp-1 ] INFO 2025/03/09 11:47:34 car connected
[lp-1 ] DEBUG 2025/03/09 11:47:34 pv timer elapse
[lp-1 ] DEBUG 2025/03/09 11:47:34 pv timer inactive
[lp-1 ] DEBUG 2025/03/09 11:47:34 charge total import: 4574.750kWh
[lp-1 ] ERROR 2025/03/09 11:47:34 vehicle soc: outdated
[lp-1 ] DEBUG 2025/03/09 11:47:34 switched phases: 3p
[lp-1 ] DEBUG 2025/03/09 11:47:34 max charge current: 16A
[lp-1 ] DEBUG 2025/03/09 11:47:34 charger enable
[lp-1 ] DEBUG 2025/03/09 11:47:34 wake-up timer: start
[site ] DEBUG 2025/03/09 11:48:04 ----
[lp-2 ] DEBUG 2025/03/09 11:48:04 charge power: 0W
[lp-2 ] DEBUG 2025/03/09 11:48:04 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/03/09 11:48:04 charge power: 0W
[lp-1 ] DEBUG 2025/03/09 11:48:04 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/09 11:48:04 grid power: -12604W
[site ] DEBUG 2025/03/09 11:48:04 grid powers: [-4108 -4176 -4321]W
[site ] DEBUG 2025/03/09 11:48:04 grid currents: [-18 -18.3 -18.8]A
[site ] DEBUG 2025/03/09 11:48:04 pv 2 power: 10451W
[site ] DEBUG 2025/03/09 11:48:04 battery 1 power: 0W
[site ] DEBUG 2025/03/09 11:48:04 pv 1 power: 3433W
[site ] DEBUG 2025/03/09 11:48:04 battery 1 soc: 100%
[site ] DEBUG 2025/03/09 11:48:04 pv power: 13884W
[site ] DEBUG 2025/03/09 11:48:04 site power: -12604W
[lp-2 ] DEBUG 2025/03/09 11:48:04 charge total import: 4941.305kWh
[lp-2 ] DEBUG 2025/03/09 11:48:04 charger status: B
[lp-2 ] DEBUG 2025/03/09 11:48:04 charge start energy: 4941.305kWh
[lp-2 ] INFO 2025/03/09 11:48:04 car connected
[lp-2 ] DEBUG 2025/03/09 11:48:04 pv timer elapse
[lp-2 ] DEBUG 2025/03/09 11:48:04 pv timer inactive
[lp-2 ] DEBUG 2025/03/09 11:48:04 charge total import: 4941.305kWh
[lp-2 ] ERROR 2025/03/09 11:48:04 vehicle soc: outdated
[lp-2 ] DEBUG 2025/03/09 11:48:04 available power 12604W > 4140W min 3p threshold
[lp-2 ] DEBUG 2025/03/09 11:48:04 phase scale3p in 0s
[lp-2 ] DEBUG 2025/03/09 11:48:04 switched phases: 3p
[lp-2 ] DEBUG 2025/03/09 11:48:04 phase timer inactive
[lp-2 ] DEBUG 2025/03/09 11:48:04 pv charge current: 18.3A = 0A + 18.3A (-12604W @ 3p)
[lp-2 ] DEBUG 2025/03/09 11:48:04 site power -12604W <= 0W enable threshold
[lp-2 ] DEBUG 2025/03/09 11:48:04 pv enable in 0s
[lp-2 ] DEBUG 2025/03/09 11:48:04 pv enable timer elapsed
[lp-2 ] DEBUG 2025/03/09 11:48:04 max charge current: 6A
[lp-2 ] DEBUG 2025/03/09 11:48:04 charger enable
[lp-2 ] DEBUG 2025/03/09 11:48:04 wake-up timer: start
[site ] DEBUG 2025/03/09 11:48:34 ----
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.8

VolkerK62, 2025-03-09:
```
[lp-1 ] ERROR 2025/03/09 11:47:34 vehicle soc: outdated
[lp-2 ] ERROR 2025/03/09 11:48:04 vehicle soc: outdated
```
Du hast kein `timeout` konfiguriert.

closed 2025-03-09 by andig

ThomasWetzel, 2025-03-09:
Vielen Dank. Das war's. :-)

In den vorigen Versionen war dies anscheinend nicht nötig.

andig, 2025-03-09:
Doch. Von Anbeginn an...