evcc-io/evcc #19632: Open-Meteo Tariff Doesn't Support Multiple Arrays Like Solcast

danielraffel, 2025-03-10:
### Describe the bug

#### **Summary**
I attempted to configure multiple **Open-Meteo** solar arrays in **EVCC**, following the approach documented for **Solcast**, where multiple solar forecasts are summed together. However, this does not seem to work with Open-Meteo. I encountered errors indicating that the configuration is not accepted.

---

### **Expected Behavior**
- **Either:** The documentation should explicitly state that Open-Meteo does not support multiple entries.
- **Or:** Open-Meteo should support multiple arrays like Solcast, allowing values to be summed.

---

### **Actual Behavior**
- The **Solcast approach** (using multiple `- type: template` entries) **does not work** with Open-Meteo.
- **Manually summing the `kwp` values** in a single Open-Meteo entry does work, but it's unclear if this is correct so I didn't bother to mention it.

---

### **Environment**
- **EVCC Version:** 0.200.8
- **Operating System:** Linux
- **Configuration Type:** Open-Meteo solar forecast

---

### **Request**
- Can Open-Meteo support multiple arrays like Solcast?
- If not, can the documentation clarify that only one Open-Meteo entry is allowed?

Thanks for looking into this!

### Steps to reproduce

1. **Tried defining multiple Open-Meteo solar arrays as a list**, similar to Solcast:
   ```yaml
   tariffs:
     solar:
       - type: template
         template: open-meteo
         lat: REDACTED
         lon: REDACTED
         dec: -28
         kwp: 5.6
         az: 90
       - type: template
         template: open-meteo
         lat: REDACTED
         lon: REDACTED
         dec: -27
         kwp: 2
         az: -90
       - type: template
         template: open-meteo
         lat: REDACTED
         lon: REDACTED
         dec: -11
         kwp: 3.2
         az: 0
   ```
   **Error received:**
   ```
   [tariff] ERROR 2025/03/09 17:34:00 invalid character 'U' looking for beginning of value
   [main ] ERROR 2025/03/09 17:34:00 creating tariff solar-template-0 failed: cannot create tariff type 'template': cannot create tariff type 'custom': invalid character 'U' looking for beginning of value
   ```

2. **Tried the single array format (one Open-Meteo entry), which worked fine**:
   ```yaml
   tariffs:
     solar:
       type: template
       template: open-meteo
       lat: REDACTED
       lon: REDACTED
       dec: -28
       kwp: 5.6
       az: 90
   ```
   **This works, but only for one array.**
...

### Configuration details

```yaml
Shared above
```

### Log details

```text
Shared above
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

TobiasHuber1980, 2025-03-10:
I think dec should be a value from 0 to 90. Not -xx

danielraffel, 2025-03-10:
Good catch. Just tried this:

```
  solar:
    - type: template
      template: open-meteo
      site: east
      interval: 1h
      lat: 37.X
      lon: -122.X
      dec: 28
      kwp: 5.6
      az: 90
    - type: template
      template: open-meteo
      site: west
      interval: 1h
      lat: 37.X
      lon: -122.X
      dec: 27
      kwp: 2
      az: -90
    - type: template
      template: open-meteo
      site: south
      interval: 1h
      lat: 37.X
      lon: -122.X
      dec: 11
      kwp: 3.2
      az: 0
```

Still getting errors:
```
[main ] FATAL 2025/03/09 22:27:28 tariff tariff [solar-template-2] cannot create tariff type 'template': invalid key: site
[main ] FATAL 2025/03/09 22:27:28 will attempt restart in: 15m0s
```