evcc-io/evcc #19636: switch phases: modbus: exception '3' (illegal data value), function '6'

Makis-EV, 2025-03-10:
### Describe the bug

Mit dem Update auf Version 0.2xx erhalte ich leider den Fehler:

Ich habe eine Alfen WallBox im Einsatz. Mit der Version 0.133 tritt der Fehler nicht auf.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Alfen EvePro
Victron
EVCC 0.208
Docker auf Synology
```

### Log details

```text
[site ] DEBUG 2025/03/10 10:00:35 ----
[lp-1 ] DEBUG 2025/03/10 10:00:35 charge power: 0W
[lp-1 ] DEBUG 2025/03/10 10:00:36 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/10 10:00:36 battery 1 power: -3641W
[site ] DEBUG 2025/03/10 10:00:36 grid power: -2887W
[site ] DEBUG 2025/03/10 10:00:36 battery 1 soc: 94%
[site ] DEBUG 2025/03/10 10:00:36 grid currents: [6.4 7.6 3.3]A
[site ] DEBUG 2025/03/10 10:00:36 pv 1 power: 8632W
[site ] DEBUG 2025/03/10 10:00:36 battery has priority at soc 94% (< 95%)
[site ] DEBUG 2025/03/10 10:00:36 site power: -2787W
[lp-1 ] DEBUG 2025/03/10 10:00:37 charge total import: 1699.092kWh
[lp-1 ] DEBUG 2025/03/10 10:00:37 charger status: B
[lp-1 ] ERROR 2025/03/10 10:00:37 switch phases: modbus: exception '3' (illegal data value), function '6'
[site ] DEBUG 2025/03/10 10:01:05 ----
[lp-1 ] DEBUG 2025/03/10 10:01:05 charge power: 0W
[lp-1 ] DEBUG 2025/03/10 10:01:06 charge currents: [0 0 0]A
[site ] DEBUG 2025/03/10 10:01:06 battery 1 power: -5218W
[site ] DEBUG 2025/03/10 10:01:06 grid power: -2656W
[site ] DEBUG 2025/03/10 10:01:06 battery 1 soc: 94%
[site ] DEBUG 2025/03/10 10:01:06 grid currents: [2.7 7.6 3.3]A
[site ] DEBUG 2025/03/10 10:01:06 pv 1 power: 8685W
[site ] DEBUG 2025/03/10 10:01:06 battery has priority at soc 94% (< 95%)
[site ] DEBUG 2025/03/10 10:01:06 site power: -2556W
[lp-1 ] DEBUG 2025/03/10 10:01:06 charge total import: 1699.092kWh
[lp-1 ] DEBUG 2025/03/10 10:01:06 charger status: B
[lp-1 ] ERROR 2025/03/10 10:01:07 switch phases: modbus: exception '3' (illegal data value), function '6'
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.208

andig, 2025-03-10:
What does

    evcc charger --log trace

say?

themakis, 2025-03-10:
How to run this command in docker?

premultiply, 2025-03-10:
> Mit dem Update auf Version 0.2xx erhalte ich leider den Fehler:
> Mit der Version 0.133 tritt der Fehler nicht auf.

Spannend, denn ich sehe keine relevanten Änderungen an der Implementierung.

Box mal neugestartet?

themakis, 2025-03-10:
Alles mehrfach neu gestartet. Sowohl WallBox als auch EVCC. Habe mittlerweile zwei Instanzen EVCC laufen. Die 0.20x die nicht lädt und eine 0,133 das ich laden kann.

andig, 2025-03-11:
Es wird 3p und Phasenumschaltung erkannt. Geändert hat sich nichts. 

> Die 0.20x die nicht lädt und eine 0,133 das ich laden kann.

Fehler gibts laut Log _nur_ bei der Phasenumschaltung. Jetzt ist die Frage was da anders sein soll. Also:

    evcc charger --log trace,db:error --phases 3

für beiden Versionen zum Vergleich.

Makis-EV, 2025-03-11:
0.208
[main  ] INFO 2025/03/11 12:51:03 evcc 0.200.8
[main  ] INFO 2025/03/11 12:51:03 using config file: /etc/evcc.yaml
[alfen ] TRACE 2025/03/11 12:51:04 modbus: send 00 01 00 00 00 06 01 03 01 32 00 06
[alfen ] TRACE 2025/03/11 12:51:04 modbus: recv 00 01 00 00 00 0f 01 03 0c 43 64 45 1e 43 62 f3 33 43 61 6e 14
[alfen ] DEBUG 2025/03/11 12:51:04 detected 3p alfen
[alfen ] TRACE 2025/03/11 12:51:04 modbus: send 00 02 00 00 00 06 01 06 04 bf 00 03
[alfen ] TRACE 2025/03/11 12:51:04 modbus: recv 00 02 00 00 00 03 01 86 03
[main  ] ERROR 2025/03/11 12:51:04 set phases: modbus: exception '3' (illegal data value), function '6'

0.133
[main  ] INFO 2025/03/11 12:52:40 evcc 0.133.0
[main  ] INFO 2025/03/11 12:52:40 using config file: /etc/evcc.yaml
[alfen ] TRACE 2025/03/11 12:52:41 modbus: send 00 01 00 00 00 06 01 03 01 32 00 06
[alfen ] TRACE 2025/03/11 12:52:41 modbus: recv 00 01 00 00 00 0f 01 03 0c 43 62 21 47 43 61 75 c2 43 60 2e 14
[alfen ] DEBUG 2025/03/11 12:52:41 detected 3p alfen
[alfen ] TRACE 2025/03/11 12:52:41 modbus: send 00 02 00 00 00 06 01 06 04 bf 00 03
[alfen ] TRACE 2025/03/11 12:52:41 modbus: recv 00 02 00 00 00 03 01 86 03
[main  ] ERROR 2025/03/11 12:52:41 set phases: modbus: exception '3' (illegal data value), function '6'

Wie ich gerade selbst sehe, ist der Fehler in beiden Logs. nur bei 0.208 wird er in der UI angezeigt.
Jedoch konnte ich feststellen, dass bei der 0.208 nicht lädt und bei 0.133 lädt.

andig, 2025-03-11:
Da es bei anderen Usern läuft kann ich nur vorschlagen, Alfen zu fragen warum die Phasenumschaltung nicht möglich ist. Soweit uns bekannt kann das jede Alfen Box. Die Load Balancing Lizenz hast Du ja sicher. 

Leider nichts mehr, das wir tun könnten...

Makis-EV, 2025-03-11:
wie gesagt, bei Version 0.133 klappt es ohne Probleme. Natürlich habe ich die Load Balancing Lizenz.
Habe eine Anfrage an Alfen gesendet.