evcc-io/evcc #19643: MQTT and influxdb error

phen0, 2025-03-10:
### Describe the bug

Since last or before last update I'm getting lots of errors and evcc is facing issues to connect via mqtt or to influxdb.
See error log below.

### Steps to reproduce

1. Update to v0.200.9
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  site: info
  lp-1: info
  warp: info

interval: 30s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: x.x.x.x
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1

- type: template
  template: sungrow-hybrid
  id: 1
  host: x.x.x.x
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2

- type: template
  template: sungrow-hybrid
  id: 1
  host: x.x.x.x
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- name: wallbox1
  type: template
  template: tinkerforge-warp
  host: x.x.x.x
  port: 1883
  topic: warp2/wallbox
  timeout: 30s
  energymanager: wem/26KN

vehicles:
- name: opel
[... 304 lines cut ...]
[mqtt  ] ERROR 2025/03/10 13:02:43 send: evcc/site/battery/1/energy: timeout
[mqtt  ] ERROR 2025/03/10 13:02:43 send: evcc/site/battery/1/power: timeout
[mqtt  ] ERROR 2025/03/10 13:02:43 send: evcc/site/battery/1/powers: timeout
[mqtt  ] ERROR 2025/03/10 13:02:43 send: evcc/site/battery/1/currents: timeout
[mqtt  ] ERROR 2025/03/10 13:02:43 send: evcc/site/battery/1/excessDCPower: timeout
[mqtt  ] ERROR 2025/03/10 13:02:43 send: evcc/site/battery/1/soc: timeout
[mqtt  ] ERROR 2025/03/10 13:02:43 send: evcc/site/battery/1/controllable: timeout
[mqtt  ] ERROR 2025/03/10 13:02:56 tcp://10.0.55.85:1883 connection lost: pingresp not received, disconnecting
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.9

andig, 2025-03-10:
> NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.

closed 2025-03-10 by andig

phen0, 2025-03-12:
Hi @andig

you were right. It only happens in docker but how come? I didn't change the config at all just updated the container.
Maybe you've a hint for me how to troubleshoot?
Many thanks