evcc-io/evcc #19645: startet nicht mehr nach update 0.200.9

dz1984sg, 2025-03-10:
### Describe the bug

Es sind 3x SolaX Ultra verbaut, vor dem update lief es 1A

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
nur über die ui konfiguriert ...
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/10 13:50:12 evcc 0.200.9
[main  ] INFO 2025/03/10 13:50:12 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/10 13:50:12 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/03/10 13:50:13 connecting evcc-360616086 at tcp://192.168.178.48:1883
[eebus ] INFO 2025/03/10 13:50:13 Local SKI: 98d9e2252a8bce9d0e287fd863e72af7dd5a4c8f
[main  ] INFO 2025/03/10 13:50:13 listening at :7070
[main  ] FATAL 2025/03/10 13:50:14 meter [db:8] cannot create meter 'db:8': cannot create meter type 'template': invalid key: storageunit
[main  ] FATAL 2025/03/10 13:50:14 will attempt restart in: 15m0s
[eebus ] ERROR 2025/03/10 13:53:14 websocket server error: http: Server closed
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/10 13:53:14 evcc 0.200.9
[main  ] INFO 2025/03/10 13:53:14 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/10 13:53:14 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/03/10 13:53:15 connecting evcc-1219506709 at tcp://192.168.178.48:1883
[eebus ] INFO 2025/03/10 13:53:15 Local SKI: 98d9e2252a8bce9d0e287fd863e72af7dd5a4c8f
[main  ] INFO 2025/03/10 13:53:15 listening at :7070
[main  ] FATAL 2025/03/10 13:53:16 meter [db:12] cannot create meter 'db:12': cannot create meter type 'template': invalid key: storageunit
[main  ] FATAL 2025/03/10 13:53:16 will attempt restart in: 15m0s
[eebus ] ERROR 2025/03/10 13:56:16 websocket server error: http: Server closed
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/10 13:56:17 evcc 0.200.9
[main  ] INFO 2025/03/10 13:56:17 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/10 13:56:17 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/03/10 13:56:17 connecting evcc-1354524150 at tcp://192.168.178.48:1883
[eebus ] INFO 2025/03/10 13:56:17 Local SKI: 98d9e2252a8bce9d0e287fd863e72af7dd5a4c8f
[main  ] INFO 2025/03/10 13:56:17 listening at :7070
[main  ] FATAL 2025/03/10 13:56:18 meter [db:9] cannot create meter 'db:9': cannot create meter type 'template': invalid key: storageunit
[main  ] FATAL 2025/03/10 13:56:18 will attempt restart in: 15m0s
[eebus ] ERROR 2025/03/10 13:59:18 websocket server error: http: Server closed
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/10 13:59:19 evcc 0.200.9
[main  ] INFO 2025/03/10 13:59:19 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/10 13:59:19 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/03/10 13:59:19 connecting evcc-359552610 at tcp://192.168.178.48:1883
[eebus ] INFO 2025/03/10 13:59:19 Local SKI: 98d9e2252a8bce9d0e287fd863e72af7dd5a4c8f
[main  ] INFO 2025/03/10 13:59:19 listening at :7070
[main  ] FATAL 2025/03/10 13:59:20 meter [db:9] cannot create meter 'db:9': cannot create meter type 'template': invalid key: storageunit
[main  ] FATAL 2025/03/10 13:59:20 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.2000.9

andig, 2025-03-10:
Sorry for the breakage. Will fix this in nightly.

andig, 2025-03-10:
Closed in https://github.com/evcc-io/evcc/commit/d194ce2783c70f0f4310aa7d0c2ee9d9ecb00f5d

closed 2025-03-10 by andig

dz1984sg, 2025-03-13:
> Closed in d194ce2

shouldn't the update already be coming?

why, have you deactivated the 2nd battery in the settings?

andig, 2025-03-13:
It's fixed in the nighly version