evcc-io/evcc #19648: SG Wärmepumpe schaltet nicht ab bei zu wenig Überschuss im Modus PV

nicx, 2025-03-10:
### Describe the bug

Ich habe meine SG fähige Wärmepumpe (Daikin Altherma EchO2) wir hier beschrieben integriert: https://docs.evcc.io/docs/devices/heating

Die Wärmepumpe lässt sich steuern, d.h. wenn ich den Loadpoint auf "Schnell" schalte wird der SG mode auf "an" gesetzt. Schalte ich den Loadpoint auf "Aus", wird der SG mode wieder in den Normalzustand versetzt. Manuell funktioniert also alles korrekt.

Wenn ich die WP nun im Modus PV betreibe, wird der SG boost mode bei erreichen der eingestellten Schwelle für min. Ladestrom (bei mir per UI konfiguriert auf 4A, also ca. 2,8kW Überschuss) aktiviert. Allerdings wird der SG mode nicht mehr deaktiviert, falls der Überschuss unter diese 4A fällt.

Mach/verstehe ich etwas falsch oder gibt es hier noch einen Bug? :)

### Steps to reproduce

-

### Configuration details

```yaml
### siehe https://docs.evcc.io/docs/devices/heating:

loadpoints:
  - title: Wärmepumpe
    charger: heatpump_control
    meter: heatpump_power

meters:
  - name: heatpump_power
    type: custom
    power:
      source: mqtt
      topic: P1P2/P/meter/U/9/Electricity_Power

chargers:
  - name: heatpump_control
    type: sgready
    setmode:
      source: switch
      switch:
      - case: 1 # normal
        set:
          source: const
          value: false
          set:
            source: mqtt
            topic: espaltherma/sg/set
            payload: 0
      - case: 2 # boost
        set:
          source: const
          value: true
          set:
            source: mqtt
            topic: espaltherma/sg/set
            payload: 2
      - case: 3 # dimm (not supported, yet)
        set:
          source: error
          error: ErrNotAvailable
    temp:
     source: mqtt
     topic: P1P2/P/P1P2MQTT/bridge0/T/1/Temperature_R5T_DHW_Tank
     timeout: 60m
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.200.9

andig, 2025-03-10:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-03-10 by andig

nicx, 2025-03-11:
@andig leider ist das Logfile man nicht aussagekräftig (daher hatte ich es gar nicht erst angehängt. Habe hier jetzt mal eines angehängt.

Ist denn meine Konfig soweit schlüssig und mein Verständnis korrekt? Also müsste beim stoppen des "Loadpoints" der case 1 wieder greifen? oder müsste ich hier bzgl. Konfiguration noch etwas anpassen?

Logfile:
```
[lp-1 ] DEBUG 2025/03/11 13:10:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:10:59 charger status: C
[lp-1 ] DEBUG 2025/03/11 13:10:59 pv charge current: 0.816A = 1A + -0.184A (127W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:10:59 projected site power 127W >= 0W disable threshold
[lp-1 ] DEBUG 2025/03/11 13:10:59 pv disable in 0s
[lp-1 ] DEBUG 2025/03/11 13:10:59 pv disable timer elapsed
[lp-1 ] DEBUG 2025/03/11 13:10:59 charger disable
[lp-1 ] DEBUG 2025/03/11 13:10:59 wake-up timer: stop
[lp-1 ] DEBUG 2025/03/11 13:11:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:11:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:12:29 charge power: 529W
[lp-1 ] DEBUG 2025/03/11 13:12:29 charger status: B
[lp-1 ] INFO 2025/03/11 13:12:29 stop charging <-
[lp-1 ] DEBUG 2025/03/11 13:12:29 pv timer reset
[lp-1 ] DEBUG 2025/03/11 13:12:29 pv timer inactive
[lp-1 ] DEBUG 2025/03/11 13:12:29 pv charge current: 0A = 0A + -0.203A (140W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:12:59 charge power: 529W
[lp-1 ] DEBUG 2025/03/11 13:13:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:13:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:13:59 charger status: B
[lp-1 ] DEBUG 2025/03/11 13:13:59 pv charge current: 0A = 0A + -0.332A (229W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:14:29 charge power: 529W
[lp-1 ] DEBUG 2025/03/11 13:14:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:15:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:15:29 charger status: B
[lp-1 ] DEBUG 2025/03/11 13:15:29 pv charge current: 0A = 0A + -0.194A (134W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:15:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:16:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:16:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:16:59 charger status: B
[lp-1 ] DEBUG 2025/03/11 13:16:59 pv charge current: 0A = 0A + -0.584A (403W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:17:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:17:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:18:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:18:29 charger status: B
[lp-1 ] DEBUG 2025/03/11 13:18:29 pv charge current: 0A = 0A + -0.191A (132W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:18:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:19:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:19:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:19:59 charger status: B
[lp-1 ] DEBUG 2025/03/11 13:19:59 pv charge current: 0A = 0A + -0.154A (106W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:20:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:20:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:21:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:21:29 charger status: B
[lp-1 ] DEBUG 2025/03/11 13:21:29 pv charge current: 4.38A = 0A + 4.38A (-3023W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:21:29 site power -3023W <= 0W enable threshold
[lp-1 ] DEBUG 2025/03/11 13:21:29 pv enable timer start: 1m0s
[lp-1 ] DEBUG 2025/03/11 13:21:29 pv enable in 1m0s
[lp-1 ] DEBUG 2025/03/11 13:21:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:22:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:22:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:22:59 charger status: B
[lp-1 ] DEBUG 2025/03/11 13:22:59 pv charge current: 2.21A = 0A + 2.21A (-1523W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:22:59 site power -1523W <= 0W enable threshold
[lp-1 ] DEBUG 2025/03/11 13:22:59 pv enable in 0s
[lp-1 ] DEBUG 2025/03/11 13:22:59 pv enable timer elapsed
[lp-1 ] DEBUG 2025/03/11 13:22:59 max charge current: 1A
[lp-1 ] DEBUG 2025/03/11 13:22:59 charger enable
[lp-1 ] DEBUG 2025/03/11 13:22:59 wake-up timer: start
[lp-1 ] DEBUG 2025/03/11 13:23:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:23:59 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:24:29 charge power: 414W
[lp-1 ] DEBUG 2025/03/11 13:24:29 charger status: C
[... 79 lines cut ...]
[lp-1 ] DEBUG 2025/03/11 13:45:29 charger status: C
[lp-1 ] DEBUG 2025/03/11 13:45:29 pv charge current: 17.5A = 16A + 1.46A (-1009W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:45:59 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:46:29 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:46:59 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:46:59 charger status: C
[lp-1 ] DEBUG 2025/03/11 13:46:59 pv charge current: 17.5A = 16A + 1.5A (-1037W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:47:29 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:47:59 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:48:29 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:48:29 charger status: C
[lp-1 ] DEBUG 2025/03/11 13:48:29 pv charge current: 17.4A = 16A + 1.41A (-974W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:48:51 set min current: 6
[lp-1 ] DEBUG 2025/03/11 13:48:59 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:49:29 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:49:59 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:49:59 charger status: C
[lp-1 ] DEBUG 2025/03/11 13:49:59 pv charge current: 16.9A = 16A + 0.867A (-598W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:50:29 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:50:59 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:51:29 charge power: 0W
[lp-1 ] DEBUG 2025/03/11 13:51:29 charger status: C
[lp-1 ] DEBUG 2025/03/11 13:51:29 pv charge current: 17.1A = 16A + 1.06A (-733W @ 3p)
[lp-1 ] DEBUG 2025/03/11 13:51:59 charge power: 0W
```

andig, 2025-03-11:
> Allerdings wird der SG mode nicht mehr deaktiviert, falls der Überschuss unter diese 4A fällt.

Das Log sagt nichts darüber, ist daher nutzlos.

nicx, 2025-03-11:
@andig hier im log hab ich etwas zum deaktivieren gefunden ("stop charching"). Wie geschrieben: der lp-1 wird aber nicht gestoppt (respektive der sg mode geändert).

was kann ich nebst Konfiguration und log noch liefern? Ein anderes log? Ein bisschen mehr Hilfe wäre echt toll.

```
[circuit-wallboxen] DEBUG 2025/03/11 13:09:29 power: 0W
[circuit-wallboxen] DEBUG 2025/03/11 13:09:29 current: 0A
[circuit-main] DEBUG 2025/03/11 13:09:29 power: -2W
[circuit-main] DEBUG 2025/03/11 13:09:29 current: 1.87A
[lpc   ] DEBUG 2025/03/11 13:09:29 power: -2W
[lpc   ] DEBUG 2025/03/11 13:09:29 current: 1.87A
[site  ] DEBUG 2025/03/11 13:09:29 grid power: -2W
[site  ] DEBUG 2025/03/11 13:09:29 battery 1 power: -81W
[site  ] DEBUG 2025/03/11 13:09:29 pv 1 power: 1801W
[site  ] DEBUG 2025/03/11 13:09:29 grid currents: [1.87 -1.5 -1.7]A
[site  ] DEBUG 2025/03/11 13:09:29 battery 1 soc: 98%
[site  ] DEBUG 2025/03/11 13:09:29 battery has priority at soc 98% (< 100%)
[site  ] DEBUG 2025/03/11 13:09:29 site power: 98W
[lp-1  ] DEBUG 2025/03/11 13:09:29 charger status: C
[lp-1  ] DEBUG 2025/03/11 13:09:29 pv charge current: 0.858A = 1A + -0.142A (98W @ 3p)
[lp-1  ] DEBUG 2025/03/11 13:09:29 projected site power 98W >= 0W disable threshold
[lp-1  ] DEBUG 2025/03/11 13:09:29 pv disable in 0s
[lp-1  ] DEBUG 2025/03/11 13:09:29 pv disable timer remaining: 0s
[site  ] DEBUG 2025/03/11 13:09:29 solar forecast: accumulated 14.317kWh, produced 8.670kWh, scale 0.606
[site  ] DEBUG 2025/03/11 13:09:59 ----
[lp-3  ] DEBUG 2025/03/11 13:09:59 charge power: 0W
[lp-3  ] DEBUG 2025/03/11 13:09:59 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/03/11 13:09:59 charge power: 414W
[lp-2  ] DEBUG 2025/03/11 13:09:59 charge power: 0W
[lp-2  ] DEBUG 2025/03/11 13:09:59 charge currents: [0 0 0]A
[circuit-wallboxen] DEBUG 2025/03/11 13:09:59 power: 0W
[circuit-wallboxen] DEBUG 2025/03/11 13:09:59 current: 0A
[circuit-main] DEBUG 2025/03/11 13:09:59 power: -0W
[circuit-main] DEBUG 2025/03/11 13:09:59 current: 1.86A
[lpc   ] DEBUG 2025/03/11 13:09:59 power: -0W
[lpc   ] DEBUG 2025/03/11 13:09:59 current: 1.86A
[site  ] DEBUG 2025/03/11 13:09:59 grid power: -0W
[site  ] DEBUG 2025/03/11 13:09:59 battery 1 power: -148W
[site  ] DEBUG 2025/03/11 13:09:59 pv 1 power: 1921W
[site  ] DEBUG 2025/03/11 13:09:59 grid currents: [1.86 -1.49 -1.69]A
[site  ] DEBUG 2025/03/11 13:09:59 battery 1 soc: 98%
[site  ] DEBUG 2025/03/11 13:09:59 battery has priority at soc 98% (< 100%)
[site  ] DEBUG 2025/03/11 13:09:59 site power: 100W
[lp-2  ] DEBUG 2025/03/11 13:09:59 charge total import: 7205.741kWh
[lp-2  ] DEBUG 2025/03/11 13:09:59 charger status: A
[site  ] DEBUG 2025/03/11 13:09:59 solar forecast: accumulated 14.336kWh, produced 8.690kWh, scale 0.606
[site  ] DEBUG 2025/03/11 13:10:29 ----
[lp-3  ] DEBUG 2025/03/11 13:10:29 charge power: 0W
[lp-3  ] DEBUG 2025/03/11 13:10:29 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/03/11 13:10:29 charge power: 414W
[lp-2  ] DEBUG 2025/03/11 13:10:29 charge power: 0W
[lp-2  ] DEBUG 2025/03/11 13:10:29 charge currents: [0 0 0]A
[circuit-wallboxen] DEBUG 2025/03/11 13:10:29 power: 0W
[circuit-wallboxen] DEBUG 2025/03/11 13:10:29 current: 0A
[circuit-main] DEBUG 2025/03/11 13:10:29 power: -9W
[circuit-main] DEBUG 2025/03/11 13:10:29 current: 1.85A
[lpc   ] DEBUG 2025/03/11 13:10:29 power: -9W
[lpc   ] DEBUG 2025/03/11 13:10:29 current: 1.85A
[site  ] DEBUG 2025/03/11 13:10:29 grid power: -9W
[site  ] DEBUG 2025/03/11 13:10:29 battery 1 power: -276W
[site  ] DEBUG 2025/03/11 13:10:29 pv 1 power: 2036W
[site  ] DEBUG 2025/03/11 13:10:29 grid currents: [1.85 -1.5 -1.69]A
[site  ] DEBUG 2025/03/11 13:10:29 battery 1 soc: 98%
[site  ] DEBUG 2025/03/11 13:10:29 battery has priority at soc 98% (< 100%)
[site  ] DEBUG 2025/03/11 13:10:29 site power: 91W
[lp-3  ] DEBUG 2025/03/11 13:10:29 charge total import: 215.598kWh
[lp-3  ] DEBUG 2025/03/11 13:10:29 charger status: A
[site  ] DEBUG 2025/03/11 13:10:29 solar forecast: accumulated 14.355kWh, produced 8.710kWh, scale 0.607
[site  ] DEBUG 2025/03/11 13:10:59 ----
[lp-3  ] DEBUG 2025/03/11 13:10:59 charge power: 0W
[... 138 lines cut ...]
[site  ] DEBUG 2025/03/11 13:13:31 solar forecast: accumulated 14.472kWh, produced 8.790kWh, scale 0.607
[site  ] DEBUG 2025/03/11 13:13:59 ----
[lp-3  ] DEBUG 2025/03/11 13:13:59 charge power: 0W
[lp-3  ] DEBUG 2025/03/11 13:13:59 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/03/11 13:13:59 charge power: 414W
[lp-2  ] DEBUG 2025/03/11 13:13:59 charge power: 0W
[lp-2  ] DEBUG 2025/03/11 13:13:59 charge currents: [0 0 0]A
[circuit-wallboxen] DEBUG 2025/03/11 13:13:59 power: 0W
[circuit-wallboxen] DEBUG 2025/03/11 13:13:59 current: 0A
[circuit-main] DEBUG 2025/03/11 13:13:59 power: 129W
[circuit-main] DEBUG 2025/03/11 13:13:59 current: 1.53A
[lpc   ] DEBUG 2025/03/11 13:13:59 power: 129W
[lpc   ] DEBUG 2025/03/11 13:13:59 current: 1.53A
[site  ] DEBUG 2025/03/11 13:13:59 grid power: 129W
[site  ] DEBUG 2025/03/11 13:13:59 battery 1 power: -1997W
[site  ] DEBUG 2025/03/11 13:13:59 pv 1 power: 3299W
[site  ] DEBUG 2025/03/11 13:13:59 grid currents: [1.42 1.53 -1.26]A
[site  ] DEBUG 2025/03/11 13:13:59 battery 1 soc: 99%
[site  ] DEBUG 2025/03/11 13:13:59 battery has priority at soc 99% (< 100%)
[site  ] DEBUG 2025/03/11 13:13:59 site power: 229W
[lp-1  ] DEBUG 2025/03/11 13:13:59 charger status: B
[lp-1  ] DEBUG 2025/03/11 13:13:59 pv charge current: 0A = 0A + -0.332A (229W @ 3p)
[site  ] DEBUG 2025/03/11 13:13:59 solar forecast: accumulated 14.489kWh, produced 8.800kWh, scale 0.607
[site  ] DEBUG 2025/03/11 13:14:29 ----
```

andig, 2025-03-11:
Du verwendest mqtt? Dann wäre es sich sinnig das mal mit zu loggen. Ersetzt vielleicht die Glaskugel ;)

nicx, 2025-03-11:
@andig hast du meinen ursprünglichen Issue gelesen? Wenn ich manuell schalte in der EVCC UI ("Schnell" bzw. "Aus"), dann funktioniert alles. Also auch das setzen der MQTT Werte laut Konfiguration funktioniert. Ein MQTT Problem kann ich damit doch ausschliessen?!

Lediglich im PV Modus wird der Loadpoint nicht automatisch deaktiviert, sobald nicht genügend Überschuss vorhanden ist. Die Aktivierung funktioniert, nur die Deaktivierung nicht.

Frage 1:
Würde dir trotzdem ein MQTT log von EVCC helfen? Wenn ja, reicht ein Debug Log?

Frage 2:
Und kann ich das Verhalten des PV Modus irgendwie manuell provozieren, zb über das Setzen des "Min. Ladestroms" des Loadpoints? Meine Idee: Wenn der PV Modus die WP aktiviert, dann setze ich den Min. Ladestrom auf einen Wert über dem aktuellen PV Überschuss, dann müsste der LP doch automatisch deaktiviert werden. Ist dem so?

nicx, 2025-03-13:
@andig ich habe nun noch ein wenig weiter experimentiert: Kann es sein, dass der loadpoint mit einem sgready-charger nur dann richtig funktioniert, wenn "1-phasig" gesetzt ist (über die UI)? Hier funktioniert es nämlich wie erwartet (siehe Logfile unten).

Hätte diese Konfiguration irgendwelche negativen Auswirkungen oder ist die nur für den Schwellwert zum Einschalten des Loadpoints wichtig (unabhängig wie die WP technisch angeschlossen ist)? Meine Wärmepumpe ist zwar nur 1-phasig angeschlossen, der zugehörige Heizstab aber 3-phasig. Der Maximalverbrauch übersteigt daher ggf. die Watt-Werte bei eingestellter "1-phasig" option.

```
...
[lp-1 ] DEBUG 2025/03/13 11:30:08 charge power: 529W
[lp-1 ] DEBUG 2025/03/13 11:30:08 charger status: B
[lp-1 ] DEBUG 2025/03/13 11:30:08 pv charge current: 1.14A = 0A + 1.14A (-262W @ 1p)
[lp-1 ] DEBUG 2025/03/13 11:30:08 site power -262W <= 0W enable threshold
[lp-1 ] DEBUG 2025/03/13 11:30:08 pv enable timer start: 1m0s
[lp-1 ] DEBUG 2025/03/13 11:30:08 pv enable in 1m0s
[mqtt ] TRACE 2025/03/13 11:30:10 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:30:15 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:30:20 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:30:25 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:30:27 recv P1P2/P/meter/U/9/Electricity_Power: '644'
[mqtt ] TRACE 2025/03/13 11:30:30 recv P1P2/P/meter/U/9/Electricity_Power: '644'
[mqtt ] TRACE 2025/03/13 11:30:35 recv P1P2/P/meter/U/9/Electricity_Power: '644'
[lp-1 ] DEBUG 2025/03/13 11:30:38 charge power: 644W
[mqtt ] TRACE 2025/03/13 11:30:40 recv P1P2/P/meter/U/9/Electricity_Power: '644'
[mqtt ] TRACE 2025/03/13 11:30:45 recv P1P2/P/meter/U/9/Electricity_Power: '644'
[mqtt ] TRACE 2025/03/13 11:30:50 recv P1P2/P/meter/U/9/Electricity_Power: '644'
[mqtt ] TRACE 2025/03/13 11:30:55 recv P1P2/P/meter/U/9/Electricity_Power: '644'
[mqtt ] TRACE 2025/03/13 11:30:57 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:00 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:05 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[lp-1 ] DEBUG 2025/03/13 11:31:08 charge power: 529W
[mqtt ] TRACE 2025/03/13 11:31:10 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:15 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:20 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:25 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:30 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:35 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[lp-1 ] DEBUG 2025/03/13 11:31:38 charge power: 529W
[lp-1 ] DEBUG 2025/03/13 11:31:38 charger status: B
[lp-1 ] DEBUG 2025/03/13 11:31:38 pv charge current: 1.7A = 0A + 1.7A (-390W @ 1p)
[lp-1 ] DEBUG 2025/03/13 11:31:38 site power -390W <= 0W enable threshold
[lp-1 ] DEBUG 2025/03/13 11:31:38 pv enable in 0s
[lp-1 ] DEBUG 2025/03/13 11:31:38 pv enable timer elapsed
[lp-1 ] DEBUG 2025/03/13 11:31:38 max charge current: 1A
[mqtt ] TRACE 2025/03/13 11:31:38 send espaltherma/sg/set: '2'
[lp-1 ] DEBUG 2025/03/13 11:31:38 charger enable
[lp-1 ] DEBUG 2025/03/13 11:31:38 wake-up timer: start
[mqtt ] TRACE 2025/03/13 11:31:40 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:45 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:50 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:55 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:31:57 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:32:00 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:32:05 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[lp-1 ] DEBUG 2025/03/13 11:32:08 charge power: 414W
[mqtt ] TRACE 2025/03/13 11:32:10 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:32:15 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:32:20 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:32:25 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:32:27 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:32:30 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:32:35 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[lp-1 ] DEBUG 2025/03/13 11:32:38 charge power: 529W
[mqtt ] TRACE 2025/03/13 11:32:40 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:32:45 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:32:50 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:32:55 recv P1P2/P/meter/U/9/Electricity_Power: '529'
[mqtt ] TRACE 2025/03/13 11:32:57 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:33:00 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:33:05 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[lp-1 ] DEBUG 2025/03/13 11:33:08 charge power: 414W
[lp-1 ] DEBUG 2025/03/13 11:33:08 charger status: C
[lp-1 ] INFO 2025/03/13 11:33:08 start charging ->
[... 218 lines cut ...]
[mqtt ] TRACE 2025/03/13 11:48:01 recv P1P2/P/P1P2MQTT/bridge0/T/1/Temperature_R5T_DHW_Tank: '48.312'
[... cut ...]
[mqtt ] TRACE 2025/03/13 11:48:49 recv P1P2/P/P1P2MQTT/bridge0/T/1/Temperature_R5T_DHW_Tank: '48.312'
[mqtt ] TRACE 2025/03/13 11:48:50 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:48:51 recv P1P2/P/P1P2MQTT/bridge0/T/1/Temperature_R5T_DHW_Tank: '48.227'
[mqtt ] TRACE 2025/03/13 11:48:55 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:49:00 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[mqtt ] TRACE 2025/03/13 11:49:05 recv P1P2/P/meter/U/9/Electricity_Power: '414'
[lp-1 ] DEBUG 2025/03/13 11:49:08 charge power: 414W
...

```

nicx, 2025-03-13:
@andig cool, also doch ein bug. ich berichte gerne nach dem nächsten release ob das problem gelöst ist.

btw: wie hätte ich das konkrete problem denn per logfile o.ä. "korrekt" als issue melden können, damit meine meldung euch weitergeholfen hätte? ich würde gerne zukünftig besser werden ;)