evcc-io/evcc #19649: Solar prognosis: show deviation and power

andig, 2025-03-10:
It would be nice to see the current forecasted power as well as the accumulated forecast error in %.

thecem, 2025-03-10:
Wo würde das übergeben werden?

naltatis, 2025-03-12:
I wouldn't want to publicly compare real and forecasted power values. We could a a debug log if people are really interested in this. The scale factor in % is added in https://github.com/evcc-io/evcc/pull/19724 alongside the 🧪 scaling feature toggle.

andig, 2025-03-12:
> I wouldn't want to publicly compare real and forecasted power values.

I would still like to see the currently forecasted value or, alternatively, have a Y axis as guidance. Otherwise these is absolutely nothing that gives me a clue on power, only on energy.

> Too much fluctuation

The forecast doesn't fluctuate and I'd like to see it's value. 

> We could a a debug log if people are really interested in this

I don't want to dig this out in the debug log.