evcc-io/evcc #19652: Wärmepumpen - Aktivität ohne Ladesteuerung tracken

Afrouper, 2025-03-10:
Hallo,

ich verwende eine Viessmann Wärmepumpe. Der Stromverbrauch wird über einen Shelly abgenommen und die Ansteuerung erfolgt über openHAB (Events von ecvv via MQTT und Weitergabe der erhöhten Temperaturen an die Viessmann API).

Erfolgt die Steuerung über evcc funktioniert alles wie gewünscht und die Ladeanzeigt in evcc arbeitet und erkennt nach Abschlus auch den Ladevorgang.

Jetzt läuft die Wärmepumpe aber natürlich auch wenn kein PV Überschuss vorhanden ist da die WP selbst feststellt das das Heizwasser oder das Warmwasser "zu kalt" ist. Dann soll die WP natürlich auch ohne Ansteuerung von evcc starten.

Ich würde nur gerne diese Ladevorgänge auch in evcc tracken:
-  Ich kann zwar den Status des Chargers auf "C" setzen, bekomme dann aber einen Logikfehler (should be disabled, but was enabled)
- Setze ich den Charger auf enabled umgehe ich die Steuerung von evcc. Ich "brauche" ja das Event von evcc wenn genügend PV Überschuss vorhanden ist um die o.g. Temperaturen anzupassen
- Es könnte das Verhalten von Smarten Schaltern passen. Dann müsste ich sie aber vermutlich etwas genauer ansteuern können.
- Ich passe bereits den "enabledThreshold" am Ladepunkt dynamisch an (Wenn die WP schon in Betrieb ist wenn die Sonne aufgeht, braucht es natürlich weniger PV Überschuss bis ich die Temperaturen erhöhen kann); hier könnte man das ggf. generischer Implementieren und den enabledThreshold dynamisch berechnen (gemessenr Verbrauch - Threshold)

Das könnte auch andere Wärmepumpen generell betreffen. Fände euer Feedback interessant.

Danke Christian

schenlap, 2025-03-10:
Hallo, ich möchte auch gerne den Verbrauch der Wärmepumpe darstellen.
@VolkerK62 
Kannst du die Config posten wie du das realisiert hast?

naltatis, 2025-03-11:
Mit https://github.com/evcc-io/evcc/issues/13050 planen wir eine zusätzliche, rein zeit-basierte Auswertung zu implementieren. Hier gibs aktuell auch schon erste Ideen zur Umsetzung. Daher würd ich diesen Issue zu einer Diskussion konvertieren.