evcc-io/evcc #19660: Wärmepumpen - Unterstützung von Buderus Wärmepumpen mit ems-esp

Tihmann, 2025-03-10:
Hallo,

ich würde auch noch gerne um Unterstützung von Buderus Wärmepumpen bitten. Meine Buderus Wärmepumpe ist über ein ems-esp Modul eingebunden, ein Open Source Projekt: https://emsesp.org/. Damit ist auf unterschiedliche Varianten möglich die Wärmepumpe auszulesen und zu steuern: über die Console mit Telnet or Serial, über HTTP mit RESTful API calls und via MQTT über topics und payloads https://docs.emsesp.org/Commands/

Meine Buderus Wärmepumpe ist SG Ready, das Interessante an ems-esp ist das man den Ausgang 4 der Wärmepumpe (Steuerung SG ready) einfach per ems-esp Aufruf invertieren kann, so dass man kein physikalisches Kabel / Relais anschließen muss (bei Invertierung ist Status offen gleich SG ready aktiv).

Tihmann, 2025-03-11:
Gerne würde ich mich auch an der Entwicklung beteiligen. Ich wollte mich gerade etwas einarbeiten, finde im GitHub aber die Templates für die Wärmepumpen nicht. Kann mir jemand da einen Hinweis geben.

Für die weitere Entwicklung wäre es hilfreich, ein bestehendes Template zu wissen, in dem Parameter über MQTT ausgelesen und gesetzt werden. Lesen gibt es hier: https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/mazda2mqtt.yaml 

aber schreiben per MQTT ?

andig, 2025-03-11:
HTTP wäre bevorzugt. Aus der Doku geht nicht wirklich hervor, wie hier der Ansatz wäre: `sgready`? Oder der WP den Überschuss mitteilen (`heatpump`)? Die Implementierung kannst Du auf der Basis selbst ausprobieren, wenns funktoiniert können wir das in ein Template gießen.

Tihmann, 2025-03-11:
Hier gibt es weitere Infos zur Aktivierung von SG ready https://bbqkees-electronics.nl/2024/10/03/using-the-smart-grid-sg-and-photovoltaic-pv-function-of-your-heat-pump-with-the-ems-gateways/ 

Eigentlich wird nur das Schalten des SG Ready Eingangs simuliert, indem die Logik invertiert wird

Tihmann, 2025-03-11:
Zusätzlich zum Schalten des SG Ready Eingangs können folgende Parameter per Software geändert werden:

- Erhöhung der Wunschtemperatur beim Heizen in K
- Erhöhter Warmwasserkomfort ja/nein
- Absenkung der Wunschtemperatur beim Kühlen in K
- Kühlen nur mit PV Energie ja/nein
- Max. Leistung für Kompressor von 0,5 kW bis 6,0 kW in 0,1 Schritten

andig, 2025-03-12:
Es wäre sinnig- wenn das geht- das auf den SG Ready Eingang zu reduzieren. Die Logik dahinter, also was die WP dann mit dem "boost" machen will, kann sie selbst entscheiden bzw. ist bei vielen WP konfigurierbar. "Erhöhter Warmwasserkomfort" sollte sich egtl. auch hinter "boost" verbergen.

Long story short: es braucht also eine `sgready` Konfiguration welche den entsprechenden SG Ready Eingang für "angehobener Betrieb" aktiviert. Siehe z.B. https://github.com/evcc-io/evcc/blob/master/templates/definition/charger/stiebel-lwa.yaml als Vorlage.

andig, 2025-03-12:
> Max. Leistung für Kompressor von 0,5 kW bis 6,0 kW in 0,1 Schritten

Das wäre eine andere Lösung- da kann evcc einen "power envelope" vorgeben.  Das kann aber _nur_ funktionieren wenn die WP das nicht als hartes Limit versteht, sondern als "verfügbare Überschussleistung". In dem Fall (und das scheint hier nicht zu gelten) bräuchte es dann einen `heatpump` Charger.

premultiply, 2025-03-12:
> Max. Leistung für Kompressor von 0,5 kW bis 6,0 kW in 0,1 Schritten

Das ist normalerweise eine externe Leistungslimitierung.
Wird z.B. eingesetzt für übergeordnete Heizungs- und Klimasteuerungen und wird klassischerweise in der Regelungstechnik mit einer 0-10V-Schnittstelle analog abgebildet um die Modulation zu begrenzen.

Für evcc eher kein wirklich sinnvoller Weg.

Tihmann, 2025-03-12:
Ich denke nicht, dass es eine externe Leistungslimitierung ist. Definitiv nicht über eine 0-10V Schnittstelle geregelt, sondern über Einstellung im Menü: https://buderus-de-de.boschtt-documents.com/download/file/file/6721891897.pdf Menü-Punkt 5.1.12: In diesem Menü werden die Einstellungen für das Photovoltaik-System (PV-System) vorgenommen. Max. Lstg. f.
Kompressor -Maximale Leistung für den Kompressorbetrieb bei aktiviertem Photovoltaik-Betrieb einstellen.

rbeudel, 2025-03-14:
Ich bekomme demnächst das neue eebus Modul das es für einige Bosch/Buderus Pumpen gibt. Mal sehen ob das ausser §14a noch etwas anderes kann.

Tihmann, 2025-03-14:
Dieses Modul ? MX400 ? https://www.buderus.de/de/unternehmen/presse/presseinfos/presseuebersicht/flexibler-vernetzen-77652

Das ist bei mir bereits verbaut, im Menü ist EEBus bereits auszuwählen. Habe aber noch nichts damit herausgefunden

rbeudel, 2025-03-14:
Ich habe Bosch, da heist es K 40F. Die einzige Information ist, das bei Verbindung mit der Steuerbox die §14a Funktion gegeben sein soll.

dirkmolitor, 2025-03-16:
> Hallo,
>
> ich würde auch noch gerne um Unterstützung von Buderus Wärmepumpen bitten. Meine Buderus Wärmepumpe ist über ein ems-esp Modul eingebunden, ein Open Source Projekt: https://emsesp.org/. [...]

Hallo, nutze an meiner Bosch WP CS7x00i AW ebenfalls das ems-esp Modul. Ich würde mich sehr freuen, wenn eine Steuerung ohne zusätzliches SG-Relais möglich wird. 
VG Dirk M.

andig, 2025-03-16:
Wir drehen uns im Kreis: https://github.com/evcc-io/evcc/issues/19660#issuecomment-2717023949. 

Die Frage ist ganz einfach: ist es möglich, die SG Ready Zustände per API anzusprechen?

Tihmann, 2025-03-16:
Ja ist es. Nicht über die API von Buderus, sondern über die API von ems-esp https://bbqkees-electronics.nl/wiki/gateway/restful-api.html

Tihmann, 2025-03-16:
Um SG Ready zu aktivieren muss Ausgang I4 auf 100000000000 gesetzt werden

andig, 2025-03-16:
@Tihmann leider hilft das nicht weiter. Es geht nicht darum "SG Ready zu aktivieren" sondern einen bestimmten SG Ready Zustand zu aktivieren- nämlich Boost/Erhöhter Betrieb.

> Um SG Ready zu aktivieren muss Ausgang I4 auf 100000000000 gesetzt werden

Ich habe keine Ahnung, was das bedeutet soll. Im Zweifel brauchts den vollständigen REST Request, bitte keine Informationsbröckchen.

Um es nochmal deutlich zu sagen: wir haben weder die entsprechenden Geräte noch Zeit oder Muße, uns die Informationen zusammen zu suchen. Der Ball liegt bei Euch!

Tihmann, 2025-03-16:
Damit wird die Eingangslogik von I4 invertiert, also I4 offen ist aktiviert

Tihmann, 2025-03-16:
Ich hatte Euren Aufruf im Blog so verstanden, dass man weitere Wärmepumpen hier als Issue aufmachen soll. Sorry wenn meine Angaben unvollständig sind. Ich brauche noch etwas Zeit um mich in die Logik und Programmierung von evcc einzuarbeiten. Insofern schließe ich hier erst einmal

andig, 2025-03-16:
Hier geht es erst im zweiten Schritt darum, wie etwas in evcc umzusetzen wäre. Zuerst mal muss geklärt werden, wie _konkret_ mit der WP zu interagieren ist. Das muss vollständig beschrieben sein und jemand mit Gerät muss es ausprobieren. Und genau das fehlt hier.

rbeudel, 2025-03-16:
Bei diesen Wärmepumpen gibt es 4 Eingänge die entweder über Kontakte mit einem Schalter zB. Shelly oder über mqtt mit ems-esp geschaltet werden können. Was genau diese Kontakte genau machen sollen kann in der Wärmepumpe eingestellt werden. Also muss von aussen nur der Eingang 4 (PV) umgeschaltet werden. Über mqtt muss im Topic "ems-esp/boiler_data" die payload "hpin4opt": "000000000000" oder eben "hpin4opt": "100000000000" gesendet werden. Wie das genau mit evcc geht kann ich nicht sagen.

andig, 2025-03-16:
@rbeudel geht das auch über HTTP? Mit welchem Request (curl …?)

rbeudel, 2025-03-16:
Geht tatsächlich auch mit curl. [https://docs.emsesp.org/Commands/](url) Hier steht es genauer beschrieben. Damit hatte ich aber noch nicht viel zu tun. Als Versuchskaninchen würde ich aber zur Verfügung stehen.

andig, 2025-03-16:
Wie gesagt:

> Das muss vollständig beschrieben sein und jemand mit Gerät muss es ausprobieren. Und genau das fehlt hier.

Es braucht den vollständigen Request, für Boost wie auch zurück zu Normal. Bitte keine Links zur Doku. Wir können das weder probieren noch haben wir Zeit uns da durch zu wühlen. Danke.

Tihmann, 2025-03-16:
@rbeudel kannst Du bitte einmal ein Foto von Deinem Menü Eingang Extern 4 Posten?

Tihmann, 2025-03-16:
Wie gesagt komme ich erst nächste Woche zur konkreten Umsetzung, aber dies aus der ems-esp Doku schon einmal als erste Gedankenstütze:

Some commands can be dangerous and require security authentication, unless disabled via an EMS-ESP setting. The authentication is in the form of an Access Token which is generated from the WebUI's Security->Manage Users and then clicking on the key button for the admin user. The 152 character long string must be included in the HTTP header like "Authorization: Bearer {ACCESS_TOKEN}". The tokens have no expiry date

# Most GETs do not need authentication
% curl http://ems-esp.local/api/thermostat/seltemp

# POSTs (with -d) need authentication tokens
% curl http://ems-esp.local/api/thermostat/seltemp \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ8.eyJ1c2VybmFtZSI6ImFkbWluIiwiYWRtaW4iOnRydWUsInZlcnNpb24iOiIzLjEuMWIwIn0.qeGT53Aom4rDYeIT1Pr4BSMdeWyf4_zN9ue2c51ZnM0' \
  -d '{ "value" : 22.5 }'

rbeudel, 2025-03-16:
[image] Das ist der Text aus dem Handbuch. Ich komme zur Zeit nicht an die Wärmepumpe.

Tihmann, 2025-03-18:
@andig Ich konnte es jetzt einmal testen. Die curl Befehle sind wie folgt (<ems-esp-ip> und <Access_Token> müssen angepasst werden). Anleitung zum Access Token: The authentication is in the form of an Access Token which is generated from the WebUI's Security->Manage Users and then clicking on the key button for the admin user. The 152 character long string must be included in the HTTP header like "Authorization: Bearer {ACCESS_TOKEN}". The tokens have no expiry date

1.) PV Anlage im Menü aktivieren, erst dann sind die PV Einstellungen verfügbar: 
curl http://<ems-esp-ip>/api/boiler/hpin4opt \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer <Access_Token>' \
  -d '{ "value" : "000000100000" }'

2.) Modus "Verstärkter Betrieb der Wärmepumpe möglich" einschalten:
curl http://<ems-esp-ip>/api/boiler/hpin4opt \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer <Access_Token>' \
  -d '{ "value" : "100000100000" }'

3.) Modus "Verstärkter Betrieb der Wärmepumpe möglich" ausschalten:
curl http://<ems-esp-ip>/api/boiler/hpin4opt \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer <Access_Token>' \
  -d '{ "value" : "000000100000" }'

4.) Modus "Erzwungener verstärkter Wärmepumpenbetrieb." einschalten:
curl http://<ems-esp-ip>/api/boiler/hpin1opt \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer <Access_Token>' \
  -d '{ "value" : "100000000000" }'
curl http://<ems-esp-ip>/api/boiler/hpin4opt \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer <Access_Token>' \
  -d '{ "value" : "100000100000" }'

5.) Modus "Erzwungener verstärkter Wärmepumpenbetrieb." ausschalten:
curl http://<ems-esp-ip>/api/boiler/hpin1opt \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer <Access_Token>' \
  -d '{ "value" : "000000000000" }'
curl http://<ems-esp-ip>/api/boiler/hpin4opt \
  -H 'Content-Type: application/json' \
  -H 'Authorization: Bearer <Access_Token>' \
  -d '{ "value" : "000000100000" }'

@rbeudel @dirkmolitor bitte einmal testen, ob das für Euch so funktioniert

Tihmann, 2025-03-18:
gerade festgestellt, dass man die Zeichen, die nicht geändert werden sollen durch "x" ersetzen kann: 

1.) PV Anlage im Menü aktivieren, erst dann sind die PV Einstellungen verfügbar:
curl http:///api/boiler/hpin4opt
-H 'Content-Type: application/json'
-H 'Authorization: Bearer <Access_Token>'
-d '{ "value" : "xxxxxx1xxxxx" }'

2.) Modus "Verstärkter Betrieb der Wärmepumpe möglich" einschalten:
curl http:///api/boiler/hpin4opt
-H 'Content-Type: application/json'
-H 'Authorization: Bearer <Access_Token>'
-d '{ "value" : "1xxxxxxxxxxx" }'

3.) Modus "Verstärkter Betrieb der Wärmepumpe möglich" ausschalten:
curl http:///api/boiler/hpin4opt
-H 'Content-Type: application/json'
-H 'Authorization: Bearer <Access_Token>'
-d '{ "value" : "0xxxxxxxxxxx" }'

4.) Modus "Erzwungener verstärkter Wärmepumpenbetrieb." einschalten:
curl http:///api/boiler/hpin1opt
-H 'Content-Type: application/json'
-H 'Authorization: Bearer <Access_Token>'
-d '{ "value" : "1xxxxxxxxxxx" }'
curl http:///api/boiler/hpin4opt
-H 'Content-Type: application/json'
-H 'Authorization: Bearer <Access_Token>'
-d '{ "value" : "1xxxxxxxxxxx" }'

5.) Modus "Erzwungener verstärkter Wärmepumpenbetrieb." ausschalten:
curl http:///api/boiler/hpin1opt
-H 'Content-Type: application/json'
-H 'Authorization: Bearer <Access_Token>'
-d '{ "value" : "0xxxxxxxxxxx" }'
curl http:///api/boiler/hpin4opt
-H 'Content-Type: application/json'
-H 'Authorization: Bearer <Access_Token>'
-d '{ "value" : "0xxxxxxxxxxx" }'

Tihmann, 2025-03-18:
Das wären die Einstellungen für den Typ sgready, für den Typ heating gibt es noch die zusätzliche Einstellung "PV max. Kompressorleistung". Diese wird wie folgt gesetzt:

curl http:///api/boiler/pvmaxcomp
-H 'Content-Type: application/json'
-H 'Authorization: Bearer <Access_Token>'
-d '{ "value" : <Wert in kW> }'

Tihmann, 2025-03-19:
@andig Welche Parameter benötigt evcc noch zur Anzeige ? Temperatur Vorlauf ? Temperatur Warmwasser? Dann würde ich diese Parameter auch noch hier darstellen

andig, 2025-03-19:
WW oder Pufferspeicher haben wir meistens.

rbeudel, 2025-03-19:
Ich habe versucht den Punkt 1 auszuführen. Dabei kam folgendes heraus:
`root@evcc:~# curl http://192.168.1.145/api/boiler/hpin4opt
-H 'Content-Type: application/json'
-H 'Authorization: Bearer cCI6IkpXVyJ1c2VybmFtZSI6ImFkbWluIiwiYWRxxxxxxxWV9.2bHpWya2C7Q12WjNUBD6_7N3RCD7CMl-EGhyQVzFdDg'
-d '{ "value" : "xxxxxx1xxxxx" }'
{"name":"hpin4opt","fullname":"Einstellung Eingang 4","circuit":"","value":"000000000000","type":"string","readable":true,"writeable":true,"visible":true}-bash: -H: Kommando nicht gefunden.
-bash: -H: Kommando nicht gefunden.
-bash: -d: Kommando nicht gefunden.
`
Den Bearer habe ich hier zur Anzeige Verändert.  Habe ich ansonsten einen Fehler drin? Es gibt in den Einstellungen von ems-esp einen Punkt: Api :Zugriffstoken-Autorisierung bei API-Aufrufen umgehen. Könnte das die Sache vereinfachen?

jk, 2025-03-19:
> Ich habe versucht den Punkt 1 auszuführen. Dabei kam folgendes heraus: `root@evcc:~# curl http://192.168.1.145/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer [...]

Ich vermute du hast das einfach kopiert, eingefügt und Enter gedrückt. Das muss entweder alles in einer Zeile stehen oder du musst die Zeilen explizit umbrechen, z. B.:

```bash
curl -X POST "http://192.168.1.145/api/boiler/hpin4opt" \
  -H "Content-Type: application/json" \
  -H "Authorization: Bearer cCI6IkpXVyJ1c2VybmFtZSI6ImFkbWluIiwiYWRxxxxxxxWV9.2bHpWya2C7Q12WjNUBD6_7N3RCD7CMl-EGhyQVzFdDg" \
  -d '{ "value" : "xxxxxx1xxxxx" }'
```

rbeudel, 2025-03-19:
Richtig vermutet. So klappts auch mit dem Befehl. Die entsprechende Anzeige in ems-esp ist von 00000... auf 000000100000 gesprungen.

Tihmann, 2025-03-19:
Warmwasserkreislauf: Aktuelle interne Temperatur
curl http://ems-esp-ip/api/boiler/dhw/curtemp

Tihmann, 2025-03-19:
> Richtig vermutet. So klappts auch mit dem Befehl. Die entsprechende Anzeige in ems-esp ist von 00000... auf 000000100000 gesprungen.

@rbeudel Ist damit bei Dir auch der Menüpunkt Photovoltaikanlage auf ja gesprungen und im Menü eine Ebene höher die Einstellmöglichkeiten für Photovoltaikanlage zu sehen ?

andig, 2025-03-19:
Dann muss man das doch nur nochmal analog der anderen Beispiele in ein sgready Template packen 😀

Tihmann, 2025-03-19:
Okay, hier brauche ich wirklich ein wenig Hilfe mit evcc. Kannst Du mir bitte ein sgready Template, ein heating Template und ein beliebiges Template mit einem curl Aufruf nennen, dann kann ich mir den Rest zusammensuchen und versuchen ein Buderus / Bosch Template zu erstellen. Danke !

rbeudel, 2025-03-19:
> [@rbeudel](https://github.com/rbeudel) Ist damit bei Dir auch der Menüpunkt Photovoltaikanlage auf ja gesprungen und im Menü eine Ebene höher die Einstellmöglichkeiten für Photovoltaikanlage zu sehen ?

Ich habe noch die 7400 Serie mit dem älteren Steuermodul. Hier wird bei der Aktivierung PV das Modul PV im Hauptmenue angezeigt, sonst nicht. Also alles klar.

andig, 2025-03-20:
Verstärkt/erzwungen musst du entscheiden und auf Boost mappen. Das Konzept kennt evcc nicht.

Tihmann, 2025-03-20:
Okay, verstehe. Und was ist dann dimm ?

andig, 2025-03-20:
Das ist §14a. Derzeit nixht genutzt, kann per `error` Plugin implementiert werden.

Tihmann, 2025-03-20:
Anbei ein erster Versuch in go. @andig bitte gerne einmal über die Syntax schauen

`template: buderus_emsesp
products:
  - brand: Buderus
    description:
      generic: Buderus (SG Ready)
  - brand: Bosch
    description:
      generic: Bosch (SG Ready)
group: heating
# requirements:
#   evcc: ["sponsorship"]
params:
  - name: emsesp_ip
    choice: ["tcpip"]
  - name: emsesp_token
  - name: tempsource
    type: choice
    choice: ["", "warmwater"]
    description:
      de: "Temperaturquelle"
      en: "Temperature source"
render: |
  type: sgready
  getmode:
    source: map
    values:
      1: 3 # Frostschutz
      2: 1 # Normal
      3: 2 # Forcierter Betrieb
      4: 2 # Sofortige Ansteuerung
    get:
      source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        address: 5000
        type: input
        encoding: uint16
  setmode:
    source: switch
    switch:
    - case: 1 # normal
      set:
        source: sequence
        set:
        - source: const
          value: 0
          set:
            source: http
            uri: http://{{ .emsesp-ip }}/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer {{ .emsesp_token }}' -d '{ "value" : "0xxxxxxxxxxx" }'
        - source: const
          value: 0
          set:
            source: http
            uri: curl http://{{ .emsesp-ip }}/api/boiler/hpin1opt -H 'Content-Type: application/json' -H 'Authorization: Bearer {{ .emsesp_token }}' -d '{ "value" : "0xxxxxxxxxxxxxx" }'
    - case: 2 # boost
      set:
        source: sequence
        set:
        - source: const
          value: 1
            source: http
            uri: http://{{ .emsesp-ip }}/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer {{ .emsesp_token }}' -d '{ "value" : "1xxxxxxxxxxx" }'
        - source: const
          value: 0
          set:
            source: http
            uri: http://{{ .emsesp-ip }}/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer {{ .emsesp_token }}' -d '{ "value" : "0xxxxxxxxxxx" }'
    - case: 3 # dimm
      set:
        source: sequence
        set:
        - source: const
          value: 1
            source: http
            uri: http://{{ .emsesp-ip }}/api/boiler/hpin1opt -H 'Content-Type: application/json' -H 'Authorization: Bearer {{ .emsesp_token }}' -d '{ "value" : "1xxxxxxxxxxxxxx" }'
        - source: const
          value: 1
            source: http
            uri: http://{{ .emsesp-ip }}/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer {{ .emsesp_token }}' -d '{ "value" : "0xxxxxxxxxxx" }'    
        - source: const
          value: 0
          set:
            source: http
            uri: http://{{ .emsesp-ip }}/api/boiler/hpin1opt -H 'Content-Type: application/json' -H 'Authorization: Bearer {{ .emsesp_token }}' -d '{ "value" : ""0xxxxxxxxxxxxxx" }'
  {{- if .tempsource }}
  temp:
    source: http
    uri: http://{{ .emsesp-ip }}/api/boiler/dhw/curtemp
    jq: .value 
  {{- end }}`

andig, 2025-03-21:
Mach bitte einen PR- da kann man auch im Code kommentieren. Header sind NICHT Teil der Url, siehe Doku

dirkmolitor, 2025-03-21:
@Tihmann : funktioniert bei mir. Danke. 
 
> 1.) PV Anlage im Menü aktivieren, erst dann sind die PV Einstellungen verfügbar: curl http://ems-esp-ip/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer <Access_Token>' -d '{ "value" : "000000100000" }' Punkt 1 muss nur einmalig aktiviert werden
> 
> 2.) Modus "Verstärkter Betrieb der Wärmepumpe möglich" einschalten: curl http://ems-esp-ip/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer <Access_Token>' -d '{ "value" : "100000100000" }'
> 
> 3.) Modus "Verstärkter Betrieb der Wärmepumpe möglich" ausschalten: curl http://ems-esp-ip/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer <Access_Token>' -d '{ "value" : "000000100000" }'
> 
> 4.) Modus "Erzwungener verstärkter Wärmepumpenbetrieb." einschalten: curl http://ems-esp-ip/api/boiler/hpin1opt -H 'Content-Type: application/json' -H 'Authorization: Bearer <Access_Token>' -d '{ "value" : "100000000000000" }' curl http://ems-esp-ip/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer <Access_Token>' -d '{ "value" : "100000100000" }'
> 
> 5.) Modus "Erzwungener verstärkter Wärmepumpenbetrieb." ausschalten: curl http://ems-esp-ip/api/boiler/hpin1opt -H 'Content-Type: application/json' -H 'Authorization: Bearer <Access_Token>' -d '{ "value" : "000000000000000" }' curl http://ems-esp-ip/api/boiler/hpin4opt -H 'Content-Type: application/json' -H 'Authorization: Bearer <Access_Token>' -d '{ "value" : "000000100000" }' ` [@rbeudel](https://github.com/rbeudel) [@dirkmolitor](https://github.com/dirkmolitor) bitte einmal testen, ob das für Euch so funktioniert

Tihmann, 2025-03-21:
https://github.com/evcc-io/evcc/pull/19977

Tihmann, 2025-03-21:
> Mach bitte einen PR- da kann man auch im Code kommentieren. Header sind NICHT Teil der Url, siehe Doku

also diese Teile der URL sind nicht korrekt ? -H 'Content-Type: application/json' -H 'Authorization: Bearer {{ .emsesp_token }}'

andig, 2025-03-21:
Das sind Header, keine Url

Tihmann, 2025-03-21:
also so ?

source: http
uri: http://{{ .emsesp_ip }}/api/boiler/hpin4opt 
headers: 
  -Content-Type: application/json
  -Authorization: Bearer {{ .emsesp_token }}
  -value" : "0xxxxxxxxxxx"

jk, 2025-03-21:
@Tihmann Fast. `-d` steht für `data` bei curl und ist dann im Fall mit dem Content-Type vermutlich ein der Body. Laut [Doku](https://docs.evcc.io/docs/reference/plugins#http-lesenschreiben) müsste das dann wahrscheinlich so aussehen:

```yaml
source: http
uri: http://{{ .emsesp_ip }}/api/boiler/hpin4opt
headers:
  - Content-Type: application/json
  - Authorization: Bearer {{ .emsesp_token }}
body: "{\"value\": \"0xxxxxxxxxxx\"}"
```

Tihmann, 2025-03-21:
Vermutlich sogar noch mit method: POST ?
```
source: http
uri: http://{{ .emsesp_ip }}/api/boiler/hpin4opt
method: POST
headers:
  - content-type: application/json
  - authorization: Bearer {{ .emsesp_token }}
body: "{\"value\": \"0xxxxxxxxxxx\"}"
```

jk, 2025-03-21:
@Tihmann Ja, vermutlich auch implizit, aber sicherlich besser du schreibst das dazu. Hab das in meinem Snippet gerade auch noch geändert.

andig, 2025-03-21:
https://github.com/evcc-io/evcc/issues/19660#issuecomment-2742616979

Tihmann, 2025-03-21:
> [@Tihmann](https://github.com/Tihmann) Fast. `-d` steht für `data` bei curl und ist dann im Fall mit dem Content-Type vermutlich der Body. Laut [Doku](https://docs.evcc.io/docs/reference/plugins#http-lesenschreiben) müsste das dann wahrscheinlich so aussehen:
> 
> source: http
> uri: http://{{ .emsesp_ip }}/api/boiler/hpin4opt
> method: POST
> headers:
>   - content-type: application/json
>   - authorization: Bearer {{ .emsesp_token }}
> body: "{\"value\": \"0xxxxxxxxxxx\"}"

@andig vielen Dank für den Link zur Doku und das Kommentieren von meinem PR. Ist das Beispiel der URI hier oben drüber so richtig? Ich würde dann den PR anpassen

andig, 2025-03-21:
Lass es uns mal probieren. Du kannst das v.a. auch lokal testen! Siehe https://github.com/evcc-io/evcc/issues/19489

andig, 2025-03-21:
body: "{\"value\": \"0xxxxxxxxxxx\"}"''

soll sicher sein

    body: {"value": "0xxxxxxxxxxx"}

Tihmann, 2025-03-21:
Habe es gerade einmal lokal getestet. Allerdings kommt hier eine Fehlermeldung:

[main  ] FATAL 2025/03/21 23:15:55 charger [heatpump_control] cannot create charger 'heatpump_control': cannot create charger type 'template': yaml: line 21: did not find expected key:

andig, 2025-03-22:
Es wäre total sinnvoll dies im PR zu diskutieren wo man den Code sieht.

MarkusMit, 2025-03-23:
Hi, ich habe eine Bosch CS6800i mit EMS-ESP, und evcc in HomeAssistant laufen. Könnte ggf. auch was testen.
Habe die PV-Steuerung dzt. über HA-Helfer und -Automatisierung realisiert, wäre aber interessiert an evcc Lösung.

Tihmann, 2025-03-23:
Der erste Entwurf klappt soweit okay. Es fehlen noch zwei Punkte:

- korrekte Umsetzung von getmode
- korrekte Umsetzung der Zieltemperatur

[68 more replies]

closed 2025-03-24 by andig