evcc-io/evcc #19673: Seit update auf Version 0.200.9 startet evcc nicht mehr

m0nn3, 2025-03-11:
### Describe the bug

Seit dem Update auf version 0.200.9 startet evcc nicht mehr und gibt mir Fehlermeldungen.
EVCC läuft unter Home Assistant

### Steps to reproduce

1. update von 0.200.5 auf 0.200.9
2. evcc starten
3. Fehlermeldung
4. EVCC auf dem PC normal installiert und mit der evcc.yaml und der evcc.db von HomeAssistant gestartet und es kommt der Fehler siehe Log2
...

### Configuration details

```yaml
Meine evcc.yaml

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: MEINEIP # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: MEINEID

interval: 10s # control cycle interval

sponsortoken: MEINTOKEN

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.359 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.082 # [currency]/kWh

chargers:
  
- name: wallbox6
  type: template
  template: innogy-ebox
    
    # Modbus TCP
  modbus: tcpip
  id: 1
  host: IPADRESSE # Hostname
  port: 5023 # Port

loadpoints:
- title: Carport
  charger: wallbox6
  phases: 3
  mincurrent: 6
  maxcurrent: 30

site:
  title: Mein Zuhause
```

### Log details

```text
Log1 HomeAssistant

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/11 11:12:29 evcc 0.200.9
[main  ] INFO 2025/03/11 11:12:29 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/11 11:12:29 using sqlite database: /data/evcc.db
[main  ] INFO 2025/03/11 11:12:29 listening at :7070
[main  ] FATAL 2025/03/11 11:12:29 meter [db:8] cannot create meter 'db:8': cannot create meter type 'template': invalid key: storageunit
[main  ] FATAL 2025/03/11 11:12:29 will attempt restart in: 15m0s

Log2 nativ auf dem Laptop (debian)

[main ] INFO 2025/03/11 11:30:43 evcc 0.200.9
[main ] INFO 2025/03/11 11:30:43 using config file: /etc/evcc.yaml
[db ] INFO 2025/03/11 11:30:43 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/03/11 11:30:43 listening at :7070
[main ] FATAL 2025/03/11 11:30:43 meter [db:8] cannot create meter 'db:8': cannot create meter type 'template': invalid key: storageunit
[main ] FATAL 2025/03/11 11:30:43 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.9

m0nn3, 2025-03-11:
Ich bin selber zu einer Lösung gekommen:

ich habe die evcc.db auf meinem Laptop gezogen, das Programm "DB Browser for SQLite Version 3.12.2" installiert.
Damit die evcc.db geöffnet und unter "Daten durchsuchen" mit Tabelle "configs" habe ich in allen Einträgen den Eintrag:

"storageunit":"1",

raus gelöscht und übernommen.

Dann die Datei wieder in HomeAssistant hochgeladen und es läuft wieder!

Tipp: in Home Assistant im Terminal musste ich die evcc.db noch aus dem Container holen und wieder zurück kopieren

#Datei vom /adddon_configs Ordner zu EVCC kopieren
docker cp /addon_configs/49686a9f_evcc/evcc.db addon_49686a9f_evcc:/data/

Da muss ich mich noch mal informieren wie man mit dem File editor in Homeassistant direkt an die Datenbank kommt.
Aber es läuft erst mal.

andig, 2025-03-11:
Ist im nightly behoben

> have verified that the issue is reproducible with the latest nightly build

Die Mühe hast du dir wohl nicht gemacht :(

closed 2025-03-11 by andig

m0nn3, 2025-03-11:
Entschuldige, ich hatte vergessen die nightly zu aktivieren beim installieren..