evcc-io/evcc #19681: Enever pricing

enboer, 2025-03-11:
When using Enever to get the prices I understand EVCC does 4 queries a day (4 for today, 4 for tomorrow). This makes sure is stays in the 250 monthly limit.
I would like to have this same price for the feedin. I can add the Enever setup for that, but then it cost again 8 tokens a day I think. So is there a way to take the electricity price also as feedin price?

StefanSchoof, 2025-03-11:
Similar issue: https://github.com/evcc-io/evcc/issues/18402

closed 2025-03-11 by andig

enboer, 2025-03-11:
I am not sure if it is the same as #18402. This #18402 limits the api calls to 8 per day. which works fine. But for a dynamic contract, I do not need separate api calls for the feedin tariff, that is the same. So what I suggest is an option that feedin takes the same values as grid tariff. An not use separate template for these values.