evcc-io/evcc #19690: Make power supply for Heating/Climating more transparent and configurable in the UI.

6onions, 2025-03-11:
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_
I think evcc somehow retrieves information from the car on whether the WB shall provide some power for the climater, via the SEAT-API, and it tries to use it already "somehow", but it is not very reliable and mostly intransparent.
So I often end up with a not 100% SOC at the end of my plan when I want to depart 
(often only 90% - making a painfull difference in distance in winter on my only 13kWh PHEV). 

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
You could implement a "supply power for climater" switch on every plan for planned climating
and another one close to the 4 mode selector buttons for instant climating. 
When they are switched on, evcc switches power on on the WB at the specified level at/for the specified time (see parameters below) independently of any SOC-levels/limits etc.

In addition there should be some means of configuration (either in the UI or in the yaml) for this.
l would at least see the first three of the following parameters: 
- Time to start power supply for climater before end of plan
- Time to leave power on after the end of the plan (in case I leave a little late, but switch of after that time if I don't leave at all)
- Time to leave power on after the instant climatising was started (e.g. 15 or 30 min) (in case I "forget" to leave)
- Power level for climater (something between 1,4 and 3,7 kW)

In addition to make fhe feature really hot/cool: :-)
- Define the power level ideally separately for winter/heating and summer/cooling (e.g. 3,7kW in winter (as my car has no heatpump) and 1,4 in summer (as this is sufficient for cooling)) -> see below
- Define a sensor for finding out if it is "winter" or "summer", e.g a temperture sensor for example by MQTT from any home-automation
- Define a temperature limit (from above sensor) for "winter/heating starts below X °C" e.g. <= 18°C
- Define a temperature limit (from above sensor) for "summer/cooling starts above Y °C" e.g. >= 24°C

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Currently I am forced to fiddle around with the mode selector in combination with the SOC limit to achive instant power supply for manual climating
(I manually switch the climater on in the car via the SEAT-App in those cases, but I need to make sure it doesn't drain the battery).

**Additional context**
_Add any other context or screenshots about the feature request here._

--

andig, 2025-03-12:
If we do this then with minimal additional options. So we'll need to chose:

- 1p or 3p or as-is?
- "now" or "minpv"?

6onions, 2025-03-13:
- 1p should not only be sufficient in any case it also makes sure the battery is not loaded more then wanted if a lower SOC-limit was set and I don’t use too much grid-power if PV-power is not sufficient.
- I am not sure if I understand the question well, but I would assume “minpv” should be sufficient/desired, for the same reasons as above 

Just important is that any SOC or energy limit is ignored in that case.

closed 2025-04-25 by andig