evcc-io/evcc #19705: Kostal Plenticore | Erzeugungsleistung ist Summe der DC Seite - nicht die relevante AC Seite

Heino1988, 2025-03-12:
### Describe the bug

Hallo zusammen,

wir ist aufgefallen, dass in der Kostal Plenticore Integration bei der Erzeugung die DC Werte genommen werden, und nicht die m.E. nach dem Wechselrichter relevanten AC-Werte. 

[image]

[image]

Sichtbar vor allem kurz vor Sonnenuntergang, wenn die Gleichstromseite mit Verlusten durch den Wechselrichter nicht mehr ausreicht für eine AC Leistung > 0W, in EVCC aber noch eine Leistung angezeigt wird.

Hoffe ich konnte es nachvollziehbar erklären, sonst gerne nachfragen.

Viele Grüße 
Heino

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Using the kostal-plenticore Integration - no manual configuration
```

### Log details

```text
needed?
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v.0.200.9

Chris45671, 2025-03-12:
Dieses Phänomen habe ich auch schon festgestellt. Ich habe zwei der Wechselrichter im Einsatz und hier ist es noch ersichtlichen, da der zweite auch per AC den Akku lädt. 

Da dies aber keine Auswirkung auf die Funktion von EVCC hat, da der Überschuss gemessen wird, hat es mich soweit nur gestört. Aber vielleicht könnte man es wirklich noch anpassen.

premultiply, 2025-03-12:
Bei Hybrid-WR nicht anders möglich.

closed 2025-03-12 by premultiply

Chris45671, 2025-03-12:
@premultiply Für mein Verständnis. Wenn ich, wie bei mir vorhanden, zwei Wechselrichter im Einsatz habe, sollte dies doch möglich sein oder? 

[image]

Dies wird mir für den zweiten Wechselrichter ausgegeben an dem keine Batterie hängt.

premultiply, 2025-03-12:
Dann probier ein Standard SunSpec Inverter Template für diesen einen Hybrid-WR ohne Batterie (warum nimmt man dann sowas?) wenn es dich so stört und der WR SunSpec unterstützt.

Mag in dem **Spezialfall** bei dem einen WR möglich sein. Ansonsten allgemein nicht. Hybrid-WR sind immer ein Kompromiss.